gphoto/libgphoto2 #471: sony alpha7r4, can't capture image

pdeman, 2020-01-28:
I am trying to capture image with a sony alpha7r4 and 
gphoto2         2.5.23.1      
libgphoto2      2.5.23.1  
libgphoto2_port 0.12.0  

but it says:
*** Error ***              
Sorry, your camera does not support generic capture
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***       

here is the result of gphoto2 --summary
gphoto2 --summary
Camera summary:                                                                
Manufacturer: Sony Corporation
Model: ILCE-7RM4
  Version: 1.0
  Serial Number: 00000000000000003285073003777959
Vendor Extension ID: 0x11 (1.0)

Capture Formats: 
Display Formats: Association/Directory, MPEG, JPEG, Unknown(b982), Unknown(b301), Undefined Type, ARW, Unknown(b310), MS Wave

Device Capabilities:
	File Download, No File Deletion, No File Upload
	No Image Capture, No Open Capture, No vendor specific capture

Storage Devices Summary:
store_00010001:
	StorageDescription: Storage Media
	VolumeLabel: None
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Generic Hierarchical
	Access Capability: Read-Only
	Maximum Capability: 62176362496 (59296 MB)
	Free Space (Bytes): 62100996096 (59224 MB)
	Free Space (Images): -1

Device Property Summary:
Battery Level(0x5001):(read only) (type=0x2) Enumeration [0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,100] value: 100% (100)
Property 0xd402:(read only) (type=0xffff) 'ILCE-7RM4'
Property 0xd406:(readwrite) (type=0xffff) '(null)'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd303:(read only) (type=0x2) 1

and here is the result of list-all-config:
gphoto2 --list-all-config
/main/actions/opcode                                                           
Label: PTP Opcode
Readonly: 0
Type: TEXT
Current: 0x1001,0xparam1,0xparam2
END
/main/status/serialnumber
Label: Serial Number
Readonly: 0
Type: TEXT
Current: 00000000000000003285073003777959
END
/main/status/manufacturer
Label: Camera Manufacturer
Readonly: 0
Type: TEXT
Current: Sony Corporation
END
/main/status/cameramodel
Label: Camera Model
Readonly: 0
Type: TEXT
Current: ILCE-7RM4
END
/main/status/deviceversion
Label: Device Version
Readonly: 0
Type: TEXT
Current: 1.0
END
/main/status/vendorextension
Label: Vendor Extension
Readonly: 0
Type: TEXT
Current: None
END
/main/status/Battery Level
Label: Battery Level
Readonly: 1
Type: TEXT
Current: 100%
END
/main/other/5001
Label: Battery Level
Readonly: 1
Type: MENU
Current: 100
Choice: 0 0
Choice: 1 1
Choice: 2 2
Choice: 3 3
Choice: 4 4
Choice: 5 5
Choice: 6 6
Choice: 7 7
Choice: 8 8
Choice: 9 9
Choice: 10 10
Choice: 11 11
Choice: 12 12
Choice: 13 13
Choice: 14 14
Choice: 15 15
Choice: 16 16
Choice: 17 17
Choice: 18 18
Choice: 19 19
Choice: 20 20
Choice: 21 21
Choice: 22 22
Choice: 23 23
Choice: 24 24
Choice: 25 25
Choice: 26 26
Choice: 27 27
Choice: 28 28
Choice: 29 29
Choice: 30 30
Choice: 31 31
Choice: 32 32
Choice: 33 33
Choice: 34 34
Choice: 35 35
Choice: 36 36
Choice: 37 37
Choice: 38 38
Choice: 39 39
Choice: 40 40
Choice: 41 41
Choice: 42 42
Choice: 43 43
Choice: 44 44
Choice: 45 45
Choice: 46 46
Choice: 47 47
Choice: 48 48
Choice: 49 49
Choice: 50 50
Choice: 51 51
Choice: 52 52
Choice: 53 53
Choice: 54 54
Choice: 55 55
Choice: 56 56
Choice: 57 57
Choice: 58 58
Choice: 59 59
Choice: 60 60
Choice: 61 100
END
/main/other/d402
Label: PTP Property 0xd402
Readonly: 1
Type: TEXT
Current: ILCE-7RM4
END
/main/other/d406
Label: PTP Property 0xd406
Readonly: 0
Type: TEXT
Current: (null)
END
/main/other/d407
Label: PTP Property 0xd407
Readonly: 1
Type: TEXT
Current: 1
END
/main/other/d303
Label: PTP Property 0xd303
Readonly: 1
Type: TEXT
Current: 1
END

and indeed there is nothing about capture image in it.
is the camera supported by libgphoto2 ? I thought I had seen it is but I can't find the up to date list again.

msmeissn, 2020-01-28:
can you switch the camera to "PC Control" mode?

pdeman, 2020-01-28:
it is in pc control mode

msmeissn, 2020-01-28:
hmm, can you attach a debuglogfile

gphoto2 --debug --debug-logfile=log.txt --summary --capture-image

and attach log.txt ?

pdeman, 2020-01-28:
here are the results: 
[log.txt](https://github.com/gphoto/libgphoto2/files/4123182/log.txt)
gphoto2 --debug --debug-logfile=log.txt --summary --capture-image
                                                                               
*** Error ***              
PTP Timeout

*** Error ***              
An error occurred in the io-library ('Timeout reading from or writing to the port'): No error description available

*** Error ***              
PTP Timeout

*** Error ***              
An error occurred in the io-library ('Timeout reading from or writing to the port'): No error description available
ERROR: Could not capture image.
ERROR: Could not capture.

*** Error ***              
PTP Timeout

*** Error ***              
An error occurred in the io-library ('Timeout reading from or writing to the port'): No error description available

msmeissn, 2020-01-28:
At this time the camera was in a corrupted state and not reacting to commands... can you replugin the camera and try again?

pdeman, 2020-01-28:
gphoto2 --debug --debug-logfile=log2.txt --summary --capture-image
Camera summary:                                                                
Manufacturer: Sony Corporation
Model: ILCE-7RM4
  Version: 1.10
  Serial Number: 00000000000000003285073003777959
Vendor Extension ID: 0x11 (1.0)
Vendor Extension Description: Sony PTP Extensions

Capture Formats: 
Display Formats: JPEG, Unknown(b301), ARW

Device Capabilities:
	File Download, No File Deletion, File Upload
	No Image Capture, No Open Capture, Sony Capture

Storage Devices Summary:

Device Property Summary:
Compression Setting(0x5004):(readwrite) (type=0x2) Enumeration [16,20,19,18,4,3,2] value: RAW (4)
White Balance(0x5005):(readwrite) (type=0x4) Enumeration [] value: Automatic (2)
F-Number(0x5007):(readwrite) (type=0x4) Range [0 - 65535, step 1] value: f/4 (400)
Focus Mode(0x500a):(readwrite) (type=0x4) Enumeration [2,32773,32772,32774,1] value: 32772
Exposure Metering Mode(0x500b):(readwrite) (type=0x4) Enumeration [32769] value: 32769
Flash Mode(0x500c):(readwrite) (type=0x4) Enumeration [] value: Automatic flash (1)
Exposure Program Mode(0x500e):(readwrite) (type=0x4) Enumeration [32768] value: 32768
Exposure Bias Compensation(0x5010):(readwrite) (type=0x3) Enumeration [] value: 0,0 stops (0)
Still Capture Mode(0x5013):(readwrite) (type=0x4) Enumeration [1,32786,32789,2,32784,32772,32771,32773,32776,32777,32780,32781,32782,32783] value: Single Shot (1)
DOC Compensation(0xd200):(readwrite) (type=0x3) Enumeration [] value: 0
DRangeOptimize(0xd201):(readwrite) (type=0x2) Enumeration [31] value: 31
Image size(0xd203):(readwrite) (type=0x2) Enumeration [1,2,3] value: 1
Shutter speed(0xd20d):(readwrite) (type=0x6) Range [0 - 4294967295, step 1] value: 65696
Property 0xd20e:(readwrite) (type=0x2) Enumeration [] value: 15
Color temperature(0xd20f):(readwrite) (type=0x4) Range [2500 - 9900, step 100] value: 5500
CC Filter(0xd210):(readwrite) (type=0x2) Range [156 - 228, step 1] value: 192
Aspect Ratio(0xd211):(readwrite) (type=0x2) Enumeration [1,3,2,4] value: 1
Focus status(0xd213):(readwrite) (type=0x2) Enumeration [] value: 1
Property 0xd214:(readwrite) (type=0x6) Range [0 - 4294967295, step 1] value: 4294967295
Objects in memory(0xd215):(readwrite) (type=0x4) Range [0 - 65535, step 1] value: 0
Property 0xd217:(readwrite) (type=0x2) Enumeration [] value: 1
Battery Level(0xd218):(readwrite) (type=0x1) Range [-1 - 100, step 1] value: 82
Picture Effect(0xd21b):(readwrite) (type=0x4) Enumeration [32768] value: 32768
AB Filter(0xd21c):(readwrite) (type=0x2) Range [156 - 228, step 1] value: 192
Property 0xd21d:(readwrite) (type=0x2) Range [0 - 2, step 1] value: 0
ISO(0xd21e):(readwrite) (type=0x6) Enumeration [] value: 16777215
Property 0xd21f:(readwrite) (type=0x2) Enumeration [] value: 1
Property 0xd220:(readwrite) (type=0x2) Enumeration [] value: 1
Property 0xd221:(readwrite) (type=0x2) Enumeration [] value: 0
Property 0xd222:(readwrite) (type=0x4) Enumeration [1,17,16] value: 16
Property 0xd223: error 2002 on query.
Property 0xd224:(readwrite) (type=0x3) Enumeration [] value: 0
Property 0xd226:(readwrite) (type=0x6) Enumeration [16777215] value: 16777215
Property 0xd227:(readwrite) (type=0x2) Enumeration [] value: 0
Property 0xd228:(readwrite) (type=0x2) Enumeration [4,5] value: 5
Property 0xd229:(readwrite) (type=0x6) Enumeration [19660810,16384010,13107210,9830410,8519690,6553610,5242890,3932170,3276810,2621450,2097162,1638410,1310730,1048586,851978,655370,524298,393226,327690,262154,65539,65540,65541,65542,65544,65546,65549,65551,65556,65561,65566,65576,65586,65596,65616,65636,65661,65696,65736,65786,65856,65936,66036,66176,66336,66536,66786,67136,67536,68036,68736,69536,70536,71936,73536] value: 65696
Property 0xd22a:(readwrite) (type=0x2) Enumeration [] value: 1
Property 0xd22b:(readwrite) (type=0x2) Enumeration [] value: 0
Property 0xd235:(readwrite) (type=0x2) Enumeration [0,1] value: 0
Property 0xd23e:(readwrite) (type=0x2) Enumeration [] value: 0
Property 0xd23f:(readwrite) (type=0x2) Enumeration [0,1,2,3,4,5,6,7,8,9,10] value: 0
Property 0xd240:(readwrite) (type=0x2) Enumeration [1,14,15,16,17,18,19] value: 1
Property 0xd241:(readwrite) (type=0x2) Enumeration [8,9,3] value: 9
Property 0xd242:(readwrite) (type=0x4) Enumeration [4,25,5,27] value: 4
Property 0xd243:(readwrite) (type=0x4) Enumeration [] value: 2
Property 0xd244:(readwrite) (type=0x4) Enumeration [] value: 2
Property 0xd245:(readwrite) (type=0x2) Enumeration [] value: 1
Property 0xd246:(readwrite) (type=0x2) Enumeration [] value: 2
Property 0xd247:(readwrite) (type=0x2) Enumeration [0,1,2,3,17,18,19,20] value: 0
Property 0xd248:(readwrite) (type=0x2) Enumeration [] value: 1
Property 0xd249:(readwrite) (type=0x6) Range [0 - 4294967295, step 1] value: 1383
[... 10 lines cut ...]
Property 0xd25f:(readwrite) (type=0x2) Enumeration [1,3,4] value: 1
Property 0xd260:(readwrite) (type=0x2) Enumeration [] value: 1
Property 0xd265:(readwrite) (type=0x2) Enumeration [] value: 0
Property 0xd266:(readwrite) (type=0x2) Enumeration [0,1] value: 0
Autofocus(0xd2c1):(readwrite) (type=0x4) Enumeration [1,2] value: 2
Capture(0xd2c2):(readwrite) (type=0x4) Enumeration [1,2] value: 2
Property 0xd2c3:(readwrite) (type=0x4) Enumeration [1,2] value: 2
Property 0xd2c5:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Still Image(0xd2c7):(readwrite) (type=0x4) Enumeration [1,2] value: 1
Movie(0xd2c8):(readwrite) (type=0x4) Enumeration [1,2] value: 2
[... cut ...]
Property 0xd2d6:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Property 0xd2d7:(readwrite) (type=0x4) Enumeration [1,2] value: 2
Property 0xd2d8:(readwrite) (type=0x4) Enumeration [1,2] value: 2
Property 0xd2d9:(readwrite) (type=0x4) Enumeration [1,2] value: 2
Property 0xd2dd:(readwrite) (type=0x1) Range [-1 - 1, step 1] value: 1
Property 0xd2de:(readwrite) (type=0x4) Enumeration [1,2] value: 1

ERROR: Could not capture image.
ERROR: Could not capture.

here is the log file: https://drive.google.com/open?id=1BWOLDI5JhQM52UvEp0RNJap5nz7DFz7L

qtnemo, 2020-01-28:
Check the "Still Img. Save Dest." under "Network", "PC Remote Function".  If it's set to "Camera Only" then you can only "trigger" but not "capture".

pdeman, 2020-02-04:
so I tried to trigger and it works. but I'ld like to know the file name.
I tried this code 
[trigger.txt](https://github.com/gphoto/libgphoto2/files/4153569/trigger.txt)
sample trigger but I removed the file copy.
and I get:
triggering capture 1
done triggering
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580822996.570626
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580822996.677394
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580822996.802707
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580822996.942271
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580822997.055561
triggering capture 2
done triggering
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580822998.268826
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580822998.380576
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580822998.507355
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580822998.623899
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580822998.740673
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580822998.847289
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580822998.962874
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580822999.086177
triggering capture 3
done triggering
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580823000.562894
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580823000.680860
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
[... 19 lines cut ...]
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580823002.526913
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580823002.644622
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580823002.762538
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580823002.880310
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580823003.003148
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
Reading PTP event failed: Timeout reading from or writing to the port (-10)
loop is at 1580823003.127792
triggering capture 5
done triggering

I never succeed to know the file name....

qtnemo, 2020-02-05:
I could not access the MicroSD card in any configuration.  If you figure it out please email me chu_bunATyahooDOTcom.  The main problem is this camera is new and not fully supported.  Many features are just missing or broken.

polakovic, 2020-02-24:
It looks that the difference between A7R4 and older models is, that the camera sends 0xc206 event with no parameter instead of expected 0xc201 event.

dhamp, 2020-09-09:
Sony released Camera Remote SDK (https://support.d-imaging.sony.co.jp/app/sdk/en/index.html) for ILCE-7SM3, ILCE-7RM4, ILCE-9M2
supported platforms: win64 (8.1,10), armv7(cortex a7){}, armv8(cortex a57/72, not sure working it on a53 or not)
it included libs with ptp2 protocol & sample app and api description in pdf files. 

@msmeissn, this sdk may helped with add support in libgphoto2 these cameras?

msmeissn, 2020-09-12:
can you try

gphoto2 --wait-event=3s --capture-image-and-download 

?

msmeissn, 2021-02-28:
likely fixed by initial wait. please test 2.5.27

closed 2021-02-28 by msmeissn

SiuyuenChow, 2022-05-17:
Have you solved the problem yet?
I recently had a similar problem.

The camera model is SONY A7M4, but the exact path and photo name cannot be found after taking photos

this is the log

ptp_sony_getalldevicepropdesc: Objects in memory(d215): value 1 -> 32769
DEBUG== 0xd215 after capture = 32769
SONY ObjectInMemory count change seen, retrieving file
Sending PTP_OC 0x1008 (Get object info) (0xffffc001) request...
Reading PTP_OC 0x1008 (Get object info) data...
Reading PTP_OC 0x1008 (Get object info) response...
PTPIP_END_DATA_PACKET
PTPIP_CMD_RESPONSE
ObjectInfo for 'DSC00174.JPG':
  Object ID: 0xffffc001
  StorageID: 0x00010000
  ObjectFormat: 0x3801
  ProtectionStatus: 0x0000
  ObjectCompressedSize: 306730
  ThumbFormat: 0x0000
  ThumbCompressedSize: 0
  ThumbPixWidth: 0
  ThumbPixHeight: 0
  ImagePixWidth: 0
  ImagePixHeight: 0
  ImageBitDepth: 0
  ParentObject: 0x00000000
  AssociationType: 0x0000
  AssociationDesc: 0x00000000
  SequenceNumber: 0x00000000
  ModificationDate: 0x00000000
  CaptureDate: 0x00000000
trying to get object size=0x4ae2a
Sending PTP_OC 0x1009 (Get object) (0xffffc001) request...
Reading PTP_OC 0x1009 (Get object) data...
Reading PTP_OC 0x1009 (Get object) response...
hdr type 8, length 18
PTPIP_END_DATA_PACKET
hdr type 8, length 18
PTPIP_CMD_RESPONSE
Append //capt0000.jpg to filesystem
Lookup folder '/'...
Found! / is 0x281569b30
Listing files in /
Lookup folder '/'...
Found! / is 0x281569b30
Querying folder /...
Delete all files in folder 0x281569b30//
file_list_func(/)
Internal append capt0000.jpg to folder /
Adding file 'capt0000.jpg' to folder '/' (type 1)...
Lookup folder / file capt0000.jpg
Lookup folder '/'...
Found! / is 0x281569b30
Setting key 'cached-images' to value '2' (libgphoto)
Saving 4 setting(s) to file "/private/var/mobile/Containers/Data/Application/8AD9F6EC-F596-45D7-A5AC-01CF9FEF31C3/Library/.gphoto/settings"
Adding file 'capt0000.jpg' from folder '/' to the fscache LRU list (type 1)...
Lookup folder / file capt0000.jpg
Lookup folder '/'...
Found! / is 0x281569b30
File 'capt0000.jpg' from folder '/' added in fscache LRU list.
Checking fscache LRU list integrity...
fscache LRU list ok with 1 items (306730 bytes)
Failed to find mime type for 0000
setting fileinfo in fs
Lookup folder / file capt0000.jpg
Lookup folder '/'...
Found! / is 0x281569b30