gphoto/libgphoto2 #472: Camera goes into Busy / locked / laptop symbol mode. Prevents further camera usage

busflix, 2020-02-04:
**What i am trying to do?**
Use camera and have pictures being backed-up on a raspberry pi(raspbian buster lite) connected via USB cable.

**Camera used**
canon 750D
canon 200D

**steps to reproduce**
gphoto2 --get-all-files --skip-existing
being executed repeatedly in  a loop. 5 sec delay after each iteration.

Camera is being used simultaneously to click more pictures. 
Image output jpg format being stored on SD card.

after some time the camera goes into busy mode(laptop logo symbol on screen) and no more pictures can be clicked forever. Camera stays locked.

running the command --summary, unlocks the camera immediately and camera can click pictures. Running --summary again puts the camera in busy mode. 

similarly the command gphoto2 --capture-tethered also toggles the camera in and out of the busy mode.

I DID NOT face the issue with 
canon 1200/1300 or any non 3 digit canon body.

**libgphoto2 and gphoto2 version 2.5.23**
built from source

**Side note**
When using Entangle(gui) in ubuntu 18.04
tethered capture works properly on the same devices.

elstoc, 2020-02-22:
Seems to be a common issue (see https://github.com/gphoto/gphoto2/issues/131). Not a long-term solution but does it work if you remove your SD card?

busflix, 2020-02-25:
This issue does not appear if the SD card is removed. But our whole purpose is to backup and send the photos immediately for further use. SD card is important to have a copy and see images on camera display.  When we use entangle( https://gitlab.com/entangle/entangle) which uses libgphoto, the issue not occur. Can anyone help to figure out what entangle does differently? Thanks

busflix, 2020-03-11:
can somebody please suggest a workaround to avoid this issue?

elstoc, 2020-03-11:
There's a new version of libgphoto2 out. Might be worth a try. Other than that the only thing I've found is to capture and then download over USB.

busflix, 2020-03-11:
Tried the latest build(as of today), but the issue still persists. any workaround to prevent the camera lock/busy mode would be a great help. Thanks

elstoc, 2020-03-11:
I feel your pain. I raised the issue in 2018 but I'm afraid I don't know nearly enough to investigate further.