gphoto/libgphoto2 #473: Kindle Fire 7 not recognised

robertbilling, 2020-02-17:
I have a Kindle Fire 7 bought around 2015, which was not recognised by the library. On investigation it turned out that the USB id in the library source was 1949:0221, the device was reporting 1949:0121.

My versions are:

gphoto2         2.5.23         gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.23         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial lockdev locking

As the device seems to have several different IDs depending on the version I have added the correct ID rather than changing an existing one.

This patch fixes the problem:

--- ./camlibs/ptp2/music-players.h.orig 2020-02-17 15:27:11.740426687 +0000
+++ ./camlibs/ptp2/music-players.h      2020-02-17 15:30:26.583154300 +0000
@@ -3377,6 +3377,9 @@
   /* https://sourceforge.net/p/libmtp/bugs/1448/ */
   { "Amazon", 0x1949, "Kindle Fire 7", 0x0221,
       DEVICE_FLAGS_ANDROID_BUGS },
+  /* R Billing tested one unit rbilling@tanglewood.online */
+  { "Amazon", 0x1949, "Kindle Fire 7 (3rd ID)", 0x0121,
+      DEVICE_FLAGS_ANDROID_BUGS },
   /* https://sourceforge.net/p/libmtp/bugs/1491/ */
   { "Amazon", 0x1949, "Kindle Fire 5", 0x0222,
       DEVICE_FLAGS_ANDROID_BUGS },

robertbilling, 2020-02-21:
Thanks very much.

msmeissn, 2020-03-01:
fixed by commit e3de3a9eca, is in 2.5.24 release.

closed 2020-03-01 by msmeissn