gphoto/libgphoto2 #476: Plese consider taggong conventoion change

kloczek, 2020-02-24:
git automatically adds repo name to autogenerated tar balls from git tags. Because current tagginbg convention is `libgphoto2-<version>` thise tar balls are with base directory `libgphoto2-libgphoto2-<version>`. Additionally in version all dots are replaced by underscores.

Plese consider change tagging convention to just `<version>` or 'v<version> (git automatically strips leading `v` in version) with dots instead underscores to match version with what is used in release and like it is used in most of the git based projects.

msmeissn, 2020-02-24:
I will see I do that for the next releases

msmeissn, 2020-02-24:
actually I changed the current release to use v2.5.24

closed 2020-02-24 by msmeissn