gphoto/libgphoto2 #477: Nikon P1000

lattard, 2020-02-25:
Hi are there any plans to include support for Nikon coolpix P1000 please ?

hfiguiere, 2020-02-25:
You need to be more specific. What have you tried and what doesn't work?

Lot of support come from working the same as another model of the same brand/vendor.

lattard, 2020-02-25:
hi, I've just installed gphoto2 and upon trying:
gphoto2 --capture-image

I get the the error message :
An error occurred in the io-library ('Could not lock the device'): Camera
is already in use.
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-60: 'Could not lock the device') ***

And since I did not find the Nikon P1000 camera in the list of supported
cameras I was wondering if it is actually supported.

When I try : gphoto2 --auto-detect
I get USB PTP Class Camera           usb:003,025

msmeissn, 2020-02-26:
this means the camera is in use , likely by gvfs ... can you umount it from nautilus and try again?

Also report the usb ids, using "lsusb"

lattard, 2020-02-26:
yes in fact it did capture  after unmounting, but the photo is not in focus, even if I am seeing that the auto focus mode is on.
Focus Mode(0x500a):(read only) (type=0x4) Enumeration [1,32784,32785,32786,32787] value: AF-S (32784)

msmeissn, 2020-02-26:
the coolpix capture is not really supported by Nikon, so it might have bugs. 
can you report:
- Record the output of lsusb to get the USB ids.
- Record the output of LANG=C gphoto2 --summary >summary.txt to get generic summary information into the summary.txt file.
- Record the output of LANG=C gphoto2 --list-all-config >config.txt to get the configuration tree into the config.txt file.

np, 2020-12-16:
Hi,

I would be interested by the support for the P1000 model.

Here is my report:

* lsusb: `Bus 001 Device 079: ID 04b0:0232 Nikon Corp.`
* [config.txt](https://github.com/gphoto/libgphoto2/files/5704614/config.txt)
* [summary.txt](https://github.com/gphoto/libgphoto2/files/5704615/summary.txt)

In particular I would be interested by the live view capability.

hfiguiere, 2020-12-16:
FWIW, it is not supported by Nikon own webcam utility:

https://downloadcenter.nikonimglib.com/en/download/sw/186.html

msmeissn, 2020-12-20:
did you try:
gphoto2 --capture-image-and-download   
?

Can you capture 
gphoto2 --summary --debug --debug-logfile=summary-debug.txt and send summary-debug.txt ?

DesertOnSea, 2024-01-28:
Very interested to know if anyone tried it on COOLPIX Nikon P1000. Thanks!