gphoto/libgphoto2 #478: Pentax driver broken with latest updates.

DrawsACircle, 2020-02-25:
**Describe the bug**

kStars/Indi cannot reports an I/O error when connecting to my Pentax K-70 whereafter the camera is disconnected, this with the latest libgphoto2 updates.
Error reported: Pentax DSLR K70 : "[ERROR] Camera_get_config failed (-7): I/O problem "
The same problem is reported by K5ii users on Pentaxforums.com
Stellarmate 1.5.0 & 1.5.1 on an RPi4.

**Name the camera**
Pentax DSLR K70

**libgphoto2 and gphoto2 version**
libgphoto2 2.5.23.1
gphoto2 2.5.20

**To Reproduce**
Connect to camera via kStars/Indi.
Or by running the --list-config command.

Logfile attached.

[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/4251655/my-logfile.txt)

msmeissn, 2020-02-26:
0.191026 gp_port_set_error [gphoto2-port.c:1190](0): 305bning af '/dev/sg0' (Adgang n346gtet) fejlede.
0.191108 gp_context_error            (0): An error occurred in the io-library ('I/O problem'): 305bning af '/dev/sg0' (Adgang n346gtet) fejlede.

can you check if the permissions of /dev/sg0 are correct?

DrawsACircle, 2020-02-26:
Hi Marcus, I ran the usb-devices command, the attached file is the output from executing a "udevadm info --attribute-walk --name=/dev/sg0" command.
Output from usb-devices:
[udevadm command.txt](https://github.com/gphoto/libgphoto2/files/4254447/udevadm.command.txt)

T:  Bus=01 Lev=02 Prnt=02 Port=00 Cnt=01 Dev#=  3 Spd=480 MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=25fb ProdID=017c Rev=01.11
S:  Manufacturer=RICOH IMAGING COMPANY, LTD.
S:  Product=PENTAX K-70
S:  SerialNumber=6394324
C:  #Ifs= 1 Cfg#= 1 Atr=c0 MxPwr=100mA
I:  If#=0x0 Alt= 0 #EPs= 2 Cls=08(stor.) Sub=06 Prot=50 Driver=usb-storage

Blueshawk, 2020-03-06:
Just for reference, My K5 was working on indi last weekend but I think I'm running stable versions on that system. (Jasem's suggestion after I hit my frustration limit break) 
@Drawsacircle: Maybe you can compile an older version or purge everything and change to stable to see if that helps? Just a thought, make sure dialout has permission to open the port too.

DrawsACircle, 2020-03-06:
As stated on indilib.org Blueshawk, I’m using the official release, no reason to risk ruining a night with clear skies. Furthermore, this has to work right out of the box, if not Stellarmate will loose it’s relevance for a lot of users who do not want to mess around with sudo commands.

knro, 2020-03-15:
Any update on this issue? is it permission or something else?

DrawsACircle, 2020-03-15:
Hi Jasem

No update yet, could be due to vacation/??
I have provided some logs for Marcus to look into, I also suspect it to be related to permission issues, the camera is detected on a hardware level, I can see it as an external drive and browse the pictures.

Regards
Keld Henningsen

Blueshawk, 2020-03-15:
@DrawsACircle  
OH that helps! You have to make sure the camera does NOT mount as a drive. Make sure it is in NSC mode and turn off automounting of usb drives.

DrawsACircle, 2020-03-15:
> @DrawsACircle
> OH that helps! You have to make sure the camera does NOT mount as a drive. Make sure the camera connection type is in mSC mode in camera menus and turn off automounting of usb drives. It also helps to put it in manual or user mode for config to work properly.
> 
> As for Stellarmate, it probably did work out of the box, but you might check it's update settings.

I did not change anything in Stellarmate, I want it to work right out of the box, I've used Stellarmate for almost 2 years with my K-70, no problems (apart from the fact that you cannot control exposure time and ISO). The present problem is related to the latest kStars/INDI/Gphoto update.
If the camera is not supposed to be auto mounted, then it's related to the latest version, not something I've done (the right out of the box thing :-) ). You should not need to be an engineer to figure out how to use Stellarmate. I am actually an electronics engineer, but sometimes you just want things to work without spending hours on bug fixing.

Blueshawk, 2020-03-15:
>  I am actually an electronics engineer, but sometimes you just want things to work without spending hours on bug fixing.

LOL! I feel your pain. You might want to consider taking up golf.

I made all my own stuff so I'm not sure about this, but you might try unchecking the automount box in Caja filemanager and reboot. Hope this helps.

@knro I know this isn't the place to report this but in case I get lazy: Stellarmate distro might need to add an automount disable in fstab, the volume manager, or udev rule for pentax sgx connections. I'm not quite sure how to do it right off. 

I have had the trouble even if I unmounted the camera once it was mounted and had to kill automount and reboot before things worked.

DrawsACircle, 2020-03-15:
> LOL! I feel your pain. You might want to consider taking up golf.
>
> I made all my own stuff so I'm not sure about this, but you might try unchecking the automount box in Caja filemanager and reboot. [...]

I can’t afford playing golf - astrophotography is a black hole budget wise, I can just imagine my wife if I started spending money on golf gear as well 😂 (I have 10+ rc planes)
I did unmount my K-70 when I was trying to figure out what was wrong, no changes. Also tried my Canon EOS1300da, no problems.

knro, 2020-03-16:
@DrawsACircle Try to disable automount and check again. Not sure why this isn't an issue with my Canon or Nikon though.

[image]

DrawsACircle, 2020-03-16:
> @DrawsACircle Try to disable automount and check again. Not sure why this isn't an issue with my Canon or Nikon though.
> 
> [image]

I just double checked Jasem, the camera was mounted because I clicked on the camera icon. 
The camera is not mounted automatically (at least when I right click the camera icon, I can choose to mount it).

knro, 2020-03-17:
Ok, I just released an update which fixes (hopefully) the permission problem. Can you please update SM and install all updates and try again?

DrawsACircle, 2020-03-17:
> Ok, I just released an update which fixes (hopefully) the permission problem. Can you please update SM and install all updates and try again? Make sure to restart SM after the updates before you test the camera again.

That did it Jasem! :-)
Only thing is I have to use the GPhoto CCD driver, the Pentax driver is not working anymore. But that's another story.