gphoto/libgphoto2 #482: Canon EOS 700D slow --capture-movie

Esokrates, 2020-03-11:
**Describe the bug**

I tried to use my EOS 700D as webcam by running

``` gphoto2 --capture-movie --stdout> fifo.mjpg ```

and view it with

```ffplay fifo.mjpg```

which is not fluent at all.

**Name the camera**
Canon EOS 700D

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version

**Debuglog**

```
gphoto2 --auto-detect --debug
0.000249 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000303 main                        (2): gphoto2 2.5.23
0.000320 main                        (2): gphoto2 has been compiled with the following options:
0.000337 main                        (2):  + gcc (C compiler used)
0.000347 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000354 main                        (2):  + exif (for displaying EXIF information)
0.000360 main                        (2):  + cdk (for accessing configuration options)
0.000366 main                        (2):  + aa (for displaying live previews)
0.000372 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000378 main                        (2):  + readline (for easy navigation in the shell)
0.000394 main                        (2): libgphoto2 2.5.23
0.000402 main                        (2): libgphoto2 has been compiled with the following options:
0.000409 main                        (2):  + all camlibs
0.000416 main                        (2):  + gcc (C compiler used)
0.000426 main                        (2):  + ltdl (for portable loading of camlibs)
0.000435 main                        (2):  + EXIF (for special handling of EXIF files)
0.000451 main                        (2): libgphoto2_port 0.12.0
0.000462 main                        (2): libgphoto2_port has been compiled with the following options:
0.000468 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.000474 main                        (2):  + gcc (C compiler used)
0.000480 main                        (2):  + ltdl (for portable loading of iolibs)
0.000487 main                        (2):  + USB (libusb1, for USB cameras)
0.000497 main                        (2):  + serial (for serial cameras)
0.000503 main                        (2):  + no resmgr (serial port access and locking)
0.000509 main                        (2):  + no ttylock (serial port locking)
0.000518 main                        (2):  + no lockdev (serial port locking)
0.000529 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000540 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000550 main                        (2): invoked with following arguments:
0.000556 main                        (2):   --auto-detect
0.000566 main                        (2):   --debug
0.000600 load_settings               (2): Creating gphoto config directory ('/home/user/.gphoto')
0.000678 load_settings               (2): Loading settings from file '/home/user/.gphoto/settings'.
0.000910 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/lib/x86_64-linux-gnu/libgphoto2_port/0.12.0'...
0.001043 foreach_func                (2): Called for filename '/usr/lib/x86_64-linux-gnu/libgphoto2_port/0.12.0/disk'.
0.001467 gp_port_library_list        (2): found fstab fsname UUID=be0fafa9-2c93-4717-bc39-06d8bbf0a232
0.001497 gp_port_library_list        (2): found fstab fsname UUID=be0fafa9-2c93-4717-bc39-06d8bbf0a232
0.001514 gp_port_library_list        (2): found fstab fsname UUID=be0fafa9-2c93-4717-bc39-06d8bbf0a232
0.001530 gp_port_library_list        (2): found fstab fsname UUID=9b786c9e-5b9c-4651-bc7a-85faf191166c
0.001551 gp_port_library_list        (2): found fstab fsname UUID=08B1-A5AD
0.001577 gp_port_library_list        (2): found fstab fsname none
0.001603 gp_port_library_list        (2): found fstab fsname tmpfs
0.001627 gp_port_library_list        (2): found fstab fsname tmpfs
0.001652 gp_port_library_list        (2): found fstab fsname tmpfs
0.001800 gp_port_library_list        (2): found mtab fsname sysfs
0.001815 gp_port_library_list        (2): found mtab fsname proc
0.001823 gp_port_library_list        (2): found mtab fsname udev
0.001844 gp_port_library_list        (2): found mtab fsname devpts
[... 217 lines cut ...]
0.043893 gp_libusb1_find_device_by_class_lib(2): Looking for USB device (class 0x6, subclass, 0x1, protocol 0x1)...
0.043923 gp_port_info_list_get_info  (2): Getting info of entry 35 (40 available)...
0.046006 gp_port_set_timeout         (2): Setting port timeout to 5000 milliseconds.
0.046073 gp_port_set_settings        (2): Setting settings...
0.046081 gp_libusb1_update           (2): (old int=0, conf=-1, alt=-1) port usb:001,022, (new int=0, conf=-1, alt=-1) port usb:001,022
0.046085 gp_libusb1_update           (2): lowlevel libusb1 port not yet opened, no need for libusb changes
0.046112 gp_abilities_list_detect_usb(2): Auto-detecting USB cameras...
0.046220 gp_libusb1_find_device_lib  (2): Looking for USB device (vendor 0x4a9, product 0x3272)... found.
0.046251 gp_libusb1_find_device_lib  (2): Detected defaults: config 1, interface 0, altsetting 0, inep 81, outep 02, intep 83, class 06, subclass 01
0.046254 gp_abilities_list_detect_usb(2): Found 'Canon EOS 700D' (0x4a9,0x3272)
0.046267 gp_port_free                (2): Freeing port...
0.046270 gp_port_close               (2): Closing port...
Model                          Port            
----------------------------------------------------------
Canon EOS 700D                 usb:001,022     
0.046680 gp_camera_free              (2): Freeing camera...
0.046685 gp_port_free                (2): Freeing port...
0.046688 gp_filesystem_reset         (2): resetting filesystem
0.046691 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
0.046693 gp_filesystem_lru_clear     (2): fscache LRU list already empty
0.046697 delete_all_folders          (2): Internally deleting all folders from '/'...
0.046699 lookup_folder               (2): Lookup folder '/'...
0.046701 lookup_folder               (2): Found! / is 0x56440129a8c0
0.046705 recurse_delete_folder       (2): Recurse delete folder 0x56440129a8c0//
```

msmeissn, 2020-03-14:
hmm, not fluent... if you record to file with just capture-movie and then play it, is it better?

This kind of viewing is really hackish though.  

Does it work better if you use e.g. "entangle" ?

tomato42, 2020-04-24:
it if fluid if you run `gphoto2 --capture-movie --stdout | ffplay -`?

Esokrates, 2020-04-26:
@msmeissn, @tomato42 Due to COVID19 it will take some time for me to get access to the camera and test again, so please be patient.

msmeissn, 2020-04-26:
No worries, this can wait :) Stay safe!

Esokrates, 2020-05-02:
> @Esokrates is it fluid if you run `gphoto2 --capture-movie --stdout | ffplay -`?

No it's stuttering unfortunately.

> hmm, not fluent... if you record to file with just capture-movie and then play it, is it better?

I couldn't play the file movie.mjpg, it opens up just as picture.
I recorded for at least over 10 seconds but it just reported 76 frames being captured.

tomato42, 2020-05-02:
> > @Esokrates is it fluid if you run `gphoto2 --capture-movie --stdout | ffplay -`?
> 
> No it's stuttering unfortunately.

did you try different cable or different port? (maybe it's negotiating USB 1.1 speeds instead of USB 2.0)

> > hmm, not fluent... if you record to file with just capture-movie and then play it, is it better?
> 
> I couldn't play the file movie.mjpg, it opens up just as picture.
> I recorded for at least over 10 seconds but it just reported 76 frames being captured.

try using  ffplay to play it (`ffplay movie.mjpeg`)

Esokrates, 2020-05-03:
@tomato42 I just have one cable, apart from that All my Ports are USB 3.0 / Thunderbird.
Playing the file with ffplay looks like fast forwarding a movie, it plays way too fast, any idea here?

Do you have the same camera and it works for you?

tomato42, 2020-05-03:
all USB 3.0 ports can negotiate USB 2.0 and even USB 1.1 speeds (in fact all mice and keyboards use USB 1.1 speeds only), including the USB-C ports. When you are copying data off of the SD card inside camera, what speeds you're getting?

if it plays too fast that indicates that the frames got dropped

I have 750D and yes, it works fine for me (fluid and lag no worse than the built-in webcam)

msmeissn, 2020-06-11:
I debugged this, and we had some logic that caused a much lower framerate than wanted.

closed 2020-06-11 by msmeissn

mre, 2020-06-15:
Thanks for your work @msmeissn.
Tried the latest version by compiling from source. (Only recompiled libgphoto2, not gphoto2 as that shouldn't be necessary AFAICT).

```
❯❯❯ gphoto2 --version
gphoto2 2.5.23

Copyright (c) 2000-2019 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         clang, popt(m), no exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.25         standard camlibs, clang, ltdl, no EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, clang, ltdl, no EXIF, USB, serial without locking
```

Ran `time gphoto2 --capture-movie` and kept it running for 1:21.29min. It captured 813 frames. That's an odd number considering that that's exactly 10.0fps. 

Am I doing something wrong and still using the old version?
If not, I wonder if we cap the framerate somewhere and that's the upper bound.

msmeissn, 2020-06-15:
2.5.25 still has the timing bug... It only is fixed in current libgphoto2 master.

I am planning to do a 2.5.26 soonish in the next weeks.

msmeissn, 2020-06-15:
you can try do 
gphoto2 --capture-movie=10s --debug --debug-logfile=/dev/null 
and see if it it gets mpore than 100 frames ,)

mre, 2020-06-15:
Thanks, here's what I get:

```
gphoto2 --capture-movie=10s --debug --debug-logfile=/dev/null
Capturing preview frames as movie to 'movie.mjpg' for 10 seconds.
Movie capture finished (84 frames)
```

Any way I can test the fix before the official release? I was compiling from `master` (commit 21147cc4152ff1d276d9dd236de2e777b686ef18). Think that should contain the fix.
For a test, I was manually bumping the version to 2.5.26 in `configure.ac:8`. After compiling, `gphoto` still shows 2.5.25, so I must be missing something...

msmeissn, 2020-06-15:
gphoto2 --version   should show 2.5.25.1 ...   check if you have the correct --prefix and --libdir during configure

mre, 2020-06-15:
You mean during configure of gphoto2, right? Still using the version from brew, which is probably the reason why it doesn't work. Compilation didn't work when I tried:

```
Undefined symbols for architecture x86_64:
  "_rl_copy_text", referenced from:
      _shell_completion_function in gphoto2-shell.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [gphoto2] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

I think that's because of missing readline dev headers, but I'll have to find out how to install them on macOS.

mre, 2020-06-15:
Okay I found [this post](http://harvey.nu/compiling_gphoto2-2.4.8_snow_leopard.html) and `./configure --without-readline` did the trick!

mre, 2020-06-15:
```
❯❯❯ gphoto2 --version
gphoto2 2.5.23.1

Copyright (c) 2000-2020 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23.1       gcc, popt(m), no exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.25.1       standard camlibs, gcc, ltdl, no EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, no EXIF, USB, serial without locking
```

In "camera" mode:

```
❯❯❯ gphoto2 --capture-movie=10s --debug --debug-logfile=/dev/null
Capturing preview frames as movie to 'movie.mjpg' for 10 seconds.
Movie capture finished (88 frames)

gphoto2 on  master took 11s
❯❯❯ gphoto2 --capture-movie=10s --debug --debug-logfile=/dev/null
Capturing preview frames as movie to 'movie.mjpg' for 10 seconds.
Movie capture finished (90 frames)

gphoto2 on  master took 10s
❯❯❯ gphoto2 --capture-movie=10s --debug --debug-logfile=/dev/null
Capturing preview frames as movie to 'movie.mjpg' for 10 seconds.
Movie capture finished (92 frames)
```

In movie mode:
```
gphoto2 on  master took 10s
❯❯❯ gphoto2 --capture-movie=10s --debug --debug-logfile=/dev/null
Capturing preview frames as movie to 'movie.mjpg' for 10 seconds.
Movie capture finished (137 frames)

gphoto2 on  master took 10s
❯❯❯ gphoto2 --capture-movie=10s --debug --debug-logfile=/dev/null
Capturing preview frames as movie to 'movie.mjpg' for 10 seconds.
Movie capture finished (135 frames)

gphoto2 on  master took 10s
❯❯❯ gphoto2 --capture-movie=10s --debug --debug-logfile=/dev/null
Capturing preview frames as movie to 'movie.mjpg' for 10 seconds.
Movie capture finished (134 frames)
```

msmeissn, 2020-06-15:
hmm. not much difference so far. :/

mre, 2020-06-15:
Attached a debug file created with
```
gphoto2 --capture-movie=10s --debug --debug-loglevel=debug --debug-logfile=debug.txt
```

[debug.txt](https://github.com/gphoto/libgphoto2/files/4780576/debug.txt)

mre, 2020-06-15:
Did some measurements on my Linux desktop machine as well.
Ubuntu 18.04 LTS, AMD FX(tm)-6300 Six-Core Processor 3.5GHz, 12GB RAM
Made a couple of measurments and the results are the same at 134 frames on average.

The debug log is... a bit long, so I split it up into multiple files and provide just the beginning and the very end here.

[debug00.txt](https://github.com/gphoto/libgphoto2/files/4783371/debug00.txt)
[debug01.txt](https://github.com/gphoto/libgphoto2/files/4783373/debug01.txt)
[debug02.txt](https://github.com/gphoto/libgphoto2/files/4783374/debug02.txt)
[debug9281.txt](https://github.com/gphoto/libgphoto2/files/4783377/debug9281.txt)

In hindsight I could have used a shorter recording duration but ¯\_(ツ)_/¯.
Let me know if you want me to move the conversation elsewhere. Don't want to spam a closed ticket. ;)

sim590, 2020-12-26:
What are the final results on this ? I have the same camera and still experiencing 134 frames per 10s in video mode which is not high enough IMO compared to others having things like 60 fps. I'm on version 2.5.26.

mre, 2020-12-26:
I haven't found a solution either. Still hoping for a patch to improve the situation.
In my opinion the issue should be reopened.

msmeissn, 2020-12-27:
I changed the wait log a bit.

For me its kind of different:

- EOS 100D - 13FPS still
- EOS 1000D - around 20FPS
- EOS 750D - 25 - 30 FPS

sim590, 2021-01-03:
Also, when I configure my Canon EOS 700D to capture video with configuration `1280x720, 60FPS`, in the camera menus I'm getting even worse performance: 5FPS by running `gphoto2 --capture-movie=100s`.