gphoto/libgphoto2 #483: Ubuntu PTP Device Busy

dgmonet, 2020-03-24:
UBUNTU is driving me nuts.  I have tried to follow the FAQs but no suggested
fix seems to help.  For a while, UBUNTU's version of gphoto2() was working
as advertised.  At some point, proper operation failed and it started to give
me "PTP Error" messages.  I rebooted, played around, and gphoto2() would
occasionally work and then revert to the PTP Error condition.  Finally,
that error has gone away and gphoto2() is now stuck at
   prompt> gphoto2 --port usb:003,013 --set-config imageformat=32
   *** Error ***              
   The property 'Image Format' / 0xd120 was not set (0x2019: PTP Device Busy).
This condition does not go away during a reboot or power cycle.  It
seems that UBUNTU now knows to allocate the camera to some other task thereby
causing gphoto2() to fail.  I have no idea what this task might be, nor
how to go looking for it to kill it.  As per previous FAQs, I have tried
killing gvfs-gphoto2-volume-monitor, but this has no effect that I can detect.

If it helps:
   prompt> uname -r
   5.3.0-42-generic
   prompt> cat /etc/os-release
   NAME="Ubuntu"
   VERSION="18.04.4 LTS (Bionic Beaver)"
   ID=ubuntu
   ID_LIKE=debian
   PRETTY_NAME="Ubuntu 18.04.4 LTS"
   VERSION_ID="18.04"
   HOME_URL="https://www.ubuntu.com/"
   SUPPORT_URL="https://help.ubuntu.com/"
   BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
   PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-poli
   cy"
   VERSION_CODENAME=bionic
   UBUNTU_CODENAME=bionic
   prompt> gphoto2 --version
   gphoto2 2.5.15
   
   Copyright (c) 2000-2017 Lutz Mueller and others
   
   gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
   redistribute copies of gphoto2 under the terms of the GNU General Public
   License. For more information about these matters, see the files named COPYING.
   
   This version of gphoto2 is using the following software versions and options:
   gphoto2         2.5.15         gcc, popt(m), exif, cdk, aa, jpeg, readline
   libgphoto2      2.5.16         all camlibs, gcc, ltdl, EXIF
   libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

As long as I am demonstrating my clueless nature about UBUNTU, I have stubbed
my toe agains another problem that appears in various FAQs but so far none
of the suggested fixes have helped.  It seemed like a good idea to use the
most recent version of libgphoto2 (2.5.24), at least back when I was not getting
the device already allocated errors.  I got the tarball and followed the
instructions.  "configure", "make clean", and "make" appear to work as
advertised.  Unfortunately, "make install" fails.  The relevant portion of
the transcript is

    /bin/mkdir -p '/home/dgm/linux/lib/libgphoto2_port/0.12.0'
    /bin/bash ./libtool   --mode=install /usr/bin/install -c   serial.la usb1.la usbdiskdirect.la usbscsi.la disk.la ptpip.la '/home/dgm/linux/lib/libgphoto2_port/0.12.0'
   libtool: warning: relinking 'serial.la'
   libtool: install: (cd /home/dgm/linux/libgphoto2-2.5.24/libgphoto2_port; /bin/bash "/home/dgm/linux/libgphoto2-2.5.24/libgphoto2_port/libtool"  --tag CC --mode=relink gcc -g -O2 -Wall -Wmissing-declarations -Wmissing-prototypes -module -no-undefined -avoid-version -export-dynamic -export-symbols ./iolib.sym -rpath /home/dgm/linux/lib/libgphoto2_port/0.12.0 -o serial.la serial/la-unix.lo ./libgphoto2_port/libgphoto2_port.la -lpthread )
   libtool: relink: echo "{ global:" > .libs/serial.ver
   libtool: relink:  cat ./iolib.sym | sed -e "s/\(.*\)/\1;/" >> .libs/serial.ver
   libtool: relink:  echo "local: *; };" >> .libs/serial.ver
   libtool: relink:  gcc -shared  -fPIC -DPIC  serial/.libs/la-unix.o   -Wl,-rpath -Wl,/home/dgm/linux/lib -L/home/dgm/linux/lib -lgphoto2_port -L/usr/lib/x86_64-linux-gnu -lltdl -ldl -lpthread  -g -O2   -Wl,-soname -Wl,serial.so -Wl,-version-script -Wl,.libs/serial.ver -o .libs/serial.so
   /usr/bin/ld: /home/dgm/linux/lib/libltdl.a(libltdl_libltdl_la-ltdl.o): relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
   /usr/bin/ld: /home/dgm/linux/lib/libltdl.a(libltdl_libltdl_la-preopen.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
   /usr/bin/ld: /home/dgm/linux/lib/libltdl.a(libltdl_libltdl_la-lt__alloc.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
   /usr/bin/ld: /home/dgm/linux/lib/libltdl.a(libltdl_libltdl_la-lt_dlloader.o): relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
   /usr/bin/ld: /home/dgm/linux/lib/libltdl.a(libltdl_libltdl_la-lt_error.o): relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
   /usr/bin/ld: /home/dgm/linux/lib/libltdl.a(libltdl_libltdl_la-slist.o): relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
   /usr/bin/ld: final link failed: Nonrepresentable section on output
   collect2: error: ld returned 1 exit status
   libtool:   error: error: relink 'serial.la' with the above command before installing it
   Makefile:799: recipe for target 'install-iolibLTLIBRARIES' failed
   make[3]: *** [install-iolibLTLIBRARIES] Error 1
   make[3]: Leaving directory '/home/dgm/linux/libgphoto2-2.5.24/libgphoto2_port'
   Makefile:1501: recipe for target 'install-am' failed
   make[2]: *** [install-am] Error 2
   make[2]: Leaving directory '/home/dgm/linux/libgphoto2-2.5.24/libgphoto2_port'
   Makefile:1191: recipe for target 'install-recursive' failed
   make[1]: *** [install-recursive] Error 1
   make[1]: Leaving directory '/home/dgm/linux/libgphoto2-2.5.24/libgphoto2_port'
   Makefile:714: recipe for target 'install-recursive' failed
   
I have no idea what serial.la might be nor how .rodata might have gotten
compiled without -fPIC - if indeed this is the issue.  There was a FAQ that
suggested doing "sudo make install", but this does not make the error
condition go away.

These problems are driving me nuts.  Any clues about how to proceed would
be gratefully appreciated.  I did not have these problems on CentOS, but
I must now use UBUNTU.  May thanks in advance.  Stay healthy.

dgmonet, 2020-03-28:
Regarding the first issue, I have done
   sudo chmod -x /usr/lib/gvfs/gvfs_gphoto2_volume_monitor
to keep this from restarting after each time I kill it.  I have verified that I am in group "plugdev".  I have tried running the gphoto2 commands using sudo, and my account is in the ACL for the USB device.  No change.  Within 30 seconds of the initial probe of the device, it comes back 0x2019 PTP Device Busy.  lsof does not report any process with this USB device open.

Regarding the failure to relink serial.la, I have tried to make libgphoto2-2.5.8, 16, 17, 22, 23, 24, and all attempt this relink and all fail with the same error message.

msmeissn, 2020-03-29:
For building, you seem to be missing a shared library variant of libltdl on your system, see if there is libtool-dev or libltdl...-dev something 

The PTP error 0x2019 is already a PTP side error, so on the Linux side the permissions are ok, just the camera is in a busy state for some reason.

What camera is this?

Can you capture a debug trace?