gphoto/libgphoto2 #484: Fuji X-E3

zipfmann, 2020-03-27:
First of all: Thanks for this great software. Worked like a charm with my older camera.

I switched to Fuji some months ago and would like to use the camera for some stop-motion-films. But unfortunately gphoto2/entangled don't work.

Maybe someone here can help :) 

**Versions**
I'm using Ubuntu 19.10. 
libgphoto Version in repositories was 2.5.23.

I've seen support for XE3 has been added in libgphoto2      2.5.24
Tried without success and switched to current-dev-version 2.5.24.1, but also without success.

```
gphoto2 --version
gphoto2 2.5.23.1

Copyright (c) 2000-2020 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23.1       gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.24.1       standard camlibs (SKIPPING lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

**Name of the camera*
Fujifilm X-E3

Firmware-Version 1.22

**USB IDs**

```
lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 04cb:02d6 Fuji Photo Film Co., Ltd USB PTP Camera
Bus 001 Device 003: ID 0a5c:6414 Broadcom Corp. BCM2045A0
Bus 001 Device 002: ID 13d3:5664 IMC Networks Lenovo EasyCamera
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

**camera summary output**

[summary.txt](https://github.com/gphoto/libgphoto2/files/4394795/summary.txt)

**camera configuration output**

[list-all-config.txt](https://github.com/gphoto/libgphoto2/files/4394800/list-all-config.txt)

**test capture**

  gphoto2 --capture-image-and-download
[capture-download.txt](https://github.com/gphoto/libgphoto2/files/4394812/capture-download.txt)

 
  gphoto2 --capture-preview

**additional info**

the fuji has two modes of usb-to-pc-connections. 
 * usb-card-reader (german: USB-Kartenleser)
 * usb-raw convert/safe (german: USB RAW-KONV./SICHER.)

my device is in mode usb-raw, because i read somewhere that it is necessary for tethering.
If switched to usb-card-reader, the output of gphoto2 --summary changes:

```
$ gphoto2 --summary
Kamerazusammenfassung:                                                         
Hersteller: FUJIFILM
Modell: X-E3
  Version: 1.22
  Seriennummer: 5935333932341710187F6330611DB2
Vendor Extension ID: 0xe (1.0)
Beschreibung der Vendor Extension: fujifilm.co.jp: 1.0; 

Aufnahmeformate: 
Anzeigeformate: JPEG, JFIF

Fähigkeiten des Geräts:
	Dateien Download, Löschen von Dateien, Hochladen von Dateien
	Keine Aufnahme von Bildern, keine Offene Aufnahme, Kein Vendor spezifischer Aufnahmemodus

Informationen über Speichermedium:
store_10000001:
	Beschreibung des Speichergeräts: External Memory
	Label des Speichergerätes: 000000000000000000000000000000000000000000000000
	Speichertyp: Entfernbarer Speicher (Karte)
	Typ des Dateisystems: Standard Digital Kamera Layout (DCIM)
	Zugriffsmöglichkeiten: Lesen und Schreiben
	Maximale Kapazität: 15931539456 (15193 MB)
	Freier Platz (Bytes): 7178158080 (6845 MB)
	Freier Speicher (Bilder): -1

Properties des Geräts:
Batterie Ladezustand(0x5001):(read only) (type=0x2) Range [0 - 10, step 1] value: 10% (10)
Property 0xd303:(read only) (type=0x2) 1
Property 0xd406:(readwrite) (type=0xffff) ''
Property 0xd407:(read only) (type=0x6) 1
Belichtungszeit(0x500d):PTP Fehler 200a beim Abfragen
F-Nummer(0x5007):PTP Fehler 200a beim Abfragen
Property 0xd38c:PTP Fehler 200a beim Abfragen
Property 0xd171:PTP Fehler 200a beim Abfragen
Property 0xd21c:PTP Fehler 200a beim Abfragen
Focus Point(0xd347):PTP Fehler 200a beim Abfragen

```

Panzke, 2020-04-16:
Same Problem for me. Here my summary in english

`
pi@photobox:~ $ gphoto2 --summary
Camera summary:                                                                
Manufacturer: FUJIFILM
Model: X-E3
  Version: 1.22
  Serial Number: 593533393234190925AB93303126E2
Vendor Extension ID: 0xe (1.0)
Vendor Extension Description: fujifilm.co.jp: 1.0; 

Capture Formats: Defined Type JPEG Unknown(b103)
Display Formats: Defined Type, JPEG, Unknown(b103)

Device Capabilities:
	File Download, File Deletion, No File Upload
	Generic Image Capture, Open Capture, No vendor specific capture

Storage Devices Summary:
store_10000001:
	StorageDescription: Still
	VolumeLabel: 1593533393234190925AB93303126E2
	Storage Type: Builtin RAM
	Filesystemtype: Generic Hierarchical
	Access Capability: Read Only with Object deletion
	Maximum Capability: 18446744073709551615 (4294967295 MB)
	Free Space (Bytes): 18446744073709551615 (4294967295 MB)
	Free Space (Images): -1
store_10000002:
	StorageDescription: Live
	VolumeLabel: 2593533393234190925AB93303126E2
	Storage Type: Builtin RAM
	Filesystemtype: Generic Hierarchical
	Access Capability: Read Only with Object deletion
	Maximum Capability: 18446744073709551615 (4294967295 MB)
	Free Space (Bytes): 18446744073709551615 (4294967295 MB)
	Free Space (Images): -1

Device Property Summary:
Battery Level(0x5001): error 200a on query.
Image Size(0x5003): error 200a on query.
White Balance(0x5005): error 200a on query.
Focus Mode(0x500a): error 200a on query.
Exposure Metering Mode(0x500b): error 200a on query.
Flash Mode(0x500c): error 200a on query.
Exposure Program Mode(0x500e): error 200a on query.
Exposure Index (film speed ISO)(0x500f): error 200a on query.
Date & Time(0x5011): error 200a on query.
Pre-Capture Delay(0x5012): error 200a on query.
Sharpness(0x5015): error 200a on query.
Burst Number(0x5018): error 200a on query.
Burst Interval(0x5019): error 200a on query.
Focus Metering Mode(0x501c): error 200a on query.
Property 0xd001: error 200a on query.
Property 0xd002: error 200a on query.
Property 0xd003: error 200a on query.
Property 0xd004: error 200a on query.
Property 0xd005: error 200a on query.
Property 0xd007: error 200a on query.
Property 0xd008: error 200a on query.
Property 0xd009: error 200a on query.
Property 0xd00a: error 200a on query.
Property 0xd00b: error 200a on query.
Property 0xd00c: error 200a on query.
Property 0xd00d: error 200a on query.
Property 0xd00e: error 200a on query.
Property 0xd00f: error 200a on query.
Property 0xd010: error 200a on query.
Property 0xd011: error 200a on query.
[... 121 lines cut ...]
Property 0xd20d: error 200a on query.
Property 0xd20e: error 200a on query.
Property 0xd20f: error 200a on query.
Property 0xd210: error 200a on query.
Property 0xd211: error 200a on query.
Property 0xd212: error 201b on query.
Property 0xd213: error 200a on query.
Property 0xd214: error 200a on query.
Property 0xd215: error 200a on query.
Property 0xd216: error 200a on query.
Property 0xd217: error 200a on query.
Property 0xd218: error 200a on query.
Property 0xd219: error 200a on query.
Property 0xd21a: error 200a on query.
Property 0xd21b: error 200a on query.
Property 0xd406:(readwrite) (type=0xffff) ''
Property 0xd407:(read only) (type=0x6) 1
Exposure Time(0x500d): error 200a on query.
F-Number(0x5007): error 200a on query.
Property 0xd38c: error 200a on query.
Property 0xd171: error 200a on query.
Property 0xd21c:(readwrite) (type=0x4) Enumeration [0,1,2,3,4,5,6,7,8,9,10,11] value: 3
Focus Point(0xd347): error 200a on query.

`

mzealey, 2020-05-31:
I think X-E series like my X-T100 does not support usb tethered shooting, only wifi-based unfortunately

msmeissn, 2020-06-14:
Yes, apparently only Fuji XT-1, XT-2, XT-3, XT-4, X-H1, ANd X_pro2 currently support capture over usb.

Panzke, 2020-06-15:
Then we hope for an update or something similar that will solve the problem :-)

Thank you for your effort!

msmeissn, 2020-06-15:
(there might be chance over WIFI, but there I think we have not yet implemented the protocol)

cguennou, 2020-12-21:
Hey guys,
Just get my xe3 too, and got stuck on the same issue. Any news regarding support for X-E series?
Thanks a lot !

msmeissn, 2020-12-22:
remote control over USB is not possible, as a camera side limitation... Nothing we can do there.

We have not tested it over WIFI.