gphoto/libgphoto2 #486: Incoporate Sony compatibility table

wferi, 2020-03-29:
I don't know whether it fits the project philosophy, but I'd have found it useful to know what features are expected to work with a given camera model. I also don't know how reliable such information is, but the feature tables for vendor remote control software may provide a useful starting point. I found these from Sony for example:
https://www.playmemoriescameraapps.com/portal/usbspec.php?eid=IS9104-NPIA09014_00-F00002
https://support.d-imaging.sony.co.jp/app/imagingedge/en/devices/#emount
I doubt such webpages are particularly stable, so linking to them might be of limited use. On the other hand, including them into `ptp_device_properties` might be too restrictive without some override possibility...

msmeissn, 2020-04-12:
could link it from the webpage. But as you write such links get outdated kind of quickly.

wferi, 2020-04-13:
A possible approach is to link them through the Wayback Machine:
https://web.archive.org/web/20180214065922/https://www.playmemoriescameraapps.com/portal/usbspec.php?eid=IS9104-NPIA09014_00-F00002
https://web.archive.org/web/20191003060434/https://support.d-imaging.sony.co.jp/app/imagingedge/en/devices/
Extracting the data from the tables isn't hard either, a simple copy&paste of the table body into LibreOffice Calc works perfectly.