gphoto/libgphoto2 #487: Locked up trying to download photos on Nikon D5200

davidaccadia, 2020-04-01:
**Describe the bug**

After taking a RAW (NEF) and then trying to download the photo the call to gp_file_new_from_fd() never returns.

**Name the camera**
Nikon DSC D5300. 
USB ID  04b0:0431

We've also been able to reproduce this issue on a D5200 and D5600

**libgphoto2 and gphoto2 version**
libgphoto2 version 2.5.24
We are not using gphoto2 but calling the libgphoto2 api directly from a custom 
c++ program.

**To Reproduce**
Steps to reproduce the behaviour:

call gp_file_new_from_fd() with a NEF file as the CameraFIle.
Never got it to fail with a JPEG but did a with RAW photos. 
Although the failure wasn't consistent

A log from a failed case is as below:

```
gPhoto DEBUG:(gp_camera_file_get) Getting file 'DSC_0268.NEF' in folder '/store_00010001/DCIM/110D5300'...
gPhoto DEBUG:(gp_camera_file_get) Getting file 'DSC_0268.NEF' in folder '/store_00010001/DCIM/110D5300'...
gPhoto DEBUG:(gp_filesystem_get_file_impl) Getting file 'DSC_0268.NEF' from folder '/store_00010001/DCIM/110D5300' (type 1)...
gPhoto DEBUG:(gp_filesystem_get_file_impl) Getting file 'DSC_0268.NEF' from folder '/store_00010001/DCIM/110D5300' (type 1)...
gPhoto DEBUG:(lookup_folder_file) Lookup folder /store_00010001/DCIM/110D5300 file DSC_0268.NEF
gPhoto DEBUG:(lookup_folder_file) Lookup folder /store_00010001/DCIM/110D5300 file DSC_0268.NEF
gPhoto DEBUG:(lookup_folder) Lookup folder '/store_00010001/DCIM/110D5300'...
gPhoto DEBUG:(lookup_folder) Lookup folder '/store_00010001/DCIM/110D5300'...
gPhoto DEBUG:(gp_filesystem_get_file_impl) Downloading 'DSC_0268.NEF' from folder '/store_00010001/DCIM/110D5300'...
gPhoto DEBUG:(gp_filesystem_get_file_impl) Downloading 'DSC_0268.NEF' from folder '/store_00010001/DCIM/110D5300'...
gPhoto DEBUG:(ptp) (storage=0x00010001, handle=0x00000000)
gPhoto DEBUG:(ptp) (storage=0x00010001, handle=0x00000000)
gPhoto DEBUG:(ptp) (storage=0x00010001, handle=0x11004000)
gPhoto DEBUG:(ptp) (storage=0x00010001, handle=0x11004000)
gPhoto DEBUG:(ptp) (storage=0x00010001, handle=0x111b8000)
gPhoto DEBUG:(ptp) (storage=0x00010001, handle=0x111b8000)
gPhoto DEBUG:(get_file_func) Getting file 'DSC_0268.NEF'.
gPhoto DEBUG:(get_file_func) Getting file 'DSC_0268.NEF'.
gPhoto DEBUG:(ptp_usb_sendreq) Sending PTP_OC 0x101b (Get partial object) (0x91b810c,0x0,0x100000) request...
gPhoto DEBUG:(ptp_usb_sendreq) Sending PTP_OC 0x101b (Get partial object) (0x91b810c,0x0,0x100000) request...
gPhoto DEBUG:(ptp_usb_getdata) Reading PTP_OC 0x101b (Get partial object) data...
gPhoto DEBUG:(ptp_usb_getdata) Reading PTP_OC 0x101b (Get partial object) data...
gPhoto DEBUG:(ptp_usb_getresp) Reading PTP_OC 0x101b (Get partial object) response...
gPhoto DEBUG:(ptp_usb_getresp) Reading PTP_OC 0x101b (Get partial object) response...
gPhoto DEBUG:(ptp_usb_sendreq) Sending PTP_OC 0x101b (Get partial object) (0x91b810c,0x100000,0x100000) request...
gPhoto DEBUG:(ptp_usb_sendreq) Sending PTP_OC 0x101b (Get partial object) (0x91b810c,0x100000,0x100000) request...
gPhoto DEBUG:(ptp_usb_getdata) Reading PTP_OC 0x101b (Get partial object) data...
gPhoto DEBUG:(ptp_usb_getdata) Reading PTP_OC 0x101b (Get partial object) data...
gPhoto DEBUG:(ptp_usb_getresp) Reading PTP_OC 0x101b (Get partial object) response...
gPhoto DEBUG:(ptp_usb_getresp) Reading PTP_OC 0x101b (Get partial object) response...
gPhoto DEBUG:(ptp_usb_sendreq) Sending PTP_OC 0x101b (Get partial object) (0x91b810c,0x200000,0x100000) request...
gPhoto DEBUG:(ptp_usb_sendreq) Sending PTP_OC 0x101b (Get partial object) (0x91b810c,0x200000,0x100000) request...
...
```
The last 6 lines repeat over and over again.  

The issue did not exist in libgphoto 2.5.22. After a bit of digging I think the issues is in the ptp2/library.c file.

I applied the attached patch to 2.5.24 commenting out parts of ptp2/library.c to how it was in 2.5.22.
 This seemed to fix the issue for the D5200 and D5300.
[partial_download_for_nikon_revert.patch.txt](https://github.com/gphoto/libgphoto2/files/4412572/partial_download_for_nikon_revert.patch.txt)

msmeissn, 2020-04-18:
gPhoto DEBUG:(ptp_usb_sendreq) Sending PTP_OC 0x101b (Get partial object) (0x91b810c,0x200000,0x100000) request...

is it repeating the lines with the same second parameter 0x200000?

Can you reproduce this with gphoto2 standalone tool?

davidaccadia, 2020-04-23:
I will try.

davidaccadia, 2020-05-07:
I can't get gphoto2 to produce the same log message. I've been running the command 
` gphoto2 --capture-image-and-download --debug --debug-loglevel=debug --force-overwrite --debug-logfile=log1.txt`  Is there a better way to drive it it?

I can confirm that Get partial object eventually repeats the same command with the same values over and over again.  However the first few times are different.
I've attached a longer log file 
[gphotolog.txt.gz](https://github.com/gphoto/libgphoto2/files/4590848/gphotolog.txt.gz)
.

msmeissn, 2020-05-24:
I have added a end of loop condition into current git, and also an error message.

davidaccadia, 2020-11-17:
Tested on 2.5.26.  Fixed

closed 2020-11-17 by davidaccadia