gphoto/libgphoto2 #488: Photo date Midnight November 30 1970

davidaccadia, 2020-04-01:
**Describe the bug**

Calls to gp_camera_file_get_info() sometimes have a CameraFileInfo
object with a mtime set to 312768000 (Midnight November 30 1970)

Only see it happen with NEF files never JPEGs

**Name the camera**
I've seen this on Nikon D5200, D5300, D5600 D7200.
(I think D3500 and D3400 too but need to confirm) 

Nikon DSC D5300.
USB ID 04b0:0431

**libgphoto2 and gphoto2 version**
libgphoto2 version 2.5.22

also able to reproduce it on libgphoto2 version 2.5.24

We are not using gphoto2 but calling the libgphoto2 api directly from a custom
c++ program.

**To Reproduce**
Steps to reproduce the behaviour:

take a photo with a call to gp_camera_trigger_capture()
then pool gp_camera_wait_for_event() with a 500 MS timeout.
Once an event GP_EVENT_FILE_ADDED has been found
call gp_camera_file_get_info() with the photo just taken.

Sometimes maybe 1 in every 20 photos the mtime will be 312768000.
But only for NEF RAW images.

msmeissn, 2020-04-12:
if you are using SD Card capture, this date and time is set by the camera.
If this is SDRAM capture, we set it using time(NULL) (these woild be the captXXXX.jpg files)

which one is it?

as it sound irregular, it might come from the camera

davidaccadia, 2020-04-14:
We are using SD capture and yes it is very irregular. Maybe 1 in every 20 photos but I've not been able to find a pattern.   When I manually check the exif data on the photo that has been downloaded it has the correct date.