gphoto/libgphoto2 #489: Canon EOS R movie capture

avisecag, 2020-04-03:
**Describe the bug**
Tried to capture a movie sequence with the Canon EOS R, but can not ste the widget to set the movierecordtarget

**Name the camera**
Canon EOS R

**libgphoto2 and gphoto2 version**
gphoto2         2.5.23
libgphoto2      2.5.24

libgphoto2_port 0.12.0

**To Reproduce**
gphoto2 --set-config viewfinder=1 --set-config movierecordtarget=Card --wait-event=10s --set-config movierecordtarget=None --wait-event-and-download=2s

gphoto2 --set-config viewfinder=1 --set-config movierecordtarget=0 --wait-event=10s --set-config movierecordtarget=1 --wait-event-and-download=2s

gphoto2 --set-config movierecordtarget=Card --wait-event=10s --set-config movierecordtarget=None --wait-event-and-download=2s

[eosR_moviecapture_debug.txt](https://github.com/gphoto/libgphoto2/files/4425484/eosR_moviecapture_debug.txt)

joelcollet, 2020-04-20:
May this be also related to viewfinder widget ?
https://github.com/gphoto/libgphoto2/issues/441

msmeissn, 2020-05-03:
can you add a --wait-event=1s  after the --set-config viewfinder=1  ?

joelcollet, 2020-05-04:
<span style="font-family:arial,helvetica,sans-serif; font-size:12px">Hello<br>
<br>
i think there is a mistake in the recipient for this mail. i have not opened  this issue nor do i have a canon eos-r<br>
<br>
Regards<br>
Joël</span>

<div class="gl_quote" style="margin-top: 20px; padding-top: 5px;">De : "Marcus Meissner"<br>
A : "gphoto/libgphoto2" ,"joelcollet" ,"Comment"<br>
Envoyé: dimanche 3 Mai 2020 17:44<br>
Objet : Re: [gphoto/libgphoto2] Canon EOS R movie capture (#489)<br>
 
<div class="gl_quoted">
<p> </p>

<p>can you add a --wait-event=1s after the --set-config viewfinder=1 ?</p>

<p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
</div>

<div class="gl_quoted"> </div>
</div>

msmeissn, 2020-05-22:
@joelcollet not sure how you got in here, I think you need to press unsubscribe on the right side.

What the original reporter can also try is --set-config capturetarget=card

shahms, 2021-01-06:
FWIW, I was experiencing the same problem on my Canon EOS RP and adding the additional wait fixed the issue.