gphoto/libgphoto2 #492: Live Network Preview on Canon Rebel T6

pufflegamerz, 2020-04-13:
I am trying to get my Canon Rebel T6 to work over my network. I used the
smartphone setting to connect to my PC, and it works mostly. I can take
pictures, but I can't use the live preview mode. It gives me: *Unable to
capture live preview: (null)*

I am doing this via Entangle, so here's some more info:

*Entangle version:* 2.0

*libgphoto2 version:* 2.5.22-3

*Camera model:* Canon EOS Rebel T6

I found this archive, but it didn't help me.
https://www.mail-archive.com/entangle-devel@gna.org/msg00100.html

I was redirected here by a Gitlab issue, here:
https://gitlab.com/entangle/entangle/-/issues/43

He pointed out some errors, then redirected me here.

Thanks.

pufflegamerz, 2020-04-13:
I tried running this command: `gphoto2 --debug --debug-logfile=my-logfile.txt --port ptpip:192.168.1.14 --capture-image` and got a lot of errors, I'll upload the log file if needed.

msmeissn, 2020-04-19:
I guess liveview works over USB?

A debuglogfile would be good.

pufflegamerz, 2020-04-19:
> I guess liveview works over USB?
Yes, it does.
> 
> A debuglogfile would be good.
Ok, I will get that soon.

pufflegamerz, 2020-04-30:
Really sorry for late reply, I have been busy.
http://petabyte.heb12.com/filedump/my-logfile.txt

pufflegamerz, 2020-04-30:
I tried another command, it looks to take longer to connect there.
`gphoto2 --debug --debug-logfile=capturepreview.txt --port ptpip:192.168.1.2 --capture-preview`
http://petabyte.heb12.com/filedump/capturepreview.txt

pufflegamerz, 2020-04-30:
And Entangle can connect to my camera, take pictures, and return them via a network connection.

msmeissn, 2020-05-03:
can you try if inserting an SD card helps?

pufflegamerz, 2020-05-03:
I have had one in. It won't do any connecting without one.

msmeissn, 2020-05-03:
Hmm. the camera sadly does not tell us why its busy :( We might not have drained enough events , but this is also unlikely.

Can you add an additional 
--wait-event=1s 
before the --capture-preview ?

pufflegamerz, 2020-05-03:
It worked a little bit more, but still has `*** Error (-1: 'Unspecified error') ***`.

`gphoto2 --debug --debug-logfile=capturepreview.txt --port ptpip:192.168.1.2 --wait-event=1s --capture-preview`
http://petabyte.heb12.com/filedump/capturepreview1.txt

I also tried it with 5 seconds.
`gphoto2 --debug --debug-logfile=capturepreview5.txt --port ptpip:192.168.1.2 --wait-event=5s --capture-preview`
http://petabyte.heb12.com/filedump/capturepreview5.txt