gphoto/libgphoto2 #493: Canon EOS M6 no BULB mode

SevenW, 2020-04-16:
I try to get to get my M6 to work in Kstars/Ekos/Indi using libgphoto2. No image capture takes place, and most likely related to bulb mode. from logs it is clear that Indi is communicating well with the camera, so no basic USB connectivity issues expected. I have explored with gphoto2 and found this:

gphoto2 --capture-image  -> success. Iamge taken and stored on SD card.
gphoto2 --capture-image --bulb=1  -> no capture takes place. error messages generated.

Steps taken to resolve:
Have build the latest released libgphoto2 and gphoto2 from source on Ubuntu 18.04
Updated M6 firmware from 1.0.0 to 1.0.1

**Name the camera**
Canon EOS M6

**libgphoto2 and gphoto2 version**
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.23.1       standard camlibs (SKIPPING lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

**To Reproduce**
gphoto2 --capture-image --bulb=1 
with the M6.

log file attached.
[M6-bulb.txt](https://github.com/gphoto/libgphoto2/files/4488175/M6-bulb.txt)

SevenW, 2020-04-17:
I have learned some more by searching on the error message "bulb not found in configuration tree".
My M6 kind of crashes when I try to set shutterspeed or aperture. After issuing the gphoto2 command I need to switch my camera OFF/ON twice before it recovers.
```
gphoto2 --set-config shutterspeed=2
```
debug logfile attached: 
[M6-set-shutterspeed.txt](https://github.com/gphoto/libgphoto2/files/4493862/M6-set-shutterspeed.txt)

However I am able to BULB capture with the following command:
```gphoto2 eosremoterelease="Press Full" --wait-event=1s --set-config eosremoterelease="Release Full" --wait-event-and-download=2s```

SevenW, 2020-04-19:
Digging a bit in the code, and logfiles I think I have found the issue. For EOS-M camera's the unprepare camera code makes a setting to the electronic view finder, which my camera does not have.

https://github.com/gphoto/libgphoto2/blob/72a75953d71a7b5dd6e444f83984c5570cadf7af/camlibs/ptp2/config.c#L565

I am going to rebuild the code with this setting commented out.

joelcollet, 2020-04-20:
Hi
i believe you have the very same issue that i have with my olympus and someone else with his sony alpha mirrorless. see here for a "solution" in indilib
https://www.indilib.org/forum/ccds-dslrs/5881-sony-alpha-5100-viewfinder-widget-error.html#52237
and here
https://github.com/gphoto/libgphoto2/issues/441

joël

SevenW, 2020-04-20:
After further testing and digging around in the ptp2 config.c code I conclude that the failure mode is in applying set-config settings to for example shutterspeed and iso.

I can capture an image, provided I make no other settings:
`gphoto2 --capture-image` succeeds
The earlier reported `gphoto2 eosremoterelease="Press Full" --wait-event=1s --set-config eosremoterelease="Release Full" --wait-event-and-download=2s` worked with 2.5.23.1, but not in 2.5.24

What fails is this:
```
gphoto2 --set-config shutterspeed=2 --capture-image
gphoto2 --set-config shutterspeed=2
```
The first scenario fails with a camera busy when `PTP_OC_CANON_EOS_RemoteReleaseOn` is applied.
```
2.147254 ptp_usb_sendreq             (2): Sending PTP_OC 0x9128 (PTP_OC_CANON_EOS_RemoteReleaseOn) (0x3,0x0) request...
2.147276 gp_port_write               (3): Writing 20 = 0x14 bytes to port...
2.147425 gp_port_write               (3): Wrote   20 = 0x14 bytes to port: (hexdump of 20 bytes)
0000  14 00 00 00 01 00 28 91-20 00 00 00 03 00 00 00  ......(. .......
0010  00 00 00 00            -                         ....            

2.147466 ptp_usb_getresp             (2): Reading PTP_OC 0x9128 (PTP_OC_CANON_EOS_RemoteReleaseOn) response...
2.147487 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
2.152150 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 19 20-20 00 00 00              .......  ...    

2.152265 ptp_usb_getresp [usb.c:514] (0): PTP_OC 0x9128 receiving resp failed: PTP Device Busy (0x2019)
2.152338 camera_trigger_canon_eos_capture [library.c:5173](0): 'ptp_canon_eos_remotereleaseon (params, 3, 0)' failed: Canon EOS M Full-Press failed (0x2019: PTP Device Busy)
2.152395 gp_context_error            (0): Canon EOS M Full-Press failed (0x2019: PTP: Toestel is bezig)
2.152528 camera_canon_eos_capture [library.c:3793](0): 'camera_trigger_canon_eos_capture (camera, context)' failed: 'Onbekende fout' (-110)
2.152583 gp_camera_capture [gphoto2-camera.c:1340](0): 'camera->functions->capture (camera, type, path, context)' failed: -110
2.154173 gp_camera_free              (2): Freeing camera...
2.154234 gp_camera_exit              (2): Exiting camera ('Canon EOS M6')...
```

The second scenario fails with some usb time out when `PTP_DPC_CANON_EOS_EVFOutputDevice` is set back to zero in the unprepare camera routine.
logging:
```
2.179640 ptp_usb_sendreq             (2): Sending PTP_OC 0x9110 (PTP_OC_CANON_EOS_SetDevicePropValueEx) request...
2.179665 gp_port_write               (3): Writing 12 = 0xc bytes to port...
2.179781 gp_port_write               (3): Wrote   12 = 0xc bytes to port: (hexdump of 12 bytes)
0000  0c 00 00 00 01 00 10 91-21 00 00 00              ........!...    

2.179812 ptp_usb_senddata            (2): Sending PTP_OC 0x9110 (PTP_OC_CANON_EOS_SetDevicePropValueEx) data...
2.179833 gp_port_write               (3): Writing 24 = 0x18 bytes to port...
2.179909 gp_port_write               (3): Wrote   24 = 0x18 bytes to port: (hexdump of 24 bytes)
0000  18 00 00 00 02 00 10 91-21 00 00 00 0c 00 00 00  ........!.......
0010  b0 d1 00 00 00 00 00 00-                         ........        

2.179971 ptp_usb_getresp             (2): Reading PTP_OC 0x9110 (PTP_OC_CANON_EOS_SetDevicePropValueEx) response...
2.179993 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
22.180780 gp_libusb1_read [libusb1.c:609](0): 'libusb_bulk_transfer (port->pl->dh, port->settings.usb.inep, (unsigned char*)bytes, size, &curread, port->timeout)' failed: Operation timed out (-7)
22.180967 gp_port_read [gphoto2-port.c:441](0): Reading 1024 = 0x400 bytes from port failed: Tijdsoverschrijding tijdens lezen van of schrijven naar de poort (-10)
22.181036 ptp_usb_getresp [usb.c:514] (0): PTP_OC 0x9110 receiving resp failed: PTP I/O Error (0x02ff)
22.181085 camera_unprepare_canon_eos_capture [config.c:565](0): 'ptp_canon_eos_setdevicepropvalue (params, PTP_DPC_CANON_EOS_EVFOutputDevice, &ct_val, PTP_DTC_UINT16)' failed: PTP I/O Error (0x02ff)
```

I have tried several things in the code, but was not able to figure out a way to set shutterspeed and capture an image.

msmeissn, 2020-05-22:
So the way it should is to set  --set-config shutterspeed=2    (the bulb value) and then using eosremoterelease like you did.

 gphoto2 --set-config eosremoterelease="Press Full" --wait-event=10s --set-config eosremoterelease="Release Full" --wait-event-and-download=2s

My M10 does not seem to do bulb after some tries sadly, so I cannot reproduce it right now.

the evfoutputdevice setting crash is weird, the camera is probably crashing at this point.
What you can try is to remove the setting it in BOTH functipons camera_prepare_canon_eos_capture  and also in camera_unprepare_canon_eos_capture

knro, 2020-09-29:
So just check if this issue is still the same or has it be addressed in recent libgphto2 updates? Do we need to do anything on the client side to make this work?

SevenW, 2020-10-08:
I did not further pursue this and got myself an Altair astro camera. This was only after I figured out that also Canon did not provide SDK support for this camera nor EOS tools that can do remote capture (at least controlling settings). Probably there is something fundamentally wrong in this camera firmware or USB hardware.

I propose to close this and consider the M6 as not supported.

dragonlost, 2021-08-01:
Hello I have exact same issue on Canon EOS M3 ( firmware 1.1, 1.2.0 and 1.2.1).
I can't change the setting and then take a photo.

The get-config command works fine and I can launch a capture if I don't change the parameter

msmeissn, 2021-08-04:
currently a known issue  without fix... also for M3 :/

Jul3k, 2021-09-17:
I can confirm this issue on a EOS M3 as well. On my mobile I have the android app "Camera Connect and Controll" installed. This app handles the M3 nicely. Is there a way do fix gphoto, by looking how this app handles the configuration of the EOS M series cameras?

msmeissn, 2021-09-21:
if you can get a sniffed traffic trace of this it would be nice.

but android app goes over wifi, right?

Jul3k, 2021-09-24:
@msmeissn Yes, the Android App connects over wifi. On Monday I can setup Wireshark to sniff the traffic of the interaction with the M3. Does Gphoto support the connection over wifi for the M Series? Maybe a comparison of the traffic between the Android App and Gphoto for setting the exposure, capturing, etc. would be useful? I do not know if there is a big difference between the USB protocol and the WiFi protocol.

Jul3k, 2021-09-27:
The problem that the shutterspeed and other properties cannot be set, is not present when the camera is connected over Wifi instead of USB. I connected in shell mode, as the camera is otherwise always exiting the remote control mode:
`gphoto2 --port ptpip:192.168.178.XX --shell`
A sniffed traffic from the Android App will probable not bring any new insights right? The App also claims that it can control the camera over USB, but I do not have a adapter to test and I would not know of a way to sniff the USB traffic.

When I tried to list the available config options, I did encounter a minor problem as Gphoto could not detect the camera:

    *** Error ***
    Could not detect any camera
    
    *** Error ***
    An error occurred in the io-library ('Unknown error'): No error description available
    *** Error (-105: 'Unknown model') ***

which could be resolved by explicitly stating the camera model:
`gphoto2 --camera "Canon EOS M3" --port ptpip:192.168.178.XX --list-all-config`
Apart from that, the Wifi control worked as expecte.
I attach two log files when the camera was connected over USB. For the first, the USB was plugged in before the camera was powered on:
[eos-m3-set-config-log.txt](https://github.com/gphoto/libgphoto2/files/7234108/eos-m3-set-config-log.txt)
For the second, the cable was plugged in after the camera was powered on. This crashes the camera to a point, where I have to remove the battery:
[eos-m3-set-config-camera-crash-log.txt](https://github.com/gphoto/libgphoto2/files/7234106/eos-m3-set-config-camera-crash-log.txt)