gphoto/libgphoto2 #495: Nikon cameras fail to capture image longer than 20 seconds

berrange, 2020-04-20:
**Describe the bug**

If the shutterspeed is set to a value greater than 20 seconds, capture will fail with a busy timeout.

In 2.5.23, it would fail for shutter speed longer than 5 seconds, but I've got the patch from https://github.com/gphoto/libgphoto2/issues/420 applied which makes it merely fail after 20 seconds instead.

The longest shutter speed on my D90 and D5100 is 30 seconds, beyond which you must use "Bulb". I'm not sure if any more modern Nikon cameras support longer than 30 second fixed exposures. If they do, then the timeout needs to be longer than whatever the maximum shutter speed is, plus some grace period.

With a 30 second max shutterspeed, I suggest a timeout of 40 seconds, to allow a solid grace period to eliminate any further false positive errors.

$ gphoto2 --auto-detect --shell 
Model                          Port                                            
----------------------------------------------------------
Nikon DSC D90 (PTP mode)       usb:003,010     
gphoto2: {/home/berrange/Pictures} /> get-config /main/capturesettings/shutterspeed
Label: Shutter Speed
Readonly: 0
Type: RADIO
Current: 30.0000s
Choice: 0 0.0002s
Choice: 1 0.0003s
Choice: 2 0.0004s
Choice: 3 0.0005s
Choice: 4 0.0006s
Choice: 5 0.0008s
Choice: 6 0.0010s
Choice: 7 0.0012s
Choice: 8 0.0015s
Choice: 9 0.0020s
Choice: 10 0.0025s
Choice: 11 0.0031s
Choice: 12 0.0040s
Choice: 13 0.0050s
Choice: 14 0.0062s
Choice: 15 0.0080s
Choice: 16 0.0100s
Choice: 17 0.0125s
Choice: 18 0.0166s
Choice: 19 0.0200s
Choice: 20 0.0250s
Choice: 21 0.0333s
Choice: 22 0.0400s
Choice: 23 0.0500s
Choice: 24 0.0666s
Choice: 25 0.0769s
Choice: 26 0.1000s
Choice: 27 0.1250s
Choice: 28 0.1666s
Choice: 29 0.2000s
Choice: 30 0.2500s
Choice: 31 0.3333s
Choice: 32 0.4000s
Choice: 33 0.5000s
Choice: 34 0.6250s
Choice: 35 0.7692s
Choice: 36 1.0000s
Choice: 37 1.3000s
Choice: 38 1.6000s
Choice: 39 2.0000s
Choice: 40 2.5000s
Choice: 41 3.0000s
Choice: 42 4.0000s
Choice: 43 5.0000s
Choice: 44 6.0000s
Choice: 45 8.0000s
Choice: 46 10.0000s
Choice: 47 13.0000s
Choice: 48 15.0000s
Choice: 49 20.0000s
Choice: 50 25.0000s
Choice: 51 30.0000s
Choice: 52 Bulb
END
gphoto2: {/home/berrange/Pictures} /> capture-image-and-download 

*** Error ***              
PTP Device Busy
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-110: 'I/O in progress') ***

**Name the camera**
Nikon DSC D90 (PTP mode)
Nikon DSC D5100 (PTP mode)

**libgphoto2 and gphoto2 version**
gphoto2         2.5.23         gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.23         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial lockdev locking

and I have applied the patch from #420 already. 

**To Reproduce**
Steps to reproduce the behavior:
Set shutterspeed to 30 seconds and capture an image.

When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.

berrange, 2020-04-20:
Looking at this https://www.nikon.co.uk/en_GB/product/digital-cameras/slr/professional/d780#tech_sp  ecs

```
1/8000 to 30 s (choose from step sizes of 1/3 and 1/2 EV, extendable to 900 s in mode M); Bulb; Time; X200
```

So to cope with that extended speed, the timeout when capturing has to be more like 910 seconds

msmeissn, 2020-04-21:
I tested with my D90 and it has no issues. Which settings to you use?
can you get a debug logfile?

berrange, 2020-04-21:
Ok, let me double check against latest git master instead of the previous stable release too, and then i'll get debug log if it still reproduces on git master

ColinSlack, 2020-04-22:
I have both a D610 and aD810, I shoot regularly at 25 and 30 second exposures without a problem.

berrange, 2020-04-22:
Ok, this was a mistake on my part. I found the fix which increased the timeout from 5 to 20 seconds #420  but failed to notice there was yet another commit after that for #465  which increased it again to 200 seconds.  The latter commit does indeed fix my problem so that's good and 2.5.24 is working correctly.   So sorry for the misleading bug report. 

I would note however, that since the D780 apparently has a max shutter speed of 900 seconds it might be wise to proactively extend the timeout yet again to 910 seconds in case someone is lucky enough to have that camera and tries to use libgphoto2 with it !

closed 2020-05-22 by msmeissn