gphoto/libgphoto2 #496: Fuji X-T3 no auto focus during movie capture

daniel-ge, 2020-04-28:
**Describe the bug**
When capturing a movie using a Fuji X-T3 and `gphoto2 --stdout --capture-movie` there is no auto-focus - and the video is blurred. However, capturing a single image with `gphoto2 --capture-image-and-download` works very well.

Maybe it is related to the line `Focus Point(0xd347): error 201b on query.` from the summary command (shown below).

**Name the camera**
```
$env LANG=C gphoto2 --auto-detect 

Model                          Port                                            
----------------------------------------------------------
Fuji Fujifilm X-T3             usb:001,011     
```
and the summary of the camera:

`$env LANG=C gphoto2 --debug --debug-logfile=summary-debug.log --summary`
<details><summary>show output from above</summary>
<p>

```
Camera summary:                                                                
Manufacturer: FUJIFILM
Model: X-T3
  Version: 3.21
  Serial Number: XXXXXXXXXXXXXXXXXXXXXXXXXXX
Vendor Extension ID: 0xe (1.0)
Vendor Extension Description: fujifilm.co.jp: 1.0; 

Capture Formats: Defined Type JPEG Unknown(b103)
Display Formats: Defined Type, JPEG, Unknown(b103)

Device Capabilities:
        File Download, File Deletion, No File Upload
        Generic Image Capture, Open Capture, No vendor specific capture

Storage Devices Summary:
store_10000001:
        StorageDescription: Still
        VolumeLabel: 1XXXXXXXXXXXXXXXXXXXXXXXXXXX
        Storage Type: Builtin RAM
        Filesystemtype: Generic Hierarchical
        Access Capability: Read Only with Object deletion
        Maximum Capability: 18446744073709551615 (17592186044415 MB)
        Free Space (Bytes): 18446744073709551615 (17592186044415 MB)
        Free Space (Images): -1
store_10000002:
        StorageDescription: Live
        VolumeLabel: 2XXXXXXXXXXXXXXXXXXXXXXXXXXX
        Storage Type: Builtin RAM
        Filesystemtype: Generic Hierarchical
        Access Capability: Read Only with Object deletion
        Maximum Capability: 18446744073709551615 (17592186044415 MB)
        Free Space (Bytes): 18446744073709551615 (17592186044415 MB)
        Free Space (Images): -1

Device Property Summary:
Battery Level(0x5001): error 200a on query.
Image Size(0x5003):(readwrite) (type=0xffff) Enumeration [
        '3120x2080',
        '3120x1760',
        '2080x2080',
        '4416x2944',
        '4416x2488',
        '2944x2944',
        '6240x4160',
        '6240x3512',
        '4160x4160',
        '4992x3328',
        '4992x2808',
        '3328x3328'
[... 198 lines cut ...]
```
$env LANG=C gphoto2 --debug \
       --debug-logfile=movie-capture-debug.log \
       --stdout --capture-movie=10 | \
 gst-launch-1.0 fdsrc fd=0 \
       ! decodebin name=dec \
       ! queue \
       ! videoconvert \
       ! xvimagesink

Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Capturing 10 preview frames as movie to 'stdout'.
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Movie capture finished (11 frames)
Got EOS from element "pipeline0".
Execution ended after 0:00:00.202260066
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```
[movie-capture-debug.log](https://github.com/gphoto/libgphoto2/files/4545437/movie-capture-debug.log)

msmeissn, 2020-05-24:
At this time we do not know how to trigger autofocus in liveview :/

daniel-ge, 2020-05-24:
Hi Marcus

Would it help having the captured wifi communication between camera and smartphone app? 

I am wondering whether it is possible to extract the related commands (given the fact that it is ptpip and without any encryption).

I could give it a try using the Wireshark ptpip dissector. 

Thanks,
Daniel

msmeissn, 2020-05-24:
That would be helpful, if you can capture it.

TormodLandet, 2020-05-29:
If you enable "Pre AF" in the menu (on the camera) it should focus