gphoto/libgphoto2 #497: Camera support request for Sanyo Xacti

Autom3, 2020-05-02:
Fill in fields below. If you don't know how, leave it free.

**Name of the camera**

Sanyo Xacti VPC-FH1

**USB IDs**

 0474:02e5

**camera summary output**

https://hastebin.com/raw/ihisoqekow

**camera configuration output**

https://hastebin.com/raw/ugoheganay

**test capture**

test if capture perhaps already works:

  gphoto2 --capture-image-and-download

  output:
```
*** Error ***              
Sorry, your camera does not support generic capture
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image-and-download

Please make sure there is sufficient quoting around the arguments.
```

  output from the debug command: https://hastebin.com/raw/rufirevaxi
 
  gphoto2 --capture-preview

  output:
```
*** Error (-6: 'Unsupported operation') ***                                    

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-preview

Please make sure there is sufficient quoting around the arguments.
```

One more point, when I run: **gphoto2 --auto-detect**, I get this output:
```
Model                          Port                                            
----------------------------------------------------------
USB PTP Class Camera           usb:001,009     
```

On my camera itself I got the option to connect to the computer using **M**TP

Thanks in advance.

hfiguiere, 2020-05-03:
It doesn't support generic PTP capture. If the camera is capable of doing capture then it is using a different protocol.

msmeissn, 2020-05-03:
the ids reported indicate its just MTP for downloading images. 

You can try 

gphoto2 --summary --debug --debug-logfile=summary-debug.log

and attach summary-debug.log here   so we can look at what the camera reports.

Autom3, 2020-05-03:
Here you go:

[summary-debug.log](https://github.com/gphoto/libgphoto2/files/4571625/summary-debug.log)

msmeissn, 2020-05-04:
i looked over it and it really only has the regular PTP and MTP capabilities, there is no "vendor" opcode or any capture related thing mentioned.

So the camera has no remnote control abilities... 

If there is Windows software that does remote control, we would like to know about it, but so far I am assuming the camera is not remote control capable.

closed 2020-05-04 by msmeissn