gphoto/libgphoto2 #499: Regression Canon PowerShot G9x mark II

luukvanderduim, 2020-05-05:
Hi!,

The command below allowed me to use the Canon PowerShot G9X mark II as a webcam using libgphoto2 2.5.23 (Ubuntu  Xfce 19.10), but no longer so using 2.5.24 (Ubuntu xfce 20.04):

`gphoto2 --stdout -I --set-config  viewfinder  --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p  -threads 0 -f v4l2 /dev/video0
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/4580246/my-logfile.txt)
`
Last week (end of april 2020) I tried libgphoto2 from git, but that resulted in the problem. (description below). I was on Ubuntu 19.10 at that time. That distributions libgphoto2 2.5.23 did result in a working 'webcam'. I have since upgraded to Ubuntu 20.04 but find the same result as with the git build I tried.

The lens does not open and no valid stream is presented to stdout, it appears.

ERROR: Movie capture error... Exiting.
Movie capture finished (0 frames)
*** Error (-1: 'Unspecified error') ***       

pipe:: Invalid data found when processing input

I have attached a debug file to help troubleshoot the issue.

If necessary, I can bisect the git but I might perhaps need some advice with that.

**Name the camera**
Canon PowerShot G9x Mark II

**libgphoto2 and gphoto2 version**
 gphoto2 --version
gphoto2 2.5.23

Copyright (c) 2000-2019 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.24         standard camlibs (SKIPPING lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

**To Reproduce**
see description above

Thanks for any help, much appreciated!

luukvanderduim, 2020-05-06:
Confirm that libgphoto2 git tag release-2.5.23 and gphoto 2.5.23.1 (current git) works.
I found that the simpler test is: ` $ gphoto2 --trigger-capture`
Will now try to find the last working commit.

luukvanderduim, 2020-05-06:
After bisecting 2.5.23 and current, the following commit appears to be the troubling commit:
```
50880658e4637aaf501a3323cdcdafeebac615c4 is the first bad commit
commit 50880658e4637aaf501a3323cdcdafeebac615c4
Author: Marcus Meissner <marcus@jet.franken.de>
Date:   Sun Oct 6 20:21:35 2019 +0200

    set / reset uilock around capture dest

 camlibs/ptp2/config.c  | 6 ++++--
 camlibs/ptp2/library.c | 2 ++
 2 files changed, 6 insertions(+), 2 deletions(-)
luuk@pineweazel:/extra/code/C/libgphoto2
```

msmeissn, 2020-05-06:
it might howevcer be this part and not the uilockign:

@@ -453,7 +455,7 @@ camera_prepare_canon_eos_capture(Camera *camera, GPContext *context) {
        C_PTP (ptp_check_eos_events (params));
        params->eos_captureenabled = 1;
 
-       if (is_canon_eos_m (params)) {
+       if (is_canon_eos_m (params) && 0) {
                PTPPropertyValue    ct_val;
 
                GP_LOG_D ("EOS M detected");

msmeissn, 2020-05-06:
this did not quote well:
`@@ -453,7 +455,7 @@ camera_prepare_canon_eos_capture(Camera *camera, GPContext *context) {
        C_PTP (ptp_check_eos_events (params));
        params->eos_captureenabled = 1;
 
-       if (is_canon_eos_m (params)) {
+       if (is_canon_eos_m (params) && 0) {
                PTPPropertyValue    ct_val;
 
                GP_LOG_D ("EOS M detected");
`

luukvanderduim, 2020-05-06:
I'll find that and test

luukvanderduim, 2020-05-06:
> It might be this part of the change:
> 
> `- if (is_canon_eos_m (params)) {`
> ` + if (is_canon_eos_m (params) && 0) {`

spot on! That indeed appears to be the culprit

msmeissn, 2020-05-06:
hmm. Some cameras need this block ,others do not :(

luukvanderduim, 2020-05-06:
`if (is_canon_eos_m (params) && strstr(params->deviceinfo.Model,"EOS")) {`
IIUC this excludes my camera model - but it needs the block?

msmeissn, 2020-05-10:
I am removing that check again.

I hope I remember why I added it, i failed to document it at that time :/

luukvanderduim, 2020-05-10:
Confirmed fix!
Thanks for all your work, I appreciate it.

closed 2020-05-10 by msmeissn