gphoto/libgphoto2 #501: Sony RX100VA stops working with libgphoto2 version 2.5.23, libgphoto2 version 2.5.22 is fine

R4C3R, 2020-05-06:
**Describe the bug**

**Name the camera**
root@ubuntu-test-pc:~# gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Sony DSC-RX100M5A (PC Control) usb:003,015     
 

**libgphoto2 and gphoto2 version**
root@ubuntu-test-pc:~# gphoto2 --version
gphoto2 2.5.23

Copyright (c) 2000-2019 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.24         standard camlibs (SKIPPING lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

**To Reproduce**
The behaviour seems to depend on the installation sequence of gphoto2. I started of with a fresh Ubuntu 18.04 install and installed the latest stable version of gphoto2 using the gphoto2-updater.sh, which was:

gphoto2         2.5.23        
libgphoto2      2.5.24   

The camera was properly detected, but issuing the command "gphoto2 --capture-image-and-download --force-overwrite" didn't do anything it seemed like, no sounds of picture taking nothing.

Then I downgraded the libgphoto2 version by altering the latest version in the gphoto2-updater.sh script for libgphoto2 to 2.5.23

gphoto2         2.5.23        
libgphoto2      2.5.23

The camera was properly detected again, but I didn't notice a change in behaviour, nothing happened.

Then I downgraded the libgphoto2 version by altering the latest version in the gphoto2-updater.sh script for libgphoto2 to 2.5.22

gphoto2         2.5.23        
libgphoto2      2.5.22

The camera was properly detected again and seemed to be working as expected. Taking pictures and downloading them. So the combination of gphoto2 2.5.23 and libgphoto2 2.5.22 seems to work fine.

After that I upgraded to libgphoto2 2.5.24 again. Now it seems to work sometimes but very unstable, sometimes pictures are taken and downloaded, BUT the process ends with an PTP i/o error

`*** Error ***              
PTP I/O Error
`
or

`*** Error ***              
PTP I/O Error
ERROR: Could not capture image.
ERROR: Could not capture.
`
or 

`*** Error ***              
An error occurred in the io-library ('Could not lock the device'): Camera is already in use.
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-60: 'Could not lock the device') ***  
`

and sometimes the process just hangs at the end, after deleting the last file

`root@ubuntu-test-pc:~# gphoto2 --capture-image-and-download                    
New file is in location /capt0000.jpg on the camera                            
Saving file as capt0000.jpg
Deleting file /capt0000.jpg on the camera
New file is in location /capt0001.arw on the camera                            
Saving file as capt0001.arw
Deleting file /capt0001.arw on the camera
New file is in location /capt0002.jpg on the camera                            
Saving file as capt0002.jpg
Deleting file /capt0002.jpg on the camera
New file is in location /capt0003.arw on the camera                            
Saving file as capt0003.arw
Deleting file /capt0003.arw on the camera
New file is in location /capt0004.jpg on the camera                            
Saving file as capt0004.jpg
Deleting file /capt0004.jpg on the camera
New file is in location /capt0005.arw on the camera                            
Saving file as capt0005.arw
Deleting file /capt0005.arw on the camera

`

When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.

I can upload a debug logfile if desired

msmeissn, 2020-05-10:
libgphoto2_port 0.12.0 iolibs: disk ptpip serial usb

this is the old libusb0, but you need libusb1 for sony cameras.

usb1 needs to be there:
libgphoto2_port 0.12.0         iolibs: disk ptpip serial >>>>>usb1<<<<<< usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial lockdev locking

Install libusb1-devel or similar package before building libgphoto2.

R4C3R, 2020-05-10:
Ok, thank you for your patience, I finally got the message...

I did the following:

`sudo apt install libusb-1.0-0-dev

wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/gphoto2-updater.sh && chmod +x gphoto2-updater.sh && sudo ./gphoto2-updater.sh

`
and I can confirm it works on Ubuntu 20.04

R4C3R, 2020-05-10:
The solution was to install libusb-1.0-0-dev and recompile gphoto2

```
gphoto2 --version
gphoto2 2.5.23

Copyright (c) 2000-2019 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.24         standard camlibs (SKIPPING lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

now it says **usb1** after **libgphoto2_port 0.12.0**

```
sudo apt install libusb-1.0-0-dev

```

and then recompiling gphoto2 did the trick

closed 2020-05-10 by R4C3R