gphoto/libgphoto2 #503: Canon camera fails to capture image with shutterspeed = 30

adverley, 2020-05-10:
**Describe the bug**

My Canon EOS 50D camera works perfectly when shooting pictures in AV or manual mode except when it is forced to use the slowest shutterspeed which is 30s. This occurs both in AV mode when it is forced to shoot with shutterspeed 30 or by setting it manually. There is no CF card in the camera, I shoot to RAM and download it immediately. I am using a Raspberry Pi 4 - 2GB. 

**Name the camera**
Canon EOS 50D                  usb:001,003     

**libgphoto2 and gphoto2 version**
gphoto2 2.5.23.1

Copyright (c) 2000-2020 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23.1       gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

**To Reproduce**
Steps to reproduce the behavior:

```
gphoto2 --set-config /main/capturesettings/shutterspeed=30
gphoto2 --capture-image-and-download
```

Output:
ERROR: Could not capture image.                                                
ERROR: Could not capture.
 
[Debug log file](https://drive.google.com/open?id=168KW4T2PlyJsvSKHxz5yN3D6S8T4VpJn).

Similar issues [here](https://github.com/gphoto/libgphoto2/issues/495) and [here](https://github.com/gphoto/gphoto2/issues/143).

adverley, 2020-05-14:
I am looking at the logs when shooting with shutterspeed 10s. You can find them [here](https://drive.google.com/open?id=1nXJXPon3TWU5xni81DKWXJInQJFTTq_1) (I removed the picture data). At line 5617, timestamp +-22s, a new object is found which is probably the image. This is +-10s after shooting the image. I guess this is due to the camera processing the image. Obviously it will take longer to process an image shot with shutterspeed 30s so maybe gphoto is not waiting long enough?  

On the original logs (with the 30s shutterspeed), an camera_unprepare_capture event happens after 63s (line 8612) so it looks related to a timeout. Maybe increasing the wait time just as in issue #495 might help out? 

I also tested the camera itself on manual mode with SS=30s, which works fine and processes the image within 6s.

adverley, 2020-05-14:
When testing with SS=25s (which works btw), the library finds the picture +-30s after the shutter closes again. The timeline then looks like this:
t=0 -> take picture
t=25 -> picture taken
t=55 -> picture found 

Can someone indicate whether it is useful to let the library wait longer for the camera to reply? And if this is the case, should I be hacking in [here](https://github.com/gphoto/libgphoto2/blob/master/camlibs/canon/canon.c#L1458)?

msmeissn, 2020-05-22:
what postprocessing does the camera do? does it usually take this long too?

Then we would need to calculate it in into delays...

msmeissn, 2020-05-22:
FWIW there currently is a max 60 second timeout, which I will rev to 90 seconds now.

msmeissn, 2020-05-24:
in 2.5.25 release, please reopen if not fixed.

closed 2020-05-24 by msmeissn