gphoto/libgphoto2 #504: Remote shutter gphoto

FredoTheGreat, 2020-05-11:
When using remote shutter on ekos/kstars/gphoto ccd the setting for the USB port /dev/ttyUSB0 does not stay saved so you have to re enter everytime.
Also if the session involves a short exposure first (focus) then the camera goes into PC mode and and subsequent long exposure with the remote shutter gets ignored because it doesn't let go of the camera.

Camera is a Nikon D7000
USB device is a kmtronics device on /dev/ttyUSB0
Latest version of gphoto etc

**To Reproduce**
Start ekos first input /dev/ttyUSB0 then connect device and run a long exposure then restart ekos settings are lost
As above but run focus first which involves short exposure then run long exposure relay on USB device fires but as camera is innocent mode its ignored

When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.

msmeissn, 2020-05-12:
gphoto is not handling the serial device remote shutter control, this is probably some of the other tools involved, either ekos or kstars.

FredoTheGreat, 2020-05-12:
Why does it hold onto the camera and not release it from PC mode when the image is done

hfiguiere, 2020-05-12:
>  USB device is a kmtronics device on /dev/ttyUSB0

As Marcus say this is not something gphoto manages.

FredoTheGreat, 2020-05-12:
It manages the camera though? The issue is not with the serial device it's with the dslr via the gphoto ccd driver

hfiguiere, 2020-05-12:
https://www.indilib.org/devices/cameras/gphoto.html

At the bottom they link where to report issues.

hfiguiere, 2020-05-12:
And

> the setting for the USB port /dev/ttyUSB0 does not stay saved

Definitely not gphoto related.

FredoTheGreat, 2020-05-12:
ok thank you for your help 
sorry i am in the wrong place :)