gphoto/libgphoto2 #505: Add support for Fujifilm X-T4

majkrzak, 2020-05-13:
**Name of the camera**

Fujifilm X-T4

**USB IDs**

Just photo mode: `Bus 005 Device 002: ID 04cb:02e6 Fuji Photo Film Co., Ltd USB PTP Camera`

Auto tethering mode: `Bus 005 Device 002: ID 04cb:02e6 Fuji Photo Film Co., Ltd USB PTP Camera`

(next sections accordingly)

**camera summary output**
```
Camera summary:                                                                
Manufacturer: FUJIFILM
Model: X-T4
  Version: 1.00
  Serial Number: 593536303034200405B7A330210FFD
Vendor Extension ID: 0x6 (1.0)
Vendor Extension Description: fujifilm.co.jp: 1.0; 

Capture Formats: 
Display Formats: JPEG, JFIF
Supported MTP Object Properties:
        JPEG/3801: dc01/StorageID dc02/ObjectFormat dc03/ProtectionStatus dc04/ObjectSize dc07/ObjectFileName dc08/DateCreated dc0b/ParentObject dc41/PersistantUniqueObjectIdentifier dc44/Name dc87/Width dc88/Height dcd3/ImageBitDepth
        JFIF/3808: dc01/StorageID dc02/ObjectFormat dc03/ProtectionStatus dc04/ObjectSize dc07/ObjectFileName dc08/DateCreated dc0b/ParentObject dc41/PersistantUniqueObjectIdentifier dc44/Name dc87/Width dc88/Height dcd3/ImageBitDepth

Device Capabilities:
        File Download, File Deletion, File Upload
        No Image Capture, No Open Capture, No vendor specific capture

Storage Devices Summary:
store_10000001:
        StorageDescription: External Memory
        VolumeLabel: 000000000000000000000000000000000000000000000000
        Storage Type: Removable RAM (memory card)
        Filesystemtype: Digital Camera Layout (DCIM)
        Access Capability: Read-Write
        Maximum Capability: 63864569856 (60906 MB)
        Free Space (Bytes): 63189286912 (60262 MB)
        Free Space (Images): -1

Device Property Summary:
Battery Level(0x5001):(read only) (type=0x2) Range [0 - 10, step 1] value: 10% (10)
Property 0xd303:(read only) (type=0x2) 1
Session Initiator Info(0xd406):(readwrite) (type=0xffff) ''
Perceived Device Type(0xd407):(read only) (type=0x6) 1
```
```
Camera summary:                                                                
Manufacturer: FUJIFILM
Model: X-T4
  Version: 1.00
  Serial Number: 593536303034200405B7A330210FFD
Vendor Extension ID: 0x6 (1.0)
Vendor Extension Description: fujifilm.co.jp: 1.0; 

Capture Formats: Defined Type JPEG Unknown(b103)
Display Formats: Defined Type, JPEG, Unknown(b103), TIFF
Supported MTP Object Properties:
        Defined Type/3800: dc01/StorageID dc02/ObjectFormat dc03/ProtectionStatus dc04/ObjectSize dc07/ObjectFileName dc08/DateCreated dc0b/ParentObject dc41/PersistantUniqueObjectIdentifier dc44/Name dc87/Width dc88/Height dcd3/ImageBitDepth
        JPEG/3801: dc01/StorageID dc02/ObjectFormat dc03/ProtectionStatus dc04/ObjectSize dc07/ObjectFileName dc08/DateCreated dc0b/ParentObject dc41/PersistantUniqueObjectIdentifier dc44/Name dc87/Width dc88/Height dcd3/ImageBitDepth
        Unknown(b103)/b103: dc01/StorageID dc02/ObjectFormat dc03/ProtectionStatus dc04/ObjectSize dc07/ObjectFileName dc08/DateCreated dc0b/ParentObject dc41/PersistantUniqueObjectIdentifier dc44/Name dc87/Width dc88/Height dcd3/ImageBitDepth
        TIFF/380d: dc01/StorageID dc02/ObjectFormat dc03/ProtectionStatus dc04/ObjectSize dc07/ObjectFileName dc08/DateCreated dc0b/ParentObject dc41/PersistantUniqueObjectIdentifier dc44/Name dc87/Width dc88/Height dcd3/ImageBitDepth

Device Capabilities:
        File Download, File Deletion, No File Upload
        Generic Image Capture, Open Capture, No vendor specific capture

[... 1679 lines cut ...]
For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image-and-download

Please make sure there is sufficient quoting around the arguments.
```
```
~ $ LANG=C gphoto2 --capture-preview
*** Error (-6: 'Unsupported operation') ***                                    

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-preview

Please make sure there is sufficient quoting around the arguments.
```
`gphoto2 --get-all-files` works

majkrzak, 2020-05-13:
If guided I might be able to help in the free time.

msmeissn, 2020-05-13:
can you capture a --debug trace?

   gphoto2 --capture-image-and-download --debug --debug-logfile=capture.log 

and atttach capture.log here?

majkrzak, 2020-05-13:
[capture.log](https://github.com/gphoto/libgphoto2/files/4622434/capture.log)

msmeissn, 2020-05-13:
can you also do it for --capture-preview ? 

gphoto2 --capture-preview --debug --debug-logfile=preview.log

majkrzak, 2020-05-13:
[preview.log](https://github.com/gphoto/libgphoto2/files/4622503/preview.log)

msmeissn, 2020-05-13:
i think we might need to have the ID listed in libgphoto2 to make some parts of the code active.

I just commited this to current libgphoto2 git if you want to try

elijahparker, 2020-05-13:
In my limited tests so far the X-T4 seems to work the same as the X-T3, so
once it's identified properly the existing code should fully support it.

majkrzak, 2020-05-13:
OK, I've managed to compile. [PKGBUILD i've used](https://github.com/gphoto/libgphoto2/files/4623508/PKGBUILD.txt)

Shot and download works partially:
```
~/D/pkgbuilds $ LANG=C gphoto2 --debug --debug-logfile=shot1.txt --capture-image-and-download 
New file is in location /store_10000001/DSCF0002.jpg on the camera             
File DSCF0002.jpg exists. Overwrite? [y|n] y
Saving file as DSCF0002.jpg
Deleting file /store_10000001/DSCF0002.jpg on the camera
New file is in location /store_10000001/DSCF0001.raf on the camera
File DSCF0001.raf exists. Overwrite? [y|n] y
Saving file as DSCF0001.raf
Deleting file /store_10000001/DSCF0001.raf on the camera
~/D/pkgbuilds $ LANG=C gphoto2 --debug --debug-logfile=shot2.txt --capture-image-and-download
                                                                               
*** Error ***              
PTP Access Denied
New file is in location /store_10000001/DSCF0004.jpg on the camera             
File DSCF0004.jpg exists. Overwrite? [y|n] y
Saving file as DSCF0004.jpg
Deleting file /store_10000001/DSCF0004.jpg on the camera
New file is in location /store_10000001/DSCF0003.raf on the camera

*** Error ***              
PTP Access Denied
ERROR: Could not get image.
```
Camera got stuck and can not be shut down without unplugging USB
[shot1.txt](https://github.com/gphoto/libgphoto2/files/4623544/shot1.txt)
[shot2.txt](https://github.com/gphoto/libgphoto2/files/4623545/shot2.txt)
(I've truncated the logs due to the github limit)

Capture preview works well

msmeissn, 2020-05-14:
Good to hear that it starts to work ! :) 

hmm, trying to download both jpg rand raw ... the jpg file is deleted and the raw file gets access problems... 

The camera might keep them linked, so perhaps deleting jpg will also delete the raw implicitly.

Does it help if you only do gphoto2 --capture-image  and download later on?

Does it help if you just capture 1 type (either jpg or raw)?

majkrzak, 2020-05-16:
Capturing only one type works well. But I just find out that downloading video over 4GiB don't. It is truncated (I just wasted some of nice sceenes :disappointed: ). It is a PTP protocol limitation or gphoto2 implementation issue?

msmeissn, 2020-05-19:
I had no success in every setting up a > 4GB download so I was not yet able to implement it ... 

It needs some special tricks, which we do not implement yet.

I can try again with the documentation only. :/

jacklenox, 2021-03-11:
Hello, just wondering if there has been any progress with this? I'm also a little confused because the [gphoto2 support page](http://gphoto.org/proj/libgphoto2/support.php) says it does support image capture with the Fujifilm X-T4. Is this incorrect?

msmeissn, 2021-03-12:
i actually had rented a xt-4 and it seems to work ok after some changes.
did you try 2.5.27?

jacklenox, 2021-03-12:
Oh, really? Hmmm. Yep, I'm up-to-date. Here's what I'm seeing in the terminal:

```
$ gphoto2 -v
gphoto2 2.5.27

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.27         gcc, popt(m), exif, cdk, no aa, jpeg, no readline
libgphoto2      2.5.27         standard camlibs (SKIPPING docupen lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

And:

```
$ gphoto2 --summary
Camera summary:                                                                
Manufacturer: FUJIFILM
Model: X-T4
  Version: 1.10
  Serial Number: 593536303034200925C2733021A681
Vendor Extension ID: 0xe (1.0)
Vendor Extension Description: fujifilm.co.jp: 1.0; 

Capture Formats: 
Display Formats: JPEG, JFIF

Device Capabilities:
	File Download, File Deletion, File Upload
	No Image Capture, No Open Capture, No vendor specific capture

Storage Devices Summary:
store_10000001:
	StorageDescription: External Memory
	VolumeLabel: 000000000000000000000000000000000000000000000000
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Digital Camera Layout (DCIM)
	Access Capability: Read-Write
	Maximum Capability: 127865454592 (121942 MB)
	Free Space (Bytes): 125542465536 (119726 MB)
	Free Space (Images): -1

Device Property Summary:
Battery Level(0x5001):(read only) (type=0x2) Range [0 - 10, step 1] value: 10% (10)
Property 0xd303:(read only) (type=0x2) 1
Property 0xd406:(readwrite) (type=0xffff) ''
Property 0xd407:(read only) (type=0x6) 1
Exposure Time(0x500d): error 200a on query.
F-Number(0x5007): error 200a on query.
LensZoomPosCaps(0xd38c): error 200a on query.
FocusPosition(0xd171): error 200a on query.
Property 0xd21c: error 200a on query.
Focus Point(0xd347): error 200a on query.
LensZoomPos(0xd170): error 200a on query.
BatteryLevel(0xd242): error 200a on query.
LiveViewImageSize(0xd174): error 200a on query.
VideoOutOnOff(0xd168): error 200a on query.
LiveViewImageQuality(0xd173): error 200a on query.
ForceMode(0xd230): error 200a on query.
Property 0xd16e: error 200a on query.
FocusArea1(0xd372): error 200a on query.
FaceDetectionMode(0xd020): error 200a on query.
RawCompression(0xd022): error 200a on query.
GrainEffect(0xd023): error 200a on query.
SetEyeAFMode(0xd024): error 200a on query.
FocusPoints(0xd025): error 200a on query.
MFAssistMode(0xd026): error 200a on query.
InterlockAEAFArea(0xd027): error 200a on query.
Shadowing(0xd029): error 200a on query.
CropMode(0xd16f): error 200a on query.
TNumber(0xd02f): error 200a on query.
FocusArea4(0xd395): error 200a on query.
HighLightTone(0xd320): error 200a on query.
ShadowTone(0xd321): error 200a on query.
LongExposureNR(0xd322): error 200a on query.
FullTimeManualFocus(0xd323): error 200a on query.
LensISSwitch(0xd346): error 200a on query.
InstantAFMode(0xd34a): error 200a on query.
PreAFMode(0xd34b): error 200a on query.
LMOMode(0xd34d): error 200a on query.
ISMode(0xd351): error 200a on query.
FocusCheckMode(0xd35e): error 200a on query.
LiveViewImageQuality(0xd173): error 200a on query.
FileNamePrefix1(0xd365): error 200a on query.
FileNamePrefix2(0xd366): error 200a on query.
FocusArea3(0xd374): error 200a on query.
TotalShotCount(0xd310): error 200a on query.
ExposurePreview(0xd359): error 200a on query.
FrameGuideGridInfo1(0xd375): error 200a on query.
FrameGuideGridInfo2(0xd376): error 200a on query.
CustomDispInfo(0xd36e): error 200a on query.
ViewMode1(0xd33f): error 200a on query.
CustomAutoPowerOff(0xd364): error 200a on query.
LockButtonMode(0xd34e): error 200a on query.
WideDynamicRange(0xd02e): error 200a on query.
LensNameAndSerial(0xd36d): error 200a on query.
LensUnknownData(0xd38a): error 200a on query.
BatteryInfo1(0xd36a): error 200a on query.
BatteryInfo2(0xd36b): error 200a on query.
FunctionLockCategory1(0xd36f): error 200a on query.
FunctionLockCategory2(0xd370): error 200a on query.
SensitivityFineTune1(0xd222): error 200a on query.
SensitivityFineTune2(0xd223): error 200a on query.
LensZoomPosCaps(0xd38c): error 200a on query.
LensFNumberList(0xd38d): error 200a on query.
LensFocalLengthList(0xd38e): error 200a on query.
Property 0xd17b: error 200a on query.
```

jacklenox, 2021-03-12:
Sorry! My mistake. I didn't realise I needed to update a setting on the camera. All working now!

msmeissn, 2021-03-12:
what was the setting for the record?

onemoz, 2021-04-20:
> what was the setting for the record?

There is a setting in the connection settings, connection mode, pick the "usb tethering auto"

closed 2021-04-22 by msmeissn

lfbmendes, 2021-04-23:
For information, Fujifilm had finally released the SDK for some cameras:

https://fujifilm-x.com/pt-br/global-news/2021/0422_3974408/

"TOKYO, April 22, 2021 – FUJIFILM Corporation (President: Kenji Sukeno) is pleased to announce the release of the FUJIFILM X Series and GFX System Digital Camera Control Software Development Kit (SDK), which will help software developers integrate the functionality of FUJIFILM X Series and GFX System cameras into any applications that are currently being developed. Available on April 22, 2021 the SDK will allow developers to initiate the automatic transfer of images, from a supported Fujifilm camera, to a computer using a Windows*1 or macOS*2 operating system. Remote access to the basic controls of any supported Fujifilm camera is also supported. "

msmeissn, 2021-04-24:
thx! it might helpo a bit, let see

riccardolardi, 2021-06-06:
I would be very interested in helping out if the release of Fuji SDK would help to better support Fuji cameras in gphoto. Not though whether the new SDK could be leveraged in regards of how gphoto works internally. Anyway I will play around with the SDK and my X-T4 and see what I can get out of it.