gphoto/libgphoto2 #506: Fuji X-T2 incorrect exposure when shutterpeed is set

bludhemn, 2020-05-19:
On a Fujifilm X-T2, there is a 100x factor difference between the shutter speed specified and image captured. 

````
dave@scorpii ~> gphoto2 --version
gphoto2 2.5.23

Copyright (c) 2000-2019 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.24         standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
````
````
dave@scorpii ~> gphoto2 --set-config-value shutterspeed=10 --capture-image-and-download --debug --debug-logfile x-t2.1.log
New file is in location /store_10000001/DSCF0001.jpg on the camera             
Saving file as DSCF0001.jpg
Deleting file /store_10000001/DSCF0001.jpg on the camera

dave@scorpii ~> exiv2 DSCF0001.jpg | grep "Exposure time"
Exposure time   : 1/10 s

dave@scorpii ~> gphoto2 --set-config-value shutterspeed=50 --capture-image-and-download --debug --debug-logfile x-t2.2.log
New file is in location /store_10000001/DSCF0002.jpg on the camera             
Saving file as DSCF0002.jpg
Deleting file /store_10000001/DSCF0002.jpg on the camera

dave@scorpii ~> exiv2 DSCF0002.jpg | grep "Exposure time"
Exposure time   : 1/2 s

dave@scorpii ~> gphoto2 --set-config-value shutterspeed=100 --capture-image-and-download --debug --debug-logfile x-t2.3.log
New file is in location /store_10000001/DSCF0003.jpg on the camera             
Saving file as DSCF0003.jpg
Deleting file /store_10000001/DSCF0003.jpg on the camera

dave@scorpii ~> exiv2 DSCF0003.jpg | grep "Exposure time"
Exposure time   : 1 s
````

bludhemn, 2020-05-19:
[x-t2.tar.gz](https://github.com/gphoto/libgphoto2/files/4648196/x-t2.tar.gz)

msmeissn, 2020-05-19:
can you also attach

gphoto2 --summary

gphoto2 --list-all-config 

output?

bludhemn, 2020-05-19:
[list-all-config.txt](https://github.com/gphoto/libgphoto2/files/4649071/list-all-config.txt)
[summary.txt](https://github.com/gphoto/libgphoto2/files/4649072/summary.txt)

Attached. Thank you.

msmeissn, 2020-05-22:
I added a shutterspeed table for the fuji xt-2.

Not sure it works, can you check current git ?

bludhemn, 2020-05-22:
Hi,
It exposes now at the default of what the bulb mode is set (1 sec), regardless of shutterspeed config.

````
dave@scorpii ~> gphoto2 --version
gphoto2 2.5.23.1

Copyright (c) 2000-2020 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23.1       gcc, popt(m), exif, no cdk, aa, jpeg, readline
libgphoto2      2.5.24.1       standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
````

````
dave@scorpii ~> env LANG=C gphoto2 --set-config-value shutterspeed=10 --capture-image-and-download --debug --debug-logfile x-t2.1.log
                                                                               
*** Error ***              
Failed to set new configuration value 10 for configuration entry shutterspeed.
New file is in location /store_10000001/DSCF0001.jpg on the camera             
Saving file as DSCF0001.jpg
Deleting file /store_10000001/DSCF0001.jpg on the camera

dave@scorpii ~> exiv2 DSCF0001.jpg | grep -i exposure
Exposure time   : 1 s
Exposure bias   : 0 EV
Exposure mode   : Manual

dave@scorpii ~> env LANG=C gphoto2 --set-config-value shutterspeed=100 --capture-image-and-download 
                                                                               
*** Error ***              
Failed to set new configuration value 100 for configuration entry shutterspeed.
New file is in location /store_10000001/DSCF0002.jpg on the camera             
Saving file as DSCF0002.jpg
Deleting file /store_10000001/DSCF0002.jpg on the camera

dave@scorpii ~> exiv2 DSCF0002.jpg | grep -i exposure
Exposure time   : 1 s
Exposure bias   : 0 EV
Exposure mode   : Manual
````

Logs attached including debug, list-all and summary. 

[X-T2.logs.tar.gz](https://github.com/gphoto/libgphoto2/files/4670660/X-T2.logs.tar.gz)

msmeissn, 2020-05-23:
can you alspo add what gphoto2 --get-config shutterspeed   reports now?

msmeissn, 2020-05-23:
(Also the value might be named differently now )

bludhemn, 2020-05-23:
````
❯ gphoto2 --get-config shutterspeed
Label: Shutter Speed
Readonly: 0
Type: RADIO
Current: 1s
Choice: 0 1/32000
Choice: 1 1/25000
Choice: 2 1/20000
Choice: 3 1/16000
Choice: 4 1/13000
Choice: 5 1/10000
Choice: 6 1/8000
Choice: 7 1/6400
Choice: 8 1/5000
Choice: 9 1/4000
Choice: 10 1/3200
Choice: 11 1/2500
Choice: 12 1/2000
Choice: 13 1/1600
Choice: 14 1/1250
Choice: 15 1/1000
Choice: 16 1/800
Choice: 17 1/640
Choice: 18 1/500
Choice: 19 1/400
Choice: 20 1/320
Choice: 21 1/250
Choice: 22 1/200
Choice: 23 1/160
Choice: 24 1/125
Choice: 25 1/100
Choice: 26 1/80
Choice: 27 1/60
Choice: 28 1/50
Choice: 29 1/40
Choice: 30 1/30
Choice: 31 1/25
Choice: 32 1/20
Choice: 33 1/15
Choice: 34 1/13
Choice: 35 1/10
Choice: 36 1/8
Choice: 37 1/6
Choice: 38 1/5
Choice: 39 1/4
Choice: 40 1/3
Choice: 41 0.4s
Choice: 42 1/2
Choice: 43 0.6s
Choice: 44 0.8s
Choice: 45 1s
Choice: 46 1.3s
Choice: 47 1.6s
Choice: 48 2s
Choice: 49 2.5s
Choice: 50 3s
Choice: 51 4s
Choice: 52 5s
Choice: 53 6s
Choice: 54 8s
Choice: 55 10s
Choice: 56 13s
Choice: 57 15s
Choice: 58 20s
Choice: 59 25s
Choice: 60 30s
Choice: 61 40s
Choice: 62 50s
Choice: 63 60s
Choice: 64 2m
Choice: 65 4m
Choice: 66 8m
Choice: 67 15m
END
````

bludhemn, 2020-05-23:
Interesting note. X-T2 (along with other Fuji models) have a Time (T) mode selection, that allows the exposure time to be set by rotating the camera dials.
[http://fujifilm-dsc.com/en/manual/x-t20/taking_photo/shooting_mode/index.html](url)
If the X-T2 is set in this mode, it appears to acknowledge the shutterspeed change as the choice value (not seconds). 

````
~
❯ gphoto2 --set-config shutterspeed=60

~
❯ gphoto2 --get-config shutterspeed | grep  'Choice: 60'
Choice: 60 30s

~
❯ gphoto2 --get-config shutterspeed | grep  'Current'
Current: 30s
````

msmeissn, 2020-05-23:
and gphoto2 --set-config shutterspeed=10s     does not work/stick?

bludhemn, 2020-05-24:
Ok, I didn't realize that the exposure time needs the 's' at the end. I tried it now with the dial set to **T** and it works ok.  It doesn't work when the dial is set to Bulb mode though. 

Please see attached for all exposures and what the camera sets it to when set to **T.** I have been watching the camera interface and it assigns it correctly, expect for the 6s. Fuji has 5s, 6.5s, 8s (no 6s). 

[results.txt](https://github.com/gphoto/libgphoto2/files/4672968/results.txt)

Would it be possible to have this work on the Bulb mode, as I am not sure if all Fuji cameras have the T(Time) option?

Thanks

SupernovaF1, 2020-06-15:
Hello
I am experience exactly the same issue on the XT3. I thought at first it would be a problem with the EKOS software I was using.
https://indilib.org/forum/ccds-dslrs/7161-exposure-time-and-iso-issues-with-fuji-xt3.html

Unfortunately, this software requires bulb mode to make it work. Have you find by any chance a workaround? 
Many thanks

SupernovaF1, 2020-06-15:
Similar thread, with a bit more information in case it helps  https://github.com/gphoto/gphoto2/issues/340

riccardolardi, 2020-09-20:
Any update on this? I was planning to buy an X-T1 for imaging with Ekos.

SupernovaF1, 2020-09-20:
I gave up and bought a Canon EOS 450d and a cooled CCD. Much happier

sperho, 2020-09-24:
Any update on whether or not exposures are now working properly with the X-series cameras?

msmeissn, 2020-09-27:
So shutterspeed seems solved I think. I tried to reproduce it on x-t3, but it worked there.

For bulb mode I think I need to write special code emitting different opcodes. I reverse engineered parts of the driver, and it seems to indicate what to do ... but that was not successful.

Do you know a free for download app that does Fuji Bulb capture on windows?

sperho, 2020-09-27:
> Do you know a free for download app that does Fuji Bulb capture on windows?

I do not know of one (unfortunately, I am not a programmer, so I am not savvy in these things).  I've bounced this off of Jasem via Stellarmate support and he connected to me and my setup via Teamviewer and we did not see a 'bulb widget' exposed by the camera.  While true bulb mode would be fantastic, even being able to access all of the shutterspeed presets built into the X-T3 and X-T4 (which beyond 60 seconds go 2m, 4m, 8m, and 15m), would be a huge leap forward.  Thank you for working on this!  You might reach out to Jasem - I just uploaded a bunch of log files of various failures using this command (that he told me to use):

gphoto2 --debug --debug-logfile=gphoto.txt --set-config shutterspeed=2m --capture-image-and-download

msmeissn, 2020-09-28:
I was just working on the Fuji XT "bulb" mode for libgphoto2, thats why is there is no such widget yet ;)

msmeissn, 2020-09-28:
the shutterspeed things above will work in "T" mode I think.

sperho, 2020-09-29:
> the shutterspeed things above will work in "T" mode I think.

I would like to test this in Ekos with the X-T3 and X-T4.  Jasem pushed an update to the StellarMate beta channel, but the revision numbers of gphoto2 did not change on my device after I updated.  Were any revision numbers incremented for the recent Fuji changes to libgphoto2?  My current revisions are:

gphoto2     2.5.20
libgphoto2 2.5.25.1
libgphoto2_port   0.12.0

msmeissn, 2020-09-30:
I did not increment the 2.5.25.1 version. If it was rebuilt you probalby have it.

sperho, 2020-09-30:
> I did not increment the 2.5.25.1 version. If it was rebuilt you probalby have it.

Thanks... It was rebuilt and I have tested it with my X-T3 in both T and B mode and I'm seeing intermittent PTP Device Busy errors and failures to capture/download.

sperho, 2020-10-01:
I'm going to test the most recent commit starting with the X-T3 to determine if the shutterspeed presets work correctly.  If everything works as expected, I will update.  If there are errors, I will post the logs.  Before I start testing, should I have the camera on B mode or T mode?  Any other details that I should be aware of before I test?

sperho, 2020-10-03:
I'm attaching a text file with a series of 4 commands that I executed with debugging enabled.  I thought I had the correct shutterspeed choice to expose for 1 second, but maybe not as the first two commands, which executed fully, exposed for 5 seconds.  The second two commands did not result in exposures, but failed due to a PTP Device busy error.  I'll post the last two log files and attempt to add the other two, but the are 100+MB in size, so they may not go.  If not, I can try to snip out the image data and try again.
[XT3_gphoto_4.txt](https://github.com/gphoto/libgphoto2/files/5322231/XT3_gphoto_4.txt)
[XT3_gphoto_3.txt](https://github.com/gphoto/libgphoto2/files/5322232/XT3_gphoto_3.txt)
[gphoto2 debug commands.txt](https://github.com/gphoto/libgphoto2/files/5322233/gphoto2.debug.commands.txt)

msmeissn, 2020-10-04:
Choice 45 is 4 seconds according to the logfile  XT3_gphoto_1_excerpt.txt:

0.467221 _put_Genericu32Table        (2): FOUND right value for 4s in the enumeration at val 4000000

I would advise to use the value settter directly e.g. --set-config-value shutterspeed=1s  

I am observing the camera reports "Busy" if it is in some form of powersave mode. If you retry the same command right afterwards it will work. This is something I can see if I can fix it.

The longer exposures need also an additional --wait-event-and-download=FILEADDED or --wait-event-and-download=150s,
as the main capture currently has a timeout of 35 seconds. Same , I wil see if I get an idea here :/

sperho, 2020-10-04:
Thank you Marcus - there are going to be a whole lot of happy Fuji users in the astrophotography community if you can figure this out... :)

msmeissn, 2020-10-04:
I adjusted the "longer exposures" mode for capture-image-and-download, it can now wait until 2 hours 

(after some time the X-t3 seems to take capture time twice, once for real image, once for black reference(?))

FWIw, I also added real bulb capture too now. Currently it autofocuses before starting bulb though, not sure if i can get rid of that.

sperho, 2020-10-04:
Thanks - I will try to get the new build installed and tested soon. 

The "capture twice" thing is an option in the X series firmware that the user can control. Yes, it is for in-camera dark noise/hot pixel/etc subtraction. Most astrophotographers turn that off so they can do dark frame subtraction manually in post processing. We put a lens cap on at the end of an imaging session and take several frames at the same camera settings and  the post processing software uses those for correcting the stacked set of light images. So, if that can be disabled or set as an option, that would be good.  

Regarding the autofocusing in bulb mode, there is a physical switch on the front of the camera that will turn off autofocus and switched the camera to manual focus. I wonder if that would be the way for the camera to know not to autofocus before releasing the shutter.

msmeissn, 2020-10-04:
I revved the dev version to 2.5.25.2 as I did some work on fuji this weekend.

(I rented again the fuji xt-3 over the weekend)

The camera has some action transitions (one for capture, focusing, bulb, etc) ... these are very hard to reverse engineer without traces and I am happy I got the bulb one ;)
So there might be a bulb capture without focusing, i just do not know the transition codes yet.

riccardolardi, 2020-10-05:
@msmeissn would you be willing to write a really short key word like "crash course" on your process? How do you go about working this out? I'm handy in Linux and have a programming background but have never done reverse engineering of cameras but would love to dig into this.

sperho, 2020-10-05:
@msmeissn I've updated to the the *.2 build and tested bulb mode.  Great progress!  See the attached log, especially entries 5-11.  All commands resulted in a successful exposure and successful download.  Like I suspected, autofocus can be prevented by putting the camera in M focus mode with the physical switch on the camera, so avoiding an autofocus if desired is feasible in this way.

Some exposure strangeness, though.  In runs 5 and 6, I set the exposures for 17 (I meant to type "17s") and 17s, respectively and you'll see that the actual exposures were 1/160th (which I see that "17" lined up with in some lookup table for 1/160), but 17s did not.  What values should I be using for sub-minute long exposures?  I also notice that bulb mode as implemented is really a T-mode, where it forces to a preset time vs an arbitrary user input.  That's ok for my application, but others might be aware.  Log file 8 is just an example of where everything worked as intended.

[gphoto2 debug commands.txt](https://github.com/gphoto/libgphoto2/files/5327868/gphoto2.debug.commands.txt)
[XT3_gphoto_5.txt](https://github.com/gphoto/libgphoto2/files/5327869/XT3_gphoto_5.txt)
[XT3_gphoto_6.txt](https://github.com/gphoto/libgphoto2/files/5327866/XT3_gphoto_6.txt)
[XT3_gphoto_8.txt](https://github.com/gphoto/libgphoto2/files/5327867/XT3_gphoto_8.txt)

msmeissn, 2020-10-11:
But there is no 17s in the shutterspeed list right? The camera likely only accepts values in that list.

I am not sure what you mean with bulb mode as T mode?  the example uses a wait-event with timed delay, but other applications could depend on user input when to do the set-config bulb=0 
With the gphoto2 commandline tool I only added this in git for now, where you can do --wait-event  and send a SIGUSR2 signal to gphoto2 to stop the waiting.

sperho, 2020-10-11:
Correct,, there is no 17s in the list.  I was thinking we were working toward a true bulb mode, which the time the shutter is open is completely arbitrary, as it would be in any camera's bulb mode.  However, if we are working from the shutter speed list that is preset in the camera, that is not bulb mode for Fuji, that is T mode.  The camera is set to "T" using the physical shutter speed control dial and then the entire preset table of exposure values that you have programmatic access to can be selected by use of a physical command dial.  All that said, I will stop thinking of this as "bulb" mode and just "use the preset exposure table" mode, which is completely fine.  Just a misunderstanding on my part.  I'm away from my equipment for awhile, but I will make another rounds of tests when I return with only values from the table.

I don't completely understand this:  "With the gphoto2 commandline tool I only added this in git for now, where you can do --wait-event and send a SIGUSR2 signal to gphoto2 to stop the waiting."

msmeissn, 2020-10-11:
just to be clear.

if you do --capture-image-and-download  you will need to chose from one of the fixed list of shutterspeeds.

if you do 
--set-config bulb=1 --wait-event=Xs --set-config bulb=0 --wait-event-and-download=2Xs 
X is an arbitrary number of seconds.

sperho, 2020-10-11:
Ah! Thank you! I understand and I'll give this a try when I return...

msmeissn, 2020-10-11:
the 2xs is just twice the time to cover time of the "dark" picture time I mentioned earlier. 
 or use --wait-event-and-download=FILEADDED   that will wait until the new file appears.

sperho, 2020-10-11:
Ok, if I disable the dark frame option in the camera settings, I should be able to use maybe 180 and 182 (2 second delay between exposure end and then download? Or just use the FILE ADDED option... Would the expected location for gphoto2 to be looking on the camera's SD card? (And then it'll download that file when it sees it?)

msmeissn, 2020-10-11:
if you disable the dark picture mode, it will only take some seconds for the second wait I would say. But probably try FILEADDED event to be safe

sperho, 2020-10-11:
Thanks for the clarification - I have some testing to do now!

sperho, 2020-10-11:
So, for 3 minute exposure, is this the correct command to try?

gphoto2 --set-config bulb=1 --wait-event=180s --set-config bulb=0 --wait-event-and-download=FILEADDED

msmeissn, 2020-10-11:
yes, this would be a 3 minute exposure and download image afterwards

tvaucher, 2020-10-11:
Hello,

Jumping on the bandwagon! I set up everything on a MacBook (closest thing that runs UNIX-like stuff atm) and it works great on my X-T2. I was about to comment on the value of `--wait-event-and-download` as I got it to work with a various combination of value (like `1` or `1s`) but `FILEADDED` seems to be a good compromise as I deactivated the in camera noise reduction for long exposure (the second "dark" exposure you mentioned).

Running this works as expected (bulb, set iso and change downloaded filename) and matches what I'd like for an astro setup
```bash
env LANG=C gphoto2 --debug --debug-logfile=x-t2-bulb-180s-iso800.log --filename img/fuji_%Y-%m-%d_%H:%M:%S.%C --set-config iso=800 --set-config bulb=1 --wait-event=180s --set-config bulb=0 --wait-event-and-download=FILEADDED
```
Now just need to download and check in Ekkos. Ngl, if it works, this would really be a breakthrough for Fuji cameras in the astro world. Thanks a lot for the hard work 👍 .

PS: just noticed that `%C` returns the file extension in lowercase, according to the doc setting `%^C` should output it uppercase but it seems to be not yet supported, right?

sperho, 2020-10-15:
I am very likely doing something incorrect, but when I issue this command:
`gphoto2 --debug --debug-logfile=x-t3-bulb-14s-iso800.log  --set-config bulb=1 --wait-event=14s --set-config bulb=0 --wait-event-and-download=FILEADDED`

the camera exposes for exactly 14s, but the file does not transfer and PTP I/O errors are thrown.  Log attached.

[x-t3-bulb-14s-iso800.log](https://github.com/gphoto/libgphoto2/files/5381928/x-t3-bulb-14s-iso800.log)

sperho, 2020-10-15:
I've also tested the INDI drivers with libgphoto2.  We're getting very close!  Using bulb mode (with force bulb mode = ON in the driver settings), I was able to expose for arbitrary durations.  I set up a sequence of 5 x 10s, 5 x 17s, 5 x 120s, and 5 x 300s exposures.  The first three durations ran flawlessly.  When it is was time for the 300s exposures to run, there were issues. Unfortunately, the log file was not capture for that series.  

I performed another similar sequence, and it failed heading into the 120s exposures due to 'downloading from camera' I/O issues and then subsequent exposure failures.  Interestingly, the file did in fact download to the directory, but Ekos/libgphoto2 didn't seem to see it.  The Ekos debug log file for this run is attached.

[log_14-53-51.txt](https://github.com/gphoto/libgphoto2/files/5387839/log_14-53-51.txt)

msmeissn, 2020-10-18:
the x-t3-bulb-14s-iso800.log ... communication crashed during download of the raw image for unknown reasons. :/
The log_14-53-51.txt is not detailed enough, but it might have run into a similar issue.

msmeissn, 2020-10-18:
as for the %^C question, currently it takes the same case as the driver reports it, which is probably lowercase.
We did not implement ^ yet.

sperho, 2020-10-18:
I don't think that there is a way to capture a lower level log file in
Ekos, buy just let me know if there is anything you'd like me to test...

riccardolardi, 2020-11-01:
Just to ask, did anyone try with an X-T1? Do all models of the family (X-T1/T2/T3/T4) share the same interface? I'm thinking of upgrading from my X-T10 to an X-T1 but only if it has potential of working with gphoto

sperho, 2020-11-01:
Still some communication issues.  I'm attaching an Ekos log in case the error thrown is useful.  A sequence of 25 x 13s exposures went well.  Then, the next sequence was to perform a series of 120s exposures. This did not go well.  This has been shared with the Ekos maintainer, too.
[log_12-04-54.txt](https://github.com/gphoto/libgphoto2/files/5471435/log_12-04-54.txt)