gphoto/libgphoto2 #509: Add support for Fuji X-T30

billndotnet, 2020-05-22:
**Name of the camera*
Fujifilm X-T30

**USB IDs**
Bus 004 Device 006: ID 04cb:02e3 Fuji Photo Film Co., Ltd

**camera summary output**
[summary.txt](https://github.com/gphoto/libgphoto2/files/4666208/summary.txt)

**camera configuration output**
[list-all-config.txt](https://github.com/gphoto/libgphoto2/files/4666207/list-all-config.txt)

**test capture**
*** Error (-1: 'Unspecified error') ***

msmeissn, 2020-05-22:
can you try to capture debug output of failed captguring?

gphoto2 --debug --debug-logfile=capture.log --capture-image

billndotnet, 2020-05-22:
[capture.log](https://github.com/gphoto/libgphoto2/files/4669602/capture.log)

Attached!

msmeissn, 2020-05-23:
The pc control mode transition fails ... 

I spent some time looking over the fuji site and some related pages, it seems the Xt-30 does not allow tethering. :/

https://www.dpreview.com/forums/thread/4451199

billndotnet, 2020-05-25:
I'm gonna harangue Fujifilm about it.

https://twitter.com/billndotnet/status/1265006151856304130

DanielSmith1239, 2022-04-28:
Hello, I believe there was a firmware update that allows for some tethering. Can support be added?

wormyrocks, 2023-08-25:
Playing with libgphoto2 on an X-T30 (first gen) with the latest 2.01 firmware update, which seems to have added several tethered capturing features for the fuji mobile app.
This is captured on an M1 macbook with the Fujifilm X Webcam 2 drivers active, which causes it to show up as a webcam (but only in Zoom and Google Meet, which is infuriating.)
The camera needs to be set to "X Webcam" mode.
On a Mac, after a while I had to run commands with `sudo` to get it to work, I assume it's conflicting with some low-level macOS driver that kicked in after a few minutes. Nonetheless, quite a few features seem to work, but mostly work partially.

[list-all-config-x-webcam.log](https://github.com/gphoto/libgphoto2/files/12441216/list-all-config-x-webcam.log)

Output of `gphoto2 -a`:
```
Abilities for camera             : Fuji Fujifilm X-T30
Serial port support              : no
USB support                      : yes
Capture choices                  :
                                 : Preview
Configuration support            : no
Delete selected files on camera  : yes
Delete all files on camera       : no
File preview (thumbnail) support : yes
File upload support              : yes
```

Output of `gphoto2 --list-all-config`:
```
/main/actions/bulb
Label: Bulb Mode
Readonly: 0
Type: TOGGLE
Current: 2
END
/main/actions/autofocusdrive
Label: Drive Fuji Autofocus
Readonly: 0
Type: TOGGLE
Current: 2
END
/main/actions/movie
Label: Movie Capture
Readonly: 0
Type: TOGGLE
Current: 2
END
/main/actions/opcode
Label: PTP Opcode
Readonly: 0
Type: TEXT
Current: 0x1001,0xparam1,0xparam2
END
/main/settings/devicename
Label: Device Name
Readonly: 0
Type: TEXT
Current: 
END
/main/settings/prioritymode
Label: Priority Mode
Readonly: 0
Type: RADIO
Current: Camera
Choice: 0 Camera
Choice: 1 USB
END
/main/status/serialnumber
Label: Serial Number
Readonly: 1
Type: TEXT
Current: 593535353332201118C5D020111DA2
END
/main/status/manufacturer
Label: Camera Manufacturer
Readonly: 1
Type: TEXT
Current: FUJIFILM
END
/main/status/cameramodel
Label: Camera Model
Readonly: 1
Type: TEXT
Current: X-T30
END
/main/status/deviceversion
Label: Device Version
Readonly: 1
Type: TEXT
Current: 2.01
END
/main/status/vendorextension
Label: Vendor Extension
Readonly: 1
Type: TEXT
Current: fujifilm.co.jp: 1.0; 
END
/main/capturesettings/capturemode
Label: Still Capture Mode
Readonly: 0
Type: RADIO
Current: Self-timer
Choice: 0 Self-timer
Choice: 1 Unknown value 0008
END
/main/capturesettings/liveviewsize
Label: Live View Size
Readonly: 0
Type: RADIO
Current: XGA
Choice: 0 XGA
Choice: 1 VGA
Choice: 2 QVGA
END
/main/other/d173
Label: LiveViewImageQuality
Readonly: 0
Type: MENU
Current: 1
Choice: 0 1
Choice: 1 3
END
/main/other/d174
Label: LiveViewImageSize
Readonly: 0
Type: MENU
Current: 1
Choice: 0 1
Choice: 1 2
Choice: 2 3
END
/main/other/d201
Label: ReleaseMode
Readonly: 0
Type: MENU
Current: 4
Choice: 0 4
Choice: 1 8
END
/main/other/d207
Label: PriorityMode
Readonly: 0
Type: MENU
Current: 1
Choice: 0 1
Choice: 1 2
END
/main/other/d20b
Label: DeviceName
Readonly: 0
Type: TEXT
Current: 
END
/main/other/d20e
Label: FreeSDRAMImages
Readonly: 1
Type: MENU
Current: 15
Choice: 0 0
Choice: 1 1
END
/main/other/d21c
Label: PTP Property 0xd21c
Readonly: 0
Type: MENU
Current: 3
Choice: 0 0
Choice: 1 1
Choice: 2 2
Choice: 3 3
Choice: 4 4
Choice: 5 5
Choice: 6 6
Choice: 7 7
Choice: 8 8
Choice: 9 9
Choice: 10 10
Choice: 11 11
END
/main/other/d230
Label: ForceMode
Readonly: 0
Type: MENU
Current: 1
Choice: 0 1
END
/main/other/d36a
Label: BatteryInfo1
Readonly: 1
Type: RANGE
Current: 12
Bottom: 0
Top: 1.67772e+07
Step: 1
END
/main/other/d36b
Label: BatteryInfo2
Readonly: 1
Type: TEXT
Current: 100,0,0
END
/main/other/d21c
Label: PTP Property 0xd21c
Readonly: 0
Type: MENU
Current: 3
Choice: 0 0
Choice: 1 1
Choice: 2 2
Choice: 3 3
Choice: 4 4
Choice: 5 5
Choice: 6 6
Choice: 7 7
Choice: 8 8
Choice: 9 9
Choice: 10 10
Choice: 11 11
END
/main/other/d174
Label: LiveViewImageSize
Readonly: 0
Type: MENU
Current: 1
Choice: 0 1
Choice: 1 2
Choice: 2 3
END
/main/other/d173
Label: LiveViewImageQuality
Readonly: 0
Type: MENU
Current: 1
Choice: 0 1
Choice: 1 3
END
/main/other/d230
Label: ForceMode
Readonly: 0
Type: MENU
Current: 1
Choice: 0 1
END
/main/other/d173
Label: LiveViewImageQuality
Readonly: 0
Type: MENU
Current: 1
Choice: 0 1
Choice: 1 3
END
/main/other/d36a
Label: BatteryInfo1
Readonly: 1
Type: RANGE
Current: 12
Bottom: 0
Top: 1.67772e+07
Step: 1
END
/main/other/d36b
Label: BatteryInfo2
Readonly: 1
Type: TEXT
Current: 100,0,0
END
```

`gphoto2 --capture-preview` works, sometimes.
`capture-tethered` works, surprisingly.
`trigger-capture` gives an error, here's the log file:
[xt30trigger-capture.log](https://github.com/gphoto/libgphoto2/files/12441317/xt30trigger-capture.log)

`capture-image-and-download` works if I manually press the shutter after triggering.

pinpox, 2024-12-19:
@wormyrocks I'm trying to do the same thing on Linux, but it fails for me:
```
❯ gphoto2 -a
                                                              
*** Error ***              
Could not detect any camera
*** Error (-105: 'Unknown model') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt -a

Please make sure there is sufficient quoting around the arguments.
```

Does anyone know how to get it to work? I have the X-T30 camera, latest firmware and set to "X-Webcam"