gphoto/libgphoto2 #510: Canon EOS 750D sound capture : unsupported operation

azmeuk, 2020-05-24:
**Describe the bug**

I would like to use my Canon EOS 750D as a webcam. Video capture works like a charm (and I would like to thank you for that) but I cannot achieve to capture the sound from my DSLR

**Name the camera**
Canon EOS 750D

**libgphoto2 and gphoto2 version**
gphoto2 2.5.23
libgphoto2      2.5.24
libgphoto2_port 0.12.0  

**To Reproduce**
```
$ env LANG=C gphoto2 --capture-sound
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***
```

Here is the [debug file](https://github.com/gphoto/libgphoto2/files/4673794/debug.txt).

So, is there a problem with the sound acquisition and this camera model?

The commands show in the "Using as webcam" part of [the documentation](http://gphoto.org/doc/remote/) or [this tutorial](https://medium.com/nerdery/dslr-webcam-setup-for-linux-9b6d1b79ae22) works very well for the video. I the gphoto2 sound capture works, I am not sure then how to make it available for the system as a microphone device. Is ffmpeg able to do this for audio the same way it does for video with v4l2?

msmeissn, 2020-05-24:
No driver so far have hooked up --capture-sound, sorry.

As far as I know, EOS cannot record or transfer sound seperately.
You can capture movie with sound (see remote doc on gphoto.org ), but not sound.

closed 2020-05-24 by msmeissn

azmeuk, 2020-05-24:
Thank you for your answer and sorry for the noise in the bugtracker.

It is ok if the sound is recorded/transfered with the video.
The procedure in the documentation create a `/dev/videoX` device that the system can use, but there is no procedure for a similar audio device. Can you give me a clue here?
I understand that `gphoto2 --stdout --capture-movie` outputs both video and audio, so I suppose I should find the right options to pass to ffmpeg?

azmeuk, 2020-05-24:
I was playing with gphoto2 and ffmpeg, and trying to redirect the audio output to pulseaudio.

```
$ gphoto2 --stdout --capture-movie | ffmpeg -i - -f pulse default
ffmpeg version n4.2.2 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 9.3.0 (Arch Linux 9.3.0-1)
  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-fontconfig --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libdav1d --enable-libdrm --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libiec61883 --enable-libjack --enable-libmfx --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxml2 --enable-libxvid --enable-nvdec --enable-nvenc --enable-omx --enable-shared --enable-version3
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
  libpostproc    55.  5.100 / 55.  5.100
Capturing preview frames as movie to 'stdout'. Press Ctrl-C to abort.
[mjpeg @ 0x55e09f5d76c0] Format mjpeg detected only with low score of 25, misdetection possible!
Input #0, mjpeg, from 'pipe:':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: mjpeg (Baseline), yuvj422p(pc, bt470bg/unknown/unknown), 1024x576, 25 tbr, 1200k tbn, 25 tbc
Output #0, pulse, to 'default':
Output file #0 does not contain any stream
```

I understand here that ffmpeg correctly guess the video stream, there is no audio stream. The video stream use the mjpeg codec but [wikipedia says](https://en.wikipedia.org/wiki/Motion_JPEG) that:

> Motion JPEG doesn't code any audio, as it's simply a concatenation of still JPEG frames

msmeissn, 2020-05-25:
--capture-movie on Canon EOS currently just capture a stream of preview jpegs.

Sound is NOT transferred from the camera in this mode.

When you record a separate movie, it will be h.264 or mp4, but this can not be streamed.

azmeuk, 2020-05-25:
So just to be sure:
Is there limitation on Canon EOS that provides sound transfer forever, or is just things to implement in libgphoto2?