gphoto/libgphoto2 #513: Canon Powershot G7X Mark I

yourealwaysbe, 2020-05-26:
**Describe the bug**

Camera closes lens when usb is attached, using `gphoto2 --set-config capture=on` does not reopen the lens. Image capture / preview is not possible.

**Name the camera**
Canon PowerShot G7 X           usb:001,010

**libgphoto2 and gphoto2 version**

gphoto2         2.5.23         gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.24         standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

**To Reproduce**

As above: connect camera, set capture=on, nothing happens.

Log attached:
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/4685453/my-logfile.txt)
.

msmeissn, 2020-05-27:
--set-config capture=on  is only used in pre-2010 powershots.

Try:

--capture-preview
--capture-image-and-download

please

yourealwaysbe, 2020-05-27:
Thanks for the quick response!
I tried --capture-preview and --capture-image-and-download to no avail.
The logs i get are as follows:
[capture-preview-logfile.txt](https://github.com/gphoto/libgphoto2/files/4688050/capture-preview-logfile.txt)
[capture-image-and-download-logfile.txt](https://github.com/gphoto/libgphoto2/files/4688052/capture-image-logfile.txt)

msmeissn, 2020-06-14:
can you perhaps try 2.5.25 libgphoto2 .. I did some Canon related fixes in this area, but cant really promise it will fix this camera.

yourealwaysbe, 2020-06-15:
Thanks for this. No luck i'm afraid. Here's an updated logfile.
[capture-preview-2.2.25-logfile.txt](https://github.com/gphoto/libgphoto2/files/4780826/capture-preview-2.2.25-logfile.txt)