gphoto/libgphoto2 #515: Can't disable bulb mode in nikon d610

MBoaretto25, 2020-05-28:
Hello, Thanks for the software, is an amazing work.

I'm having trouble disabling the bulb mode in my Nikon D610.

Camera Model:
```
Nikon Corporation D610
```
Gphoto Version:
```
gphoto2         2.5.23         
libgphoto2      2.5.24         
libgphoto2_port 0.12.0     
```
System: 
```

Distributor ID: Ubuntu
Description:    Ubuntu 18.04.3 LTS
x86_64 x86_64 GNU/Linux
```

For now bulb mode has only value:
```
gphoto2 --get-config=/main/actions/bulb: 
Label: Bulb Mode
Readonly: 0
Type: TOGGLE
Current: 2
END
```

However When i use:
```
gphoto2 --set-config=/main/actions/bulb=0
*** Error ***
Failed to set new configuration value 0 for configuration entry /main/actions/bulb.
*** Error (-1: 'Unspecified error') ***
```
I think that there is no other possible value than *1* for this mode. 
But i need to disable it because i believe that is whats is causing my camera to take 4 seconds to take a picture.

I couldn`t paste the debug log because of the character limitation.

I appreciate any help.

MBoaretto25, 2020-05-28:
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/4697649/my-logfile.txt)

msmeissn, 2020-05-29:
The bulb config is an action ... it can either start or stop a bulb capture, it is not running by default.

You want to look at "shutterspeed" or "shutterspeed2" configuration setting, there likely is the 4 seconds.

MBoaretto25, 2020-05-29:
Thanks for the answer, I did change these value and the delay went down.

msmeissn, 2020-05-30:
thanks for the feedback!

closed 2020-05-30 by msmeissn