gphoto/libgphoto2 #516: Enable higher resolution for preview / webcam mode for Fuji X-T3

TormodLandet, 2020-05-29:
On a Fujifilm X-T3 camera, freshly reset, connected over USB, with the following settings:

- Single shot (S) mode
- "Pre AF" on

I get a "working" webcam (autofocus etc), but only 640x480 resolution (Ubuntu 20.04). If I connect to a Windows 10 laptop with the new "Fujifilm X Webcam" software I get 1024x768, which *keeps on working* also when reconnected to gphoto2 on Ubuntu. So, this is an in-camera setting of sorts (the higher resolution dissappears after camera reset, but you can turn the camera on and off).

I captured "--list-all-config" before and after the Windows connection (before: low res, after: higher res). The diff is:

```diff
--- list-all-config_after_reset.txt	2020-05-29 12:41:04.415595409 +0200
+++ list-all-config_after_windows.txt	2020-05-29 12:48:32.154338410 +0200
@@ -11,14 +11,14 @@
 Current: 0x1001,0xparam1,0xparam2
 END
 /main/settings/datetime
 Label: Camera Date and Time
 Readonly: 0
 Type: DATE
-Current: 1590748800
-Printable: fr. 29. mai 2020 kl. 12.40 +0200
+Current: 1590749220
+Printable: fr. 29. mai 2020 kl. 12.47 +0200
 Help: Use 'now' as the current time when setting.
 
 END
 /main/settings/copyright
 Label: Copyright
 Readonly: 0
@@ -392,13 +392,13 @@
 Choice: 27 -3
 END
 /main/other/5011
 Label: Date & Time
 Readonly: 0
 Type: TEXT
-Current: 20200529T124000
+Current: 20200529T124700
 END
 /main/other/5012
 Label: Pre-Capture Delay
 Readonly: 0
 Type: MENU
 Current: 0
@@ -749,13 +749,13 @@
 Current: 0,16,61440000,9,0,0,0,0
 END
 /main/other/d20e
 Label: PTP Property 0xd20e
 Readonly: 1
 Type: MENU
-Current: 0
+Current: 32
 Choice: 0 0
 Choice: 1 1
 END
 /main/other/d211
 Label: PTP Property 0xd211
 Readonly: 1
@@ -936,13 +936,13 @@
 Choice: 5 0
 END
 /main/other/d21c
 Label: PTP Property 0xd21c
 Readonly: 0
 Type: MENU
-Current: 4
+Current: 3
 Choice: 0 0
 Choice: 1 1
 Choice: 2 2
 Choice: 3 3
 Choice: 4 4
 Choice: 5 5
```
I tried changing these, but any change to d21c does not seem to stick (and does nothing) and changing d20e is not possible (read only), and the value after (32) is outside the allowable range.

Are these all the settings that can be changed over USB, or only a selection that libgphoto2 exposes?

msmeissn, 2020-09-27:
0xd20e is the number of images free in SDRAM.

closed 2020-09-27 by msmeissn

jakeisnt, 2021-01-22:
Hey @TormodLandet, wondering if you had to do anything special to get this to work besides setting single shooting and Pre-AF mode. I'm unable to get my X-T3 to work with that configuration. Happy to use another channel to contact you if preferred.

msmeissn, 2021-01-23:
FWIW libgphoto2 2.5.26   has a liveviewsize option now if you are just looking for that.