gphoto/libgphoto2 #517: Support Fuji X-T series control over wifi

mzealey, 2020-05-31:
All Fuji X-T series work with the fuji wifi app; even cheaper models such as my X-T100 support shooting/live view over wifi even though they don't support it over USB. There has been some work in reverse-engineering the protocol and I have added and fixed a few features to a working prototype at https://github.com/mzealey/fuji-cam-wifi-tool and implemented a live view with focus. The protocol appears to be a reasonably significant fork from PTPIP and I have made a bit of headway in https://github.com/mzealey/libgphoto2 fuji-xt-ptpip branch in getting it to partially initiate by doing lots of nasty hacks in the ptp code. I don't really have enough knowledge of the PTP protocol to take this further but I can certainly provide packet captures from the fuji-cam-wifi-tool, wifi dumps from the app and testing. I suspect that by changing some of the core ptpip functions (eg header fields, response values) this could probably be bought to 90% working.

Here is a dump from gphoto2 against my hacked branch:

```
gphoto2 --debug  --port ptpip:192.168.0.1:55740 --capture-image
...
0.081917 gp_port_info_list_lookup_path(2): Looking for path 'ptpip:192.168.0.1:55740' (43 entries available)...
0.081924 gp_port_info_list_get_info  (2): Getting info of entry 37 (43 available)...
0.081928 gp_camera_set_port_info     (2): Setting port info for port 'Generic Port' at 'ptpip:192.168.0.1:55740'...
0.082027 gp_port_set_settings        (2): Setting settings...
0.082034 gp_setting_set              (2): Setting key 'port' to value 'ptpip:192.168.0.1:55740' (gphoto2)
0.082038 save_settings               (2): Saving 3 setting(s) to file "/home/mark/.gphoto/settings"
0.082101 gp_camera_init              (2): Initializing camera...
0.082106 gp_camera_init              (2): Loading '/usr/local/lib/libgphoto2/2.5.25.1/ptp2'...
0.083127 gp_port_open                (2): Opening  port...
0.083166 ptp_ptpip_connect           (2): connecting to ptpip:192.168.0.1:55740.
0.085151 ptp_ptpip_init_command_request(3): ptpip/init_cmd data: (hexdump of 82 bytes)
0000  52 00 00 00 01 00 00 00-f2 e4 53 8f ad a5 48 5d  R.........S...H]
0010  87 b2 7f 0b d3 d5 de d0-02 78 a8 c0 6d 00 61 00  .........x..m.a.
0020  72 00 6b 00 6f 00 73 00-00 00 00 00 00 00 00 00  r.k.o.s.........
0030  00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  ................
0040  00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  ................
0050  00 00                  -                         ..              

0.085193 ptp_ptpip_init_command_request [ptpip.c:488](0): return 82 / len 82
0.086772 ptp_ptpip_generic_read      (3): ptpip/generic_read header: (hexdump of 8 bytes)
0000  44 00 00 00 02 00 00 00-                         D.......        

0.086801 ptp_ptpip_generic_read      (3): ptpip/generic_read data: (hexdump of 60 bytes)
0000  00 00 00 00 08 70 b0 61-0a 8b 45 93 b2 e7 93 57  .....p.a..E....W
0010  .. .. .. .. .. 00 2d 00-54 00 31 00 30 00 30 00  ......-.T.1.0.0.
0020  00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  ................
0030  00 00 00 00 00 00 00 00-00 00 00 00              ............    

0.086806 ptp_ptpip_connect           (2): ptpip connected!
0.086812 gp_port_set_timeout         (2): Setting port timeout to 8000 milliseconds.
0.086823 ptp                         (2): PTP: Opening session
0.086858 ptp_ptpip_sendreq           (2): Sending PTP_OC 0x1002 (Open session) (0x1) request...
0.086922 ptp_ptpip_sendreq           (3): ptpip/oprequest data: (hexdump of 16 bytes)
0000  10 00 00 00 01 00 02 10-00 00 00 00 01 00 00 00  ................

0.086945 ptp_ptpip_getresp           (2): Reading PTP_OC 0x1002 (Open session) response...
0.088052 ptp_ptpip_generic_read      (3): ptpip/generic_read header: (hexdump of 8 bytes)
0000  0c 00 00 00 03 00 01 20-                         .......         

0.088064 ptp_ptpip_generic_read      (3): ptpip/generic_read data: (hexdump of 4 bytes)
0000  00 00 00 00            -                         ....            

0.088069 ptp_ptpip_getresp           (2): PTPIP_CMD_RESPONSE
0.088077 ptp_ptpip_sendreq           (2): Sending PTP_OC 0x1016 (Set device property value) (0xdf01) request...
0.088138 ptp_ptpip_sendreq           (3): ptpip/oprequest data: (hexdump of 16 bytes)
0000  10 00 00 00 02 00 16 10-01 00 00 00 01 df 00 00  ................

0.088161 ptp_ptpip_senddata          (2): Sending PTP_OC 0x1016 (Set device property value) data...
0.088166 ptp_ptpip_senddata          (3): ptpip/senddata header: (hexdump of 20 bytes)
0000  14 00 00 00 09 00 00 00-01 00 00 00 04 00 00 00  ................
0010  00 00 00 00            -                         ....            

0.088243 ptp_ptpip_senddata          (3): ptpip/senddata data: (hexdump of 16 bytes)
0000  10 00 00 00 0c 00 00 00-01 00 00 00 00 00 47 3e  ..............G>

0.088257 ptp_ptpip_getresp           (2): Reading PTP_OC 0x1016 (Set device property value) response...
0.112677 ptp_ptpip_generic_read      (3): ptpip/generic_read header: (hexdump of 8 bytes)
0000  0c 00 00 00 03 00 01 20-                         .......         

0.112698 ptp_ptpip_generic_read      (3): ptpip/generic_read data: (hexdump of 4 bytes)
0000  01 00 00 00            -                         ....            

0.112702 ptp_ptpip_getresp           (2): PTPIP_CMD_RESPONSE
0.112712 ptp_ptpip_sendreq           (2): Sending PTP_OC 0x1016 (Set device property value) (0xdf24) request...
0.112775 ptp_ptpip_sendreq           (3): ptpip/oprequest data: (hexdump of 16 bytes)
0000  10 00 00 00 02 00 16 10-02 00 00 00 24 df 00 00  ............$...

0.112805 ptp_ptpip_senddata          (2): Sending PTP_OC 0x1016 (Set device property value) data...
0.112811 ptp_ptpip_senddata          (3): ptpip/senddata header: (hexdump of 20 bytes)
0000  14 00 00 00 09 00 00 00-02 00 00 00 04 00 00 00  ................
0010  00 00 00 00            -                         ....            

0.112884 ptp_ptpip_senddata          (3): ptpip/senddata data: (hexdump of 16 bytes)
0000  10 00 00 00 0c 00 00 00-02 00 00 00 00 00 47 3e  ..............G>

0.112894 ptp_ptpip_getresp           (2): Reading PTP_OC 0x1016 (Set device property value) response...
0.113551 ptp_ptpip_generic_read      (3): ptpip/generic_read header: (hexdump of 8 bytes)
0000  0c 00 00 00 03 00 04 20-                         .......         

0.113566 ptp_ptpip_generic_read      (3): ptpip/generic_read data: (hexdump of 4 bytes)
0000  01 00 00 00            -                         ....            

0.113570 ptp_ptpip_getresp [ptpip.c:410](0): Unsure on type 20040003
0.113574 gp_port_set_timeout         (2): Setting port timeout to 20000 milliseconds.
0.113580 ptp_ptpip_sendreq           (2): Sending PTP_OC 0x1001 (Get device info) request...
0.113641 ptp_ptpip_sendreq           (3): ptpip/oprequest data: (hexdump of 12 bytes)
0000  0c 00 00 00 01 00 01 10-03 00 00 00              ............    

0.113650 ptp_ptpip_getdata           (2): Reading PTP_OC 0x1001 (Get device info) data...
0.114448 ptp_ptpip_generic_read      (3): ptpip/generic_read header: (hexdump of 8 bytes)
0000  0c 00 00 00 03 00 04 20-                         .......         

0.114462 ptp_ptpip_generic_read      (3): ptpip/generic_read data: (hexdump of 4 bytes)
0000  02 00 00 00            -                         ....            

0.114467 ptp_ptpip_getdata [ptpip.c:322](0): got reply type 537133059

0.114473 camera_init [library.c:8797](0): 'ptp_getdeviceinfo(params, &params->deviceinfo)' failed: 'PTP General Error' (0x2002)
0.114485 gp_context_error            (0): PTP General Error

```

msmeissn, 2020-06-02:
You get PTP error 0x2004:

ptp.h:#define PTP_RC_InvalidTransactionID     0x2004

0.113551 ptp_ptpip_generic_read      (3): ptpip/generic_read header: (hexdump of 8 bytes)
0000  0c 00 00 00 03 00 04 20-                         .......      
0000  01 00 00 00            -                         ....            
here it starts to break down.

This means the data send before somehow was not correctly received.

Type 3 would usually be an Init_event_requets. But it should type 7 as this is a response code.
perhaps remap 3 to 7.

The data send request before that looks OK, but perhaps the Fuji expects it differently.

mzealey, 2020-06-02:
As I say the fuji protocol is like ptp but quite different. Here is the 'correct' initialization sequence as hex dump from wireshark so yes those codes changed a bit. The code to do this is at https://github.com/mzealey/fuji-cam-wifi-tool/blob/master/lib/src/commands.cpp#L310 referencing commands at https://github.com/mzealey/fuji-cam-wifi-tool/blob/master/lib/include/message.hpp#L12 . A success response looks like the length plus `uint8_t const type[4] = {0x03, 0x00, 0x01, 0x20};`

It also has an additional port:
```
const int control_server_port = 55740;
const int async_response_server_port = 55741;
const int jpg_stream_server_port = 55742;
```

initialization sequence (port 55740). I'm not sure how fixed this is but I assume it's based on capture from the app:

```
00000000  52 00 00 00 01 00 00 00  f2 e4 53 8f ad a5 48 5d   R....... ..S...H]
00000010  87 b2 7f 0b d3 d5 de d0  02 78 a8 c0 48 00 61 00   ........ .x..H.a.
00000020  63 00 6b 00 65 00 64 00  43 00 6c 00 69 00 65 00   c.k.e.d. C.l.i.e.
00000030  6e 00 74 00 00 00 00 00  00 00 00 00 00 00 00 00   n.t..... ........
00000040  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
00000050  00 00                                              ..
    00000000  44 00 00 00 02 00 00 00  00 00 00 00 08 70 b0 61   D....... .....p.a
    00000010  0a 8b 45 93 b2 e7 93 57  .. .. .. .. .. .. .. ..   .......W ........
    00000020  54 00 31 00 30 00 30 00  00 00 00 00 00 00 00 00   T.1.0.0. ........
    00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ........ ........
    00000040  00 00 00 00                                        ....
00000052  10 00 00 00 01 00 02 10  01 00 00 00 01 00 00 00   ........ ........
    00000044  0c 00 00 00 03 00 01 20  01 00 00 00               .......  ....
00000062  10 00 00 00 01 00 16 10  02 00 00 00 01 df 00 00   ........ ........
00000072  0e 00 00 00 02 00 16 10  02 00 00 00 05 00         ........ ......
    00000050  0c 00 00 00 03 00 01 20  02 00 00 00               .......  ....
00000080  10 00 00 00 01 00 15 10  03 00 00 00 24 df 00 00   ........ ....$...
    0000005C  10 00 00 00 02 00 15 10  03 00 00 00               ........ ....
    00000068  03 00 02 00                                        ....
    0000006C  0c 00 00 00 03 00 01 20  03 00 00 00               .......  ....
00000090  10 00 00 00 01 00 16 10  04 00 00 00 24 df 00 00   ........ ....$...
000000A0  10 00 00 00 02 00 16 10  04 00 00 00 ff 00 02 00   ........ ........
    00000078  0c 00 00 00 03 00 01 20  04 00 00 00               .......  ....
000000B0  0c 00 00 00 01 00 2b 90  05 00 00 00               ......+. ....
    00000084  b2 01 00 00 02 00 2b 90  05 00 00 00               ......+. ....
    00000090  0a 00 00 00 14 00 00 00  19 d0 04 00 00 01 00 01   ........ ........
    000000A0  00 02 02 00 00 00 01 00  12 00 00 00 0c 50 04 00   ........ .....P..
    000000B0  01 02 00 02 00 02 01 00  02 00 16 00 00 00 12 50   ........ .......P
    000000C0  04 00 01 00 00 00 00 02  03 00 00 00 02 00 04 00   ........ ........
    000000D0  1e 00 00 00 7c d1 06 00  01 00 00 00 00 00 00 00   ....|... ........
    000000E0  00 01 00 00 00 00 07 0d  09 10 01 00 00 00 12 00   ........ ........
    000000F0  00 00 45 d2 02 00 01 02  02 02 04 00 00 01 02 03   ..E..... ........
    00000100  4e 00 00 00 10 50 04 00  01 00 00 00 00 02 1f 00   N....P.. ........
    00000110  78 ec c5 ed 13 ef 60 f0  ad f1 fb f2 48 f4 95 f5   x.....`. ....H...
    00000120  e3 f6 30 f8 7d f9 cb fa  18 fc 65 fd b3 fe 00 00   ..0.}... ..e.....
    00000130  4d 01 9b 02 e8 03 35 05  83 06 d0 07 1d 09 6b 0a   M.....5. ......k.
    00000140  b8 0b 05 0d 53 0e a0 0f  ed 10 3b 12 88 13 24 00   ....S... ..;...$.
    00000150  00 00 05 50 04 00 01 02  00 02 00 02 0a 00 02 00   ...P.... ........
    00000160  04 00 06 00 01 80 02 80  03 80 06 80 0a 80 0b 80   ........ ........
    00000170  0c 80 16 00 00 00 07 50  04 00 01 18 01 18 01 02   .......P ........
    00000180  03 00 ff ff dc 00 18 01  26 00 00 00 01 d0 04 00   ........ &.......
    00000190  01 01 00 02 00 02 0b 00  01 00 02 00 03 00 04 00   ........ ........
    000001A0  05 00 0b 00 06 00 07 00  08 00 09 00 0a 00 88 00   ........ ........
    000001B0  00 00 2a d0 06 00 01 c8  00 00 00 40 06 00 00 02   ..*..... ...@....
    000001C0  1d 00 00 19 00 80 80 0c  00 80 40 06 00 80 20 03   ........ ..@... .
    000001D0  00 80 90 01 00 80 00 c8  00 40 00 64 00 40 00 32   ........ .@.d.@.2
    000001E0  00 00 10 27 00 00 40 1f  00 00 00 64 00 40 00 32   ...'..@. ...d.@.2
    000001F0  00 40 00 19 00 00 88 13  00 00 a0 0f 00 00 80 0c   .@...... ........
    00000200  00 00 c4 09 00 00 d0 07  00 00 40 06 00 00 e2 04   ........ ..@.....
    00000210  00 00 e8 03 00 00 20 03  00 00 80 02 00 00 f4 01   ...... . ........
    00000220  00 00 90 01 00 00 40 01  00 00 fa 00 00 00 c8 00   ......@. ........
    00000230  00 00 64 00 00 40                                  ..d..@
    00000236  0c 00 00 00 03 00 01 20  05 00 00 00               .......  ....
000000BC  14 00 00 00 01 00 1c 10  06 00 00 00 00 00 00 00   ........ ........
000000CC  00 00 00 00                                        ....
    00000242  0c 00 00 00 03 00 01 20  06 00 00 00               .......  ....
000000D0  10 00 00 00 01 00 15 10  07 00 00 00 12 d2 00 00   ........ ........
    0000024E  98 00 00 00 02 00 15 10  07 00 00 00               ........ ....
[... cut ...]
    000002BA  00 00 18 d0 02 00 00 00  07 50 80 02 00 00 01 d0   ........ .P......
    000002CA  02 00 00 00 2a d0 40 06  00 00 41 df 00 00 00 00   ....*.@. ..A.....
    000002DA  26 df 01 00 00 00 27 df  01 00 00 00               &.....'. ....
    000002E6  0c 00 00 00 03 00 01 20  07 00 00 00               .......  ....
000000E0  0c 00 00 00 01 00 03 10  08 00 00 00               ........ ....
    000002F2  0c 00 00 00 03 00 01 20  08 00 00 00               .......  ....
000000EC  08 00 00 00 ff ff ff ff                            ........ 
```

msmeissn, 2020-06-11:
55740 is just 15740 (PTPIP port) + 40000 , so I guess this signifies the fuji-ptpip-like protocol ...

msmeissn, 2020-06-11:
{0x03, 0x00, 0x01, 0x20};

This is a  PTP response (type 0x0003) , code is 0x2001 (PTP_RC_OK)

What you pasted looks quite similar to a regular USB bulk stream above.

4 byte size 2 byte type 2 byte code 4 byte sequencene  and then data
xx xx xx xx tt tt cc cc ss ss ss ss  data....

malc0mn, 2020-08-26:
I made a working PTP/IP implementation for a Fuji X-T1 camera in Go: https://github.com/malc0mn/ptp-ip

Might be helpful to get a working implementation in libgphoto2 but Fuji deviates from the standard quite a lot :/

msmeissn, 2020-08-26:
I will see if I can integrate your work in libgphoto2 over time. 

Will need to borrow a X-T1 at some point in time for verification ;)

malc0mn, 2020-08-26:
It will need separate camera 'plugins' or something like that, I'm guessing that Fuji is not the only vendor deviating this much.
I'm planning on thesting my work on a Nikon D700 as well, who knows...

@mzealey the init sequence you post looks identical to the one on the X-T1 which would mean that the implementation I made would also work with the X-T100. That would be very cool :smile:!
For example, I can see [this one](https://github.com/malc0mn/ptp-ip/blob/10de86c522f5b53fbe84872c66acf38665d02a26/ip/packets_fuji.go#L219) in your dumps which was a crucial discovery when I was reverse engineering. I also referred to the fuji-cam-wifi-tool but judging from the code, they are/were not really sure exactly what the data being sent to and fro was all about.
I really wanted to understand it all, so I documented all I found in the code.

msmeissn, 2020-08-26:
PTP/IP usage so far was quite similar between cameras.

I think I will add some seperate fuji code, but still try reusing as much as possible from the generic code.

malc0mn, 2020-08-26:
Good to know. I only partly implemented the 'standard' ptp/ip stuff since I had to deviate from it quite instantly.
I was planning to have a generic implementation as well.
What vendors adhere to the standard? Is Nikon one of them? Would allow me to also finish my generic implementation :wink:

msmeissn, 2020-08-26:
Nikon works with generic PTP/IP.

For Canon and Sony I currently have no memory of how my testing went , and have no cameras available. (I think Canon EOS and Sony might just work.)

The Canon PowerShot needs some upnp magic before activating ptp/ip which I have not figured out.

malc0mn, 2020-08-26:
Thanks a bunch for the info! I'll be using that D700 then for the generic side :smile:.

msmeissn, 2020-08-29:
I borrowed a Xt-4, but its hard to get access to the camera.

- In the fixed wireless tethering mode, the camera gets an IP from my wifi router, by does not have any port open apparently.

- In the smartphone mode I can attach to the cameras AP, but it does not give me an IP ... is it adhoc mode only?

msmeissn, 2020-08-30:
While I made it connect, and have fuji specific ptpip code now added ...

The camera does not support the most basic PTP GetDeviceInfo command over wifi (it returns an empty data block and success).

I think I need to synthesize deviceinfo content :(

msmeissn, 2020-08-30:
i did  that ... 

- preview over wifi works , can capture 25 fps
- capture image itself does not
- camera disconnects after 1 gphoto2 session which is a pain

hfiguiere, 2020-08-31:
>     * camera disconnects after 1 gphoto2 session which is a pain

This is what I'm seeing (if I understand what you are saying) with the Canon G7X MarkII. I have to use gphoto2 --shell to do more than one thing over PTP/IP.

So maybe a more generic solution ought to be found for that problem since it's not unique.