gphoto/libgphoto2 #518: Nikon D7000 Bulb capture: should it work?

gubihu, 2020-06-01:
On Nikon D7000 "Buld" is listed as available shutter speed, yet bulb capture does not work:

LC_ALL=C gphoto2 --debug --debug-logfile=d7000-logfile.txt --shell
gphoto2: {/home/astroberry/gphoto} /> get-config /main/capturesettings/shutterspeed
Label: Shutter Speed
Readonly: 0
Type: RADIO
Current: 0.0004s
Choice: 0 0.0001s
Choice: 1 0.0002s
Choice: 2 0.0003s
Choice: 3 0.0004s
Choice: 4 0.0005s
Choice: 5 0.0006s
Choice: 6 0.0008s
Choice: 7 0.0010s
Choice: 8 0.0012s
Choice: 9 0.0015s
Choice: 10 0.0020s
Choice: 11 0.0025s
Choice: 12 0.0031s
Choice: 13 0.0040s
Choice: 14 0.0050s
Choice: 15 0.0062s
Choice: 16 0.0080s
Choice: 17 0.0100s
Choice: 18 0.0125s
Choice: 19 0.0166s
Choice: 20 0.0200s
Choice: 21 0.0250s
Choice: 22 0.0333s
Choice: 23 0.0400s
Choice: 24 0.0500s
Choice: 25 0.0666s
Choice: 26 0.0769s
Choice: 27 0.1000s
Choice: 28 0.1250s
Choice: 29 0.1666s
Choice: 30 0.2000s
Choice: 31 0.2500s
Choice: 32 0.3333s
Choice: 33 0.4000s
Choice: 34 0.5000s
Choice: 35 0.6250s
Choice: 36 0.7692s
Choice: 37 1.0000s
Choice: 38 1.3000s
Choice: 39 1.6000s
Choice: 40 2.0000s
Choice: 41 2.5000s
Choice: 42 3.0000s
Choice: 43 4.0000s
Choice: 44 5.0000s
Choice: 45 6.0000s
Choice: 46 8.0000s
Choice: 47 10.0000s
Choice: 48 13.0000s
Choice: 49 15.0000s
Choice: 50 20.0000s
Choice: 51 25.0000s
Choice: 52 30.0000s
Choice: 53 Bulb
END
gphoto2: {/home/astroberry/gphoto} /> set-config-index /main/capturesettings/shutterspeed=52
gphoto2: {/home/astroberry/gphoto} /> capture-image
New file is in location /capt0000.nef on the camera                            
gphoto2: {/home/astroberry/gphoto} /> set-config-index /main/capturesettings/shutterspeed=53
gphoto2: {/home/astroberry/gphoto} /> capture-image

*** Error ***              
Shutter Speed Bulb
ERROR: Could not capture image.
ERROR: Could not capture.
gphoto2: {/home/astroberry/gphoto} /> 

also:

LC_ALL=C gphoto2 --debug --debug-logfile=d7000-bulb-logfile.txt --bulb 40 --capture-image-and-download
Bulb mode enabled (exposure time: 40s).                                        

*** Error ***              
bulb not found in configuration tree.
ERROR: Could not set bulb capture, result -1.
*** Error (-1: 'Unspecified error') ***       

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs
gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Nikon DSC D7000 (PTP mode)     usb:001,009     

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version
gphoto2 --version
gphoto2 2.5.20

Copyright (c) 2000-2018 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.20         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.24.1       standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

**To Reproduce**
Steps to reproduce the behavior:

When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.

msmeissn, 2020-06-02:
bulb capture is special, you need to

set-config bulb=1
wait-event 100s
set-config bulb=0
wait-event-and-download=5s

or similar.

--capture-image will not work in bulb mode currently.

gubihu, 2020-06-02:
I understand that bulb is special. But should not these work?:

LC_ALL=C gphoto2 --debug --debug-logfile=d7000-bulb-logfile.txt --bulb 40 --capture-image-and-download
Bulb mode enabled (exposure time: 40s).

*** Error ***
bulb not found in configuration tree.
ERROR: Could not set bulb capture, result -1.
*** Error (-1: 'Unspecified error') ***

I have also tried these:

/> set-config bulb=1

*** Error ***              
bulb not found in configuration tree.
*** Error (-1: 'Unspecified error') ***

msmeissn, 2020-06-02:
Hmm, the D7000 seems to lack "TerminateCapture" operation, which we use for Bulb capture.

I cross checked the SDK , it does not look like it is available to control "bulb capture" from the PC for the D7000.

gubihu, 2020-06-02:
Thank you very much. So bulb is just simple does not work with the D7000, I have to live with that fact.

How can I know which model is supported for bulb and which is not? It can be an important factor when considering bying a new machine.
The D7000 seems to be very well supported, a lots of things works, just this one does not. So one can be never be sure.

closed 2020-06-02 by gubihu

msmeissn, 2020-06-02:
Nikons allow SDK downloads ... I have them here, might extract the list at some point.
Theere look for "TerminateCapture" 

I checked, the D7100 has remote Bulb capture, so I guess also the D7200 and D7500 too.

gubihu, 2020-06-02:
Thank you for your time and patient.