gphoto/libgphoto2 #519: Live preview in Canon  PowerShot SX740 HS not working

oe8bck, 2020-06-04:
If I try to record a movie, an error is stated:

> $ env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt 
> --capture-movie=10
> Capturing 10 preview frames as movie to 'movie.mjpg'.
> ERROR: Movie capture error... Exiting.
> Movie capture finished (0 frames)

Logfile starts with:
---
0.000026 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN
SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000113 main                        (2): gphoto2 2.5.23.1
0.000157 main                        (2): gphoto2 has been compiled with the
following options:
0.000195 main                        (2):  + gcc (C compiler used)
0.000232 main                        (2):  + popt (mandatory, for handling
command-line parameters)
0.000269 main                        (2):  + exif (for displaying EXIF information)
0.000305 main                        (2):  + no cdk (for accessing configuration
options)
0.000342 main                        (2):  + no aa (for displaying live previews)
0.000378 main                        (2):  + jpeg (for displaying live previews in
JPEG format)
0.000415 main                        (2):  + readline (for easy navigation in the
shell)
0.000452 main                        (2): libgphoto2 2.5.24.1
0.000489 main                        (2): libgphoto2 has been compiled with the
following options:
0.000528 main                        (2):  + standard camlib set (ax203 canon
digigr8 dimagev directory jl2005a jl2005c kodak_dc240 lumix mars pentax ptp2
ricoh_g3 sierra sonix sq905 st2205 topfield tp6801)
0.000565 main                        (2):  + gcc (C compiler used)
0.000601 main                        (2):  + ltdl (for portable loading of camlibs)
0.000637 main                        (2):  + EXIF (for special handling of EXIF files)
0.000674 main                        (2): libgphoto2_port 0.12.0
0.000712 main                        (2): libgphoto2_port has been compiled with the
following options:
0.000748 main                        (2):  + iolibs: disk ptpip serial usb1
usbdiskdirect usbscsi
0.000784 main                        (2):  + gcc (C compiler used)
0.000821 main                        (2):  + ltdl (for portable loading of iolibs)
0.000857 main                        (2):  + EXIF (for vusb)
0.000894 main                        (2):  + USB (libusb1, for USB cameras)
0.000967 main                        (2):  + serial (for serial cameras)
0.001007 main                        (2):  + no resmgr (serial port access and locking)
0.001043 main                        (2):  + no ttylock (serial port locking)
0.001080 main                        (2):  + no lockdev (serial port locking)
0.001118 main                        (2): CAMLIBS env var not set, using
compile-time default instead
0.001154 main                        (2): IOLIBS env var not set, using compile-time
default instead
0.001191 main                        (2): invoked with following arguments:
0.001227 main                        (2):   --debug
0.001264 main                        (2):   --debug-logfile=my-logfile.txt
0.001301 main                        (2):   --capture-movie=10
0.001378 load_settings               (2): Creating gphoto config directory
('/home/podex/.gphoto')
0.001524 load_settings               (2): Loading settings from file
'/home/podex/.gphoto/settings'.
0.001657 main                        (2): The user has not specified both a model
and a port. Try to figure them out.
0.001710 gp_port_info_list_load      (2): Using ltdl to load io-drivers from
'/usr/local/lib/libgphoto2_port/0.12.0'...
0.001889 foreach_func                (2): Called for filename
'/usr/local/lib/libgphoto2_port/0.12.0/disk'.
0.002222 gp_port_library_list        (2): found fstab fsname
UUID=d85c9f4f-d8e4-4b68-a7c0-4e7b6fd1f651
---

I can send you the complete logfile, just drop me a mail!

KR,
Christof

msmeissn, 2020-06-07:
a german article https://www.fotointern.ch/archiv/2015/01/21/samsung-nx1-wichtiges-firmware-update-fuer-videofilmer/ 

Says firmware 1.2 adds USB remote control features ... which firmware version do you have currently? Can you see if you can upgrade if not yet at 1.2?

msmeissn, 2020-06-07:
The SX series are a bit problematic there.

Can you try:

gphoto2  --get-config output

        to get all output variables

Can you then try

        gphoto2 --set-config viewfinder=1 --set-config output="PC" --capture-movie=10

and for output also try other values from the --get-config output
containing "PC" or "MOBILE" or "MOBILE2" ?

oe8bck, 2020-06-07:
Hi,

Thanks for the hints!

But they are not working. :-(

In the meantime, I updated the firmware to version 1.0.2 and now I 
cannot set the camera in record mode most of the time. If I plug the USB 
cable in, I get the PC symbol on the display. If I press the power 
button, the camera is switched off.

If I switch it on with the cable not attached and then attach the cable, 
unmount the drive (which is mounted automatically in my Linux Mint) the 
camera sometimes stays in record mode. If this is the case, the command 
triggers a "Error 70" message on the camera.

Obviously, the SX series is a little bit awkward.

KR,
Christof

-- 
---------------------------------------------------------------------
Christof Bodner <oe8bck@gmail.com>
Linux - Life is too short for reboots
- First they ignore you. Then they laugh at you. Then they
fight you(*). Then you win. -- Ghandi
---------------------------------------------------------------------
email: mailto: oe8bck@gmail.com
Tel.: +43-650-7215383
To contact me, use https://www.signal.org/
---------------------------------------------------------------------
GnuPG public key: 8A265334
Fingerprint CF71 08D2 18B8 A824 37A5 B80E 0888 37E1 8A26 5334
---------------------------------------------------------------------

oe8bck, 2021-01-01:
Hi again!

I think I have to report that things get worse in the mean time! I use Linux Mint 20 (Linux ThinkPad 5.4.0-52-generic #57-Ubuntu SMP Thu Oct 15 10:57:00 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux) and the camera is not shown any more if I plug it in the USB port. No connection if connection is made camera switched on or off or in play mode. It does not show up if I fire "lsusb".

What happened there? Why is the Canon PowerShot SX740 HS not showing up any more?

msmeissn, 2021-01-01:
if it is not showing with "lsusb", something lower level is amiss. exchange the USB cable, the USB port ... also check if the camera has enough battery power.

If its a kernel issue, it will be visible in "dmesg", but check the hardware first.

oe8bck, 2021-01-01:
WOW, thanks for the fast response! (I think you are also in "hard lockdown"?)

Of course I changed the cable and the battery is (almost) full and checked "dmesg".

BUT

Thanks to your hint I did not plug it into the USB 3.0 HUB but directly to the laptop. And now it shows up :-) Sorry for the PEBKAC on my side.
```
[ 7049.407883] usb 2-2: new high-speed USB device number 23 using xhci_hcd
[ 7049.556372] usb 2-2: New USB device found, idVendor=04a9, idProduct=32e4, bcdDevice= 0.02
[ 7049.556374] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 7049.556375] usb 2-2: Product: Canon Digital Camera
[ 7049.556376] usb 2-2: Manufacturer: Canon Inc.
```
So now I can run
```
$ LANG=C gphoto2 --get-config output
Label: Camera Output                                                           
Readonly: 0
Type: RADIO
Current: TFT
Choice: 0 Off
Choice: 1 TFT
Choice: 2 PC
Choice: 3 TFT + PC
Choice: 4 MOBILE
Choice: 5 TFT + MOBILE
Choice: 6 PC + MOBILE
Choice: 7 TFT + PC + MOBILE
Choice: 8 MOBILE2
Choice: 9 TFT + MOBILE2
Choice: 10 PC + MOBILE2
Choice: 11 TFT + PC + MOBILE2
END
```
and 
```
$ LANG=C  gphoto2 --set-config viewfinder=1 --set-config output="PC" --capture-movie=10
Capturing 10 preview frames as movie to 'movie.mjpg'.                          
ERROR: Movie capture error... Exiting.
Movie capture finished (0 frames)
$ LANG=C  gphoto2 --set-config viewfinder=1 --set-config output="MOBILE" --capture-movie=10
Capturing 10 preview frames as movie to 'movie.mjpg'.                          
ERROR: Movie capture error... Exiting.
Movie capture finished (0 frames)
$ LANG=C  gphoto2 --set-config viewfinder=1 --set-config output="MOBILE2" --capture-movie=10
Capturing 10 preview frames as movie to 'movie.mjpg'.                          
ERROR: Movie capture error... Exiting.
Movie capture finished (0 frames)
```
Should I send you the logfile? PM?

msmeissn, 2021-01-01:
can you attach the logfile here? alternatively mail me to marcus@jet.franken.de

msmeissn, 2021-01-01:
I got the logfile... What I see is that any try to select output other that "TFT" (1) ... the camera does not acknowledge the new value and flips it back to "1" (TFT) immediately. 

I am trying to think if this is something we can force otherwise :/

msmeissn, 2021-01-01:
could you try "TFT + PC" perhaps? or "TFT + MOBILE" ?

oe8bck, 2021-01-01:
```
$ LANG=C  gphoto2 --set-config viewfinder=1 --set-config output="TFT + MOBILE" --capture-movie=10
Capturing 10 preview frames as movie to 'movie.mjpg'.                          
ERROR: Movie capture error... Exiting.
Movie capture finished (0 frames)
[1]+  Fertig                  xed
$ LANG=C  gphoto2 --set-config viewfinder=1 --set-config output="TFT + PC" --capture-movie=10
Capturing 10 preview frames as movie to 'movie.mjpg'.                          
ERROR: Movie capture error... Exiting.
Movie capture finished (0 frames)
```

oe8bck, 2021-01-01:
Whenever I start gphoto2, the last picture taken is shown like in play mode and not a live (pre)view.

oe8bck, 2021-01-01:
```
$ LANG=C  gphoto2 --summary
Camera summary:                                                                
Manufacturer: Canon.Inc
Model: Canon PowerShot SX740 HS
  Version: 3-1.0.2
  Serial Number: e...
Vendor Extension ID: 0xb (1.0)
Vendor Extension Description: 

Capture Formats: JPEG
Display Formats: Association/Directory, Script, DPOF, MS AVI, MS Wave, JPEG, CRW, Unknown(b103), Unknown(bf02), Defined Type, Unknown(b104), Unknown(b105)

Device Capabilities:
	File Download, File Deletion, File Upload
	No Image Capture, No Open Capture, Canon EOS Capture, Canon EOS Capture 2
	Canon Wifi support

Storage Devices Summary:
store_00020001:
	StorageDescription: SD
	VolumeLabel: 
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Digital Camera Layout (DCIM)
	Access Capability: Read-Write
	Maximum Capability: 63831015424 (60874 MB)
	Free Space (Bytes): 60964339712 (58140 MB)
	Free Space (Images): -1

Device Property Summary:
Property 0xd402:(read only) (type=0xffff) 'Canon PowerShot SX740 HS'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd406:(readwrite) (type=0xffff) 'Unknown Initiator'
Property 0xd303:(read only) (type=0x2) 1
Battery Level(0x5001):(read only) (type=0x2) Enumeration [100,0,75,0,50] value: 100% (100)
```

oe8bck, 2021-01-01:
BTW same behaviour if I just want to capture a move:
```
$ LANG=C  gphoto2 --capture-movie=10
Capturing 10 preview frames as movie to 'movie.mjpg'.                          
ERROR: Movie capture error... Exiting.
Movie capture finished (0 frames)
```
The last still picture taken is shown after starting gphoto2

msmeissn, 2021-01-02:
its basically mostly the same issue all the time. the camera reports 0xa102 error back to us. :/

msmeissn, 2021-01-03:
can you check if the camrea is in "movie" mode, and switch it to "still picture" mode and see if that helps?

oe8bck, 2021-01-04:
If I switched the camera on with the USB cable plugged in, the desktop symbol is shown. Then I eject the (automounted) drive ibn order to be able to start gphoto2.
Then I set the selection wheel to the camera, the lens extracts and live video shown in the display.

```env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-movie=10```
The lens retracted, expanded again, showed the last image taken for a second, retracted the lens again and when the program stopps, the desktop symbol is shown. 

No movie is taken. Sometimes a short live preview is shown, I cannot say what is the reason but only short.

Same if I set the mode wheel to AUTO, SCN or M, 0xa102 error is always in the log-file. :-(

msmeissn, 2021-01-06:
The powershot series sadly is not reliable in this regard ... it just might not be possible :(

oe8bck, 2021-01-06:
Got it, thanks for the support anyway!!!!!!

The Powershot is also not stable when I try to transfer the photos via WiFi to my Android phone. There are many possibilities and sometimes one of them works. This is not nice, so if I would have known this before, maybe I would have bought another brand.

KR,
Christof

closed 2021-01-06 by oe8bck