gphoto/libgphoto2 #520: libgphoto2 is slow on nvidia jetson nano (ARMv8)

notnot, 2020-06-11:
I am experimenting with a nvidia jetson nano as a robot controller. My robot is normally controlled with a regular PC and it would be interesting to control it with a much smaller and cheaper device.
I have a Sony A5100 camera connected via USB, and am controlling this camera with libgphoto2. It does work, in principle, but the download speed of preview and full resolution photos is low. On the PC i could get about 25 previews a second (the upper limit of the camera i think), on the nano only about 2. The previews are small (1024x680) .jpg files, on average 75KB large. I would expect to see a higher throughput!
The nano has quad core ARMv8 and 4xUSB3.1. 

Any idea where the bottleneck might be? Would it be worth it to tray and upgrade libusb for instance, or are there other low level libraries that libgphoto2 uses?

By the way, it is great seeing libgphoto2 compile on such a tiny system, and doing its thing!

msmeissn, 2020-06-11:
If you capture a logfile with the commandline tool, it will contain timestamps we can check out.

e.g.

gphoto2 --capture-movie=1s --debug --debug-logfile=preview.log

notnot, 2020-06-11:
Ok, i have that data available but it is kinda large to  copy and paste  it here?
Iĺl attach it.
[preview.log](https://github.com/gphoto/libgphoto2/files/4765869/preview.log)

notnot, 2020-06-11:
Hmm, the preview.log that i posted above mentions an error near the end:
0.173235 gp_context_error            (0): Sorry, your Sony camera does not seem to return a JPEG image in LiveView mode

This is strange, because the A5100 does provide liveview that can be captured. Maybe it was a spurious error. I repeated the test and now i have a preview.log without that error. I hope it illuminates you.
[preview.log](https://github.com/gphoto/libgphoto2/files/4767727/preview.log)

notnot, 2020-06-12:
Good news! I was inspecting the preview logs and noticed that somehow gphoto2 was using an older libgphoto2, which turned out to be installed on the nvidia jetson nano already. I installed 2.5.25 yesterday but it was not used. After deleting all the old stuff, i have it working well again, and now the preview download speed seems fine :) It takes 15 to 20 ms for one frame, which took +- 320 ms before the fix!
All i can say now is that you did a great job going from libgphoto2 2.5.16 to 2.5.25! For your interest, i will attach a new preview.log here, which was generated by the updated system.
[preview.log](https://github.com/gphoto/libgphoto2/files/4767858/preview.log)

msmeissn, 2020-06-12:
good to hear it is solved:)

closed 2020-06-12 by msmeissn