gphoto/libgphoto2 #521: Canon EOS M Full-Press failed (0x2019: PTP Device Busy)

jumaesmo, 2020-06-11:
Hello everyone, I have an error with the Canon SX530 HS, when capturing the image I present the following error

gphoto2 2.5.20 and 2.5.23

0.000389 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000565 main                        (2): gphoto2 2.5.20
0.000605 main                        (2): gphoto2 has been compiled with the following options:
0.000637 main                        (2):  + gcc (C compiler used)
0.000664 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000692 main                        (2):  + exif (for displaying EXIF information)
0.000718 main                        (2):  + no cdk (for accessing configuration options)
0.000744 main                        (2):  + no aa (for displaying live previews)
0.000769 main                        (2):  + no jpeg (for displaying live previews in JPEG format)
0.000795 main                        (2):  + no readline (for easy navigation in the shell)
0.000827 main                        (2): libgphoto2 2.5.21
0.000862 main                        (2): libgphoto2 has been compiled with the following options:
0.000889 main                        (2):  + all camlibs
0.000914 main                        (2):  + gcc (C compiler used)
0.000939 main                        (2):  + ltdl (for portable loading of camlibs)
0.000964 main                        (2):  + EXIF (for special handling of EXIF files)
0.000998 main                        (2): libgphoto2_port 0.12.0
0.001033 main                        (2): libgphoto2_port has been compiled with the following options:
0.001060 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.001086 main                        (2):  + gcc (C compiler used)
0.001111 main                        (2):  + ltdl (for portable loading of iolibs)
0.001138 main                        (2):  + USB (libusb1, for USB cameras)
0.001164 main                        (2):  + serial (for serial cameras)
0.001189 main                        (2):  + no resmgr (serial port access and locking)
0.001214 main                        (2):  + no ttylock (serial port locking)
0.001239 main                        (2):  + no lockdev (serial port locking)
0.001266 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.001291 main                        (2): IOLIBS env var not set, using compile-time default instead
0.001316 main                        (2): invoked with following arguments:
0.001341 main                        (2):   --capture-image
0.001364 main                        (2):   --debug
0.001390 main                        (2):   --debug-logfile
0.001413 main                        (2):   log.tx
0.001561 load_settings               (2): Creating gphoto config directory ('/root/.gphoto')
0.001857 load_settings               (2): Loading settings from file '/root/.gphoto/settings'.
0.002083 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.002148 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/local/lib/libgphoto2_port/0.12.0'...
0.002639 foreach_func                (2): Called for filename '/usr/local/lib/libgphoto2_port/0.12.0/disk'.
0.003584 gp_port_library_list        (2): found fstab fsname proc
0.003680 gp_port_library_list        (2): found fstab fsname /dev/mmcblk0p6
0.003778 gp_port_library_list        (2): found fstab fsname /dev/mmcblk0p7
0.004171 gp_port_library_list        (2): found mtab fsname /dev/root
0.004240 gp_port_library_list        (2): found mtab fsname devtmpfs
0.004288 gp_port_library_list        (2): found mtab fsname sysfs
0.004329 gp_port_library_list        (2): found mtab fsname proc
0.004369 gp_port_library_list        (2): found mtab fsname tmpfs
0.004454 gp_port_library_list        (2): found mtab fsname devpts
0.004498 gp_port_library_list        (2): found mtab fsname tmpfs
0.004577 gp_port_library_list        (2): found mtab fsname tmpfs
0.004658 gp_port_library_list        (2): found mtab fsname tmpfs
0.004758 gp_port_library_list        (2): found mtab fsname cgroup
0.004802 gp_port_library_list        (2): found mtab fsname cgroup
0.004911 gp_port_library_list        (2): found mtab fsname cgroup
0.004955 gp_port_library_list        (2): found mtab fsname cgroup
0.004997 gp_port_library_list        (2): found mtab fsname cgroup
0.005211 gp_port_library_list        (2): found mtab fsname cgroup
0.005255 gp_port_library_list        (2): found mtab fsname cgroup
0.005294 gp_port_library_list        (2): found mtab fsname mqueue
0.005334 gp_port_library_list        (2): found mtab fsname systemd-1
0.005375 gp_port_library_list        (2): found mtab fsname sunrpc
0.005461 gp_port_library_list        (2): found mtab fsname debugfs
0.005560 gp_port_library_list        (2): found mtab fsname configfs
0.005667 gp_port_library_list        (2): found mtab fsname /dev/mmcblk0p6
0.005745 gp_port_library_list        (2): found mtab fsname tmpfs
0.005837 gp_port_library_list        (2): found mtab fsname gvfsd-fuse
0.005881 gp_port_library_list        (2): found mtab fsname fusectl
[... 959 lines cut ...]
0.744354 _cb_irq                     (2): Transfer 0x179ee20 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
0.744393 _cb_irq                     (2): 0x17a5668 with status 3
0.744423 _cb_irq                     (2): Transfer 0x17a5668 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
0.744460 _cb_irq                     (2): 0x17a5828 with status 3
0.744490 _cb_irq                     (2): Transfer 0x17a5828 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
0.744527 _cb_irq                     (2): 0x17a59e8 with status 3
0.744557 _cb_irq                     (2): Transfer 0x17a59e8 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
0.744594 _cb_irq                     (2): 0x17a5ba8 with status 3
0.744625 _cb_irq                     (2): Transfer 0x17a5ba8 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
0.745204 gp_filesystem_reset         (2): resetting filesystem
0.745246 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
0.745275 gp_filesystem_lru_clear     (2): fscache LRU list already empty
0.745304 delete_all_folders          (2): Internally deleting all folders from '/'...
0.745333 lookup_folder               (2): Lookup folder '/'...
0.745361 lookup_folder               (2): Found! / is 0x1786b40
0.745425 recurse_delete_folder       (2): Recurse delete folder 0x1786b40//
0.745456 gp_port_free                (2): Freeing port...
0.745483 gp_port_close               (2): Closing port...
0.746336 gp_filesystem_reset         (2): resetting filesystem
0.746374 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
0.746403 gp_filesystem_lru_clear     (2): fscache LRU list already empty
0.746431 delete_all_folders          (2): Internally deleting all folders from '/'...
0.746459 lookup_folder               (2): Lookup folder '/'...
0.746487 lookup_folder               (2): Found! / is 0x1786b40
0.746517 recurse_delete_folder       (2): Recurse delete folder 0x1786b40//

hfiguiere, 2020-06-11:
> gphoto2 2.5.20 and 2.5.23

The log say libgphoto2 2.5.21. 
You probably want 2.5.25

jumaesmo, 2020-06-12:
in version 
gphoto2         2.5.23.1       gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.25.1       standard camlibs (SKIPPING lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

debug file

0.000482 main                        (2): INCLUYA SIEMPRE LA SIGUIENTE LÍNEA CUANDO ENVÍE MENSAJES DE DEPURACIÓN A LA LISTA DE DISTRIBUCIÓN DE CORREO:
0.000609 main                        (2): gphoto2 2.5.23.1
0.001041 main                        (2): gphoto2 has been compiled with the following options:
0.001062 main                        (2):  + gcc (C compiler used)
0.001069 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.001076 main                        (2):  + exif (for displaying EXIF information)
0.001081 main                        (2):  + no cdk (for accessing configuration options)
0.001086 main                        (2):  + no aa (for displaying live previews)
0.001090 main                        (2):  + no jpeg (for displaying live previews in JPEG format)
0.001093 main                        (2):  + no readline (for easy navigation in the shell)
0.001104 main                        (2): libgphoto2 2.5.25.1
0.001110 main                        (2): libgphoto2 has been compiled with the following options:
0.001114 main                        (2):  + standard camlib set SKIPPING SOME (ax203 canon digigr8 dimagev directory jl2005a jl2005c kodak_dc240 mars pentax ptp2 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801 SKIPPING lumix)
0.001118 main                        (2):  + gcc (C compiler used)
0.001122 main                        (2):  + ltdl (for portable loading of camlibs)
0.001126 main                        (2):  + EXIF (for special handling of EXIF files)
0.001132 main                        (2): libgphoto2_port 0.12.0
0.001137 main                        (2): libgphoto2_port has been compiled with the following options:
0.001141 main                        (2):  + iolibs: disk ptpip serial usb usbdiskdirect usbscsi
0.001144 main                        (2):  + gcc (C compiler used)
0.001148 main                        (2):  + ltdl (for portable loading of iolibs)
0.001152 main                        (2):  + EXIF (for vusb)
0.001155 main                        (2):  + USB (libusb0, for USB cameras)
0.001159 main                        (2):  + serial (for serial cameras)
0.001163 main                        (2):  + no resmgr (serial port access and locking)
0.001166 main                        (2):  + no ttylock (serial port locking)
0.001170 main                        (2):  + no lockdev (serial port locking)
0.001174 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.001177 main                        (2): IOLIBS env var not set, using compile-time default instead
0.001181 main                        (2): invoked with following arguments:
0.001185 main                        (2):   --debug
0.001189 main                        (2):   --debug-logfile
0.001193 main                        (2):   log.txt
0.001212 load_settings               (2): Creating gphoto config directory ('/root/.gphoto')
0.001250 load_settings               (2): Loading settings from file '/root/.gphoto/settings'.
0.001345 gp_camera_free              (2): Freeing camera...
0.001351 gp_port_free                (2): Freeing port...
0.001357 gp_filesystem_reset         (2): resetting filesystem
0.001361 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
0.001364 gp_filesystem_lru_clear     (2): fscache LRU list already empty
0.001368 delete_all_folders          (2): Internally deleting all folders from '/'...
0.001373 lookup_folder               (2): Lookup folder '/'...
0.001377 lookup_folder               (2): Found! / is 0x55881f001820
0.001381 recurse_delete_folder       (2): Recurse delete folder 0x55881f001820//
0.001386 delete_all_files            (2): Delete all files in folder 0x55881f001820//

msmeissn, 2020-06-13:
The SX series cameras capture support is a bit fragile and we are trying to improve it, with more or less luck.

Can you update to libgphoto2 2.5.25 and see if this helps?

jumaesmo, 2020-06-13:
Hello, how are you ?, I already generated tests with version 2.5.25.1 and the failure continues.

alhayden, 2020-06-26:
I've run into the exact same problem with the same camera model.
In fact, just trying to `set-config capture=on` yields
```*** Error ***              
Failed to set new configuration value on for configuration entry capture.
*** Error (-1: 'Unspecified error') *** ```

Version info:
```
0.000271 main                        (2): gphoto2 2.5.23
0.000287 main                        (2): gphoto2 has been compiled with the following options:
0.000297 main                        (2):  + gcc (C compiler used)
0.000306 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000314 main                        (2):  + exif (for displaying EXIF information)
0.000322 main                        (2):  + no cdk (for accessing configuration options)
0.000329 main                        (2):  + no aa (for displaying live previews)
0.000337 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000345 main                        (2):  + readline (for easy navigation in the shell)
0.000365 main                        (2): libgphoto2 2.5.25
0.000375 main                        (2): libgphoto2 has been compiled with the following options:
0.000386 main                        (2):  + standard camlib set (ax203 canon digigr8 dimagev directory jl2005a jl2005c kodak_dc240 lumix mars pentax ptp2 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801)
0.000394 main                        (2):  + gcc (C compiler used)
0.000402 main                        (2):  + ltdl (for portable loading of camlibs)
0.000410 main                        (2):  + EXIF (for special handling of EXIF files)
0.000425 main                        (2): libgphoto2_port 0.12.0
0.000435 main                        (2): libgphoto2_port has been compiled with the following options:
0.000442 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.000450 main                        (2):  + gcc (C compiler used)
0.000458 main                        (2):  + ltdl (for portable loading of iolibs)
0.000466 main                        (2):  + EXIF (for vusb)
0.000474 main                        (2):  + USB (libusb1, for USB cameras)
0.000482 main                        (2):  + serial (for serial cameras)
0.000490 main                        (2):  + no resmgr (serial port access and locking)
0.000497 main                        (2):  + no ttylock (serial port locking)
0.000505 main                        (2):  + no lockdev (serial port locking)
0.000514 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000521 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000530 main                        (2): invoked with following arguments:
0.000538 main                        (2):   --debug
0.000546 main                        (2):   --debug-logfile=my-logfile.txt
0.000554 main                        (2):   --set-config
0.000562 main                        (2):   capture=on
```
[the whole logfile](https://github.com/gphoto/libgphoto2/files/4834743/my-logfile.txt)

ssiabruk, 2020-07-15:
Hi, Marcus.
I have the same problem.
Can I help with anything?

> gphoto2 --version
gphoto2         2.5.23.1       gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.25.1       standard camlibs (SKIPPING docupen lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
---
> gphoto2 --auto-detect
Canon PowerShot SX530 HS       usb:001,011     
-----

> gphoto2 --capture-image                                                                              
*** Error ***              
Canon EOS M Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-110: 'I/O in progress') ***
log:
ptp_usb_getresp [usb.c:514] (0): PTP_OC 0x9128 receiving resp failed: PTP Device Busy (0x2019)
camera_trigger_canon_eos_capture [library.c:5396](0): 'ptp_canon_eos_remotereleaseon (params, 3, 0)' failed: Canon EOS M Full-Press failed (0x2019: PTP Device Busy)
gp_context_error            (0): Canon EOS M Full-Press failed (0x2019: PTP Device Busy)
camera_canon_eos_capture [library.c:4035](0): 'camera_trigger_canon_eos_capture (camera, context)' failed: 'Unknown error' (-110)
gp_camera_capture [gphoto2-camera.c:1340](0): 'camera->functions->capture (camera, type, path, context)' open failed: -110
-----

> gphoto2 --capture-preview
*** Error (-1: 'Unspecified error') ***
log:
ptp_usb_getresp             (2): Reading PTP_OC 0x9153 (EOS_GetViewFinderData) response...
ptp_usb_getpacket           (2): Returning previously buffered response packet.
ptp_usb_getresp [usb.c:514] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
camera_capture_preview [library.c:3178](0): 'ret' failed: get_viewfinder_image failed (0xa102: Camera Not Ready)
gp_camera_capture_preview [gphoto2-camera.c:1404](0): 'camera->functions->capture_preview (camera, file, context)' failed: -1

[capture-image.txt](https://github.com/gphoto/libgphoto2/files/4924791/capture-image.txt)
[capture-preview.txt](https://github.com/gphoto/libgphoto2/files/4924793/capture-preview.txt)

gvogler, 2021-09-17:
Hi!

I am seeing the same errors with the SX530.

I am using these versions: gphoto2   2.5.27.1 // libgphoto2      2.5.22 // libgphoto2_port 0.12.0

The camera is detected as follows:
Model                          Port                                            
----------------------------------------------------------
USB PTP Class Camera           usb:001,008  

Error after capture reads as previously mentioned:

`
gphoto2 --capture-image
                                                                               
*** Error ***              
Canon EOS M Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-110: 'I/O in progress') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image

Please make sure there is sufficient quoting around the arguments.
`

PJF16, 2023-03-17:
See here: https://github.com/gphoto/libgphoto2/issues/334#issuecomment-1467894827

pufik, 2023-11-09:
Hi All. Any ideas how to overcome issue. I just connected Canon SX530 HS and got the same error. Maybe I can help to debug issue. I tried to restart system and do call with wait-event, but it really doesn't work. When command is send camera is reacting on it via turning off display, but then error is returned.
My gphoto versions:
`This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.27         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.27         standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking`

pufik, 2023-11-10:
> please update libgphoto2 to 2.5.31 first and retest. the relevant code got updated a lot of times since 2.5.27.
I have tried on new version:
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.30         standard camlibs, gcc, no ltdl, EXIF
libgphoto2_port 0.12.1         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking

Also i have updated firmware for camera to 1.1.0.0 latest available on Canon site:

I will update  libgphoto2 to 2.5.31 and back to you with result. Btw, Canon EOS M200 was working great...

Meanwhile my logfile:
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/13318369/my-logfile.txt)

pufik, 2023-11-10:
> please update libgphoto2 to 2.5.31 first and retest. the relevant code got updated a lot of times since 2.5.27.

I have tested with latest release. Please see log. Thanks in advance! If you want me to debug or test something then let me know  how can I help.
[canon_sx530_hs.txt](https://github.com/gphoto/libgphoto2/files/13318573/canon_sx530_hs.txt)

mvmn, 2023-11-10:
Same issue here, using liggphoto2 2.5.31 (installed via Homebrew with gphoto2 - https://formulae.brew.sh/formula/gphoto2 )

The camera seems to detect USB connection and switches off all capture functions. The lens is retracted, and the screen is turned off. One can get config, read files off the camera etc - but no captures work.

I thought that setting EOS Viewfinder property to 1 might fix this - as it does on some DSLRs. But the property won't save (you set it to 1, read config back - and it's still zero) and thus doesn't have any effect.

mvmn, 2023-11-12:
I have been able to work around the issue by using CHDK (Canon Hack Development Kit) firmware customization, which allowed me to keep the camera in a shooting (record) mode rather than file transfer (playback) mode when connected to a PC.

Now I have different issue - the capture-preview output for same camera differs on a MacBook Pro, where it is a JPEG, and on a Raspbery Pi, where it is a PBM (Portable Bitmap Image). 

Is there a way to force JPEG? Why is it even different? Not sure where I should ask these questions though, as for this particular issue they seem to be off topic. But I would appreciate any help.

flipphillips, 2024-02-14:
Same error, PowerShot sx70 hs, everything up to date.

[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/14289162/my-logfile.txt)

bugggaman, 2024-07-08:
> I have been able to work around the issue by using CHDK (Canon Hack Development Kit) firmware customization, which allowed me to keep the camera in a shooting (record) mode rather than file transfer (playback) mode when connected to a PC.
> 
> Now I have different issue - the capture-preview output for same camera differs on a MacBook Pro, where it is a JPEG, and on a Raspbery Pi, where it is a PBM - Portable Bitmap Image, or PPM - Portable PixMap, not exactly sure which one it is.
> 
> Anyhow, is there a way to force JPEG? Why is it even different? Not sure where I should ask these questions though, as for this particular issue they seem to be off topic. But I would appreciate any help.
> 
> UPD: to answer my own question about the format of preview - turns out the version of gphoto2 I had on a macOS was compiled with a `jpeg` option, but the one I had on Raspberry Pi wasn't.
> 
> ```
> This version of gphoto2 is using the following software versions and options:
> gphoto2         2.5.28         clang, popt(m), exif, no cdk, no aa, jpeg, readline
> ```
> 
> vs
> 
> ```
> This version of gphoto2 is using the following software versions and options:
> gphoto2         2.5.28.1       gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
> ```
> 
> UPD2: Hmm, now I have jpeg there - but still preview is captured in PPM and feeding capture-movie into ffmpeg fails with error `Unhandled MIME type 'image/x-portable-pixmap'`. Posted this issue as [gphoto/gphoto2#614](https://github.com/gphoto/gphoto2/issues/614)

How did you customize the firmware in CHDK? I haven't seen much documentation on this and I have the same camera as you and struggling with the USB connection.

mvmn, 2024-07-08:
> How did you customize the firmware in CHDK? I haven't seen much documentation on this and I have the same camera as you and struggling with the USB connection.

I simply installed CHDK onto the camera, @bugggaman

bugggaman, 2024-07-08:
> > How did you customize the firmware in CHDK? I haven't seen much documentation on this and I have the same camera as you and struggling with the USB connection.
> 
> I simply installed CHDK onto the camera, @bugggaman

I understand that, I have CHDK on my camera as well. What I am trying to do, is keep the camera connected in shooting mode while attached to USB. I am assuming you have some script that prevents playback mode from turning on when the USB is plugged in, or something along those lines. Can you share that script? @mvmn

mvmn, 2024-07-08:
I didn't use any scripts or any customizations - for the cameras I tried (SX530) it worked by simply installing CHDK onto it

bugggaman, 2024-07-08:
Did you change any settings in the CHDK menus? @mvmn