gphoto/libgphoto2 #523: set-config does not work for RADIO values on newer Sony cameras.

chubun, 2020-06-15:
Camera: Sony Alpha 7R4 

I also see a number of similar posts for other Sony cameras with the same problem.  The enumerate values for some configurations of RADIO type on newer Sony cameras are not defined.  For example, from the summary.txt (attached):

ISO(0xd21e):(readwrite) (type=0x6) **Enumeration []** value: 200
Exposure Bias Compensation(0x5010):(readwrite) (type=0x3) **Enumeration []** value: 0.0 stops (0)

As a result, the error below is shown when we tried to set these values in our code using set_config_value_string:
gplib> ErrContext: Target value is not in enumeration.

Any suggestion how to solve or work around this problem?  Thanks.

[summary.txt](https://github.com/gphoto/libgphoto2/files/4783047/summary.txt)

msmeissn, 2020-06-21:
Hmm, the camera seems to have its ISO values in 0xd226 and not 0xd21e ... I added code to help here

msmeissn, 2020-06-21:
and exp copensation seems better in 0xd224

msmeissn, 2020-06-21:
can you try current libgphoto2 git master  and see if it works?

chubun, 2020-06-22:
I'll try it tomorrow and let you know.  Thank you for a quick response.

chubun, 2020-06-22:
The updated version seems to show the correct enumeration.  But there are two entries for iso in the summary.txt

ISO(0xd21e):(readwrite) (type=0x6) Enumeration [] value: 102400
ISO(0xd226):(readwrite) (type=0x6) Enumeration [16777215,50,64,80,100,125,160,200,250,320,400,500,640,800,1000,1250,1600,2000,2500,3200,4000,5000,6400,8000,10000,12800,16000,20000,25600,32000,40000,51200,64000,80000,102400] value: 102400

And there is a new error:

0.000622 load_settings               (2): Creating gphoto config directory ('/home/installer/.gphoto')
0.000710 load_settings               (2): Loading settings from file '/home/installer/.gphoto/settings'.
0.000908 main                        (2): Uploading file...
*** Error (-1: 'Unspecified error') ***       

Below is the content of the settings file:

gphoto2=model=Sony DSC-A7r IV (Control)
gphoto2=port=usb:004,002
[debug.txt](https://github.com/gphoto/libgphoto2/files/4815365/debug.txt)
[summary.txt](https://github.com/gphoto/libgphoto2/files/4815366/summary.txt)

msmeissn, 2020-06-23:
what does 
gphoto2 --get-config iso
show ?

msmeissn, 2020-06-23:
you tried 
gphoto2 set-config iso=30 
while it should be
gphoto2 --set-config iso=30 
which caused this upload error you repored

chubun, 2020-06-23:
Sorry for the silly mistake.  It works much better with the right command.  Once in a while the value will not change after the command returns.  Attached are the debug for normal case (debug_OK.txt) and for the failed case (debug_noChange.txt).
[debug_noChange.txt](https://github.com/gphoto/libgphoto2/files/4820696/debug_noChange.txt)
[debug_OK.txt](https://github.com/gphoto/libgphoto2/files/4820697/debug_OK.txt)

msmeissn, 2020-06-25:
hmm , i had a small bug in my code ... can you update libgphoto2 git again and see if it works better? 
It might also not work, in both cases a debug trace would be appreciated

chubun, 2020-06-25:
I probably won't have time till the weekend.  Will try as soon as I can.

chubun, 2020-06-26:
This version seems to have more problems.
First, the type now is "Menu" instead of "Radio".
$>gphoto2 --get-config ISO 
Label: ISO                                                                     
Readonly: 0
Type: MENU
Current: 40000
END

Second, if I change the ISO value using the camera control, the first get-config will return the previous value (attached previousISO.dbg).  Then if I run get-config again, it will return the current value (correctISO.dbg).

Third, --set-config won't work at all (setISOfailed.dbg).

By the way, this is on a Raspberry PI.

[setISOfailed.txt](https://github.com/gphoto/libgphoto2/files/4834810/setISOfailed.txt)
[correctISO.txt](https://github.com/gphoto/libgphoto2/files/4834811/correctISO.txt)
[previousISO.txt](https://github.com/gphoto/libgphoto2/files/4834812/previousISO.txt)
[summary.txt](https://github.com/gphoto/libgphoto2/files/4834813/summary.txt)

msmeissn, 2020-06-26:
you need to lowercase "iso" ... it uses a entire different valuesetter if you use "ISO" ...

chubun, 2020-06-26:
No idea why I typed use cap.  With ISO in lower case, set-config return -110 for iso.  I can set f-num without any problem, and also power cycle the cam, so it's not like that the device is really busy.  Attached is the new log.
[devbusy.txt](https://github.com/gphoto/libgphoto2/files/4837984/devbusy.txt)

msmeissn, 2020-06-27:
I added some more changes to git... can you try?

chubun, 2020-06-29:
Both set-config and get-config have problems.  

set-config seems to able to set the iso, but the value won't stick.  For example, if the cam is set to M mode, then set-config is run with a low value, the preview screen will go dark for a moment, but then back to normal (debug file: setiso_M.txt).
[setiso_M.txt](https://github.com/gphoto/libgphoto2/files/4847667/setiso_M.txt)

For get-config, if I change the iso value on the cam, the first get-config will report the previous value (debug file: getiso_previous.txt).  Subsequent get-configs will reports the correct value (debug file: getiso_current.txt)
[getiso_current.txt](https://github.com/gphoto/libgphoto2/files/4847690/getiso_current.txt)
[getiso_previous.txt](https://github.com/gphoto/libgphoto2/files/4847691/getiso_previous.txt)

msmeissn, 2020-06-30:
with the ZV-1 here it seems to stick after setting.

I see a slight delay on being it available with get iso , but it is quite quick here...

chubun, 2020-06-30:
I tested on two different cameras on Ubuntu desktop and Raspberry Pi.  The results are the same.  Two get-config's to show the correct value (for example, after recycling power on the camera, the first get will always show "Auto ISO", and the second get will show the current value).  set-value will cause the preview screen to flicker, but the iso value stays the same.  Let me know if  you want to collect anymore logs.

chubun, 2020-12-30:
It has been a while, but I have a chance to try the latest version.  Both iso and exposurecompensation (ev) can be set but not reliably.  It seems something need to be properly initialized before iso and ev can be set.  For sample, if I connect the camera and immediately try to get/set-config the output will look like the following:

==============
SET: Note: property is read-only.

gphoto2 --set-config exposurecompensation=0
Loading camera drivers from '/usr/l... |                     \   0.0%  
       ...
Loading camera drivers from '/usr/l... |-------------------- /  94.7%                                                                                        
*** Error ***              
Property exposurecompensation is read only.
*** Error (-1: 'Unspecified error') ***       
...

=============
GET: Note: there are no choices for RADIO 

gphoto2 --get-config exposurecompensation
Label: Exposure Compensation                                                   
Readonly: 1
Type: RADIO
Current: -0.7
END

**********
********** However, after I play with getting/setting other properties (f-number, etc.), the commands work fine.
**********

=============
SET:

gphoto2 --set-config exposurecompensation=1
(no output, but the correct ev value shown in the camera display)

=============
GET: (Note: choices are listed)

gphoto2 --get-config exposurecompensation
Label: Exposure Compensation                                                   
Readonly: 0
Type: RADIO
Current: 0
Choice: 0 5
Choice: 1 4.7
...
Choice: 30 -5
END

msmeissn, 2021-01-03:
what gphoto2 version do you have currently? 

I added some initial wait code that might help here.

a debuglogfile might be good again

qtnemo, 2021-01-05:
Software versions:
gphoto2                2.5.26.1
libgphoto2            2.5.26.1
libgphoto2_port    0.12.0

Log file attached.
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/5771436/my-logfile.txt)

qtnemo, 2021-01-15:
By the way in the official Sony SDK for arm/windows, a call to "Set Key Position" with value 2 (PC Remote) (default value is 1 which means using Camera setting) will fix the "property is read-only" problem.

msmeissn, 2021-01-16:
any idea which config value this set key positioon this presents?

qtnemo, 2021-01-18:
Can I get into troubles posting info obtaining from reading someone's copyrighted code?  Is this considered reverse engineering?

msmeissn, 2021-01-19:
(not a lawyer, so I cant say)... if you just extrct the protocol info i think we are okayish ;)

Entropy512, 2021-02-23:
@qtnemo a Wireshark trace of USB when you fire off "Set key position" via the API would be useful here and I don't think it would be illegal.

@msmeissn - that "set key position" might be beneficial for #438 and other use cases involving switching between file transfer and normal mode.  I'll need to run some tests with Sony's official tools to see if it can pop the camera back into PC Control mode after doing transfers.

Unfortunately I only own an A7M3, so can't actually use the new Sony SDK to sniff traffic.

msmeissn, 2021-02-25:
yeah if you capture a usbtrace with e.g. usbpcap I can take a look.

msmeissn, 2021-03-14:
i am sending prioritymode = 1 now to the camera, can you check if this helps (with current git?)

msmeissn, 2022-01-04:
hopefully fixed, please reopen if not

closed 2022-01-04 by msmeissn