gphoto/libgphoto2 #525: Fuji X-A3

q8diver, 2020-06-18:
Fill in fields below. If you don't know how, leave it free.

Fuji X-A3

Use the Marketing name.

USB IDs
e.g. by running: lsusb

**Bus 001 Device 003: ID 04cb:02ce Fuji Photo Film Co., Ltd **
run

gphoto2 --summary > summary.txt

[Camera summary:
Manufacturer: FUJIFILM
Model: X-A3
Version: 2.21
Serial Number: 592D353332303116101068156WL30016
Vendor Extension ID: 0xe (1.0)
Vendor Extension Description: fujifilm.co.jp: 1.0;

Capture Formats: Apple Quicktime
Display Formats: JPEG

Device Capabilities:
File Download, File Deletion, File Upload
No Image Capture, No Open Capture, No vendor specific capture

Storage Devices Summary:
store_00010001:
StorageDescription: External Memory
VolumeLabel: 000000000000000000000000000000000000000000000000
Storage Type: Removable RAM (memory card)
Filesystemtype: Digital Camera Layout (DCIM)
Access Capability: Read Only with Object deletion
Maximum Capability: 31897681920 (30420 MB)
Free Space (Bytes): 31897632768 (30419 MB)
Free Space (Images): -1

Device Property Summary:
Battery Level(0x5001):(read only) (type=0x2) Range [0 - 3, step 1] value: 1% (1)
White Balance(0x5005): error 2007 on query.
F-Number(0x5007): error 2007 on query.
Focus Mode(0x500a): error 2007 on query.
Property 0xd407:(read only) (type=0x6) 1
Property 0xd406:(readwrite) (type=0xffff) ''
Property 0xd303:(read only) (type=0x2) 1
Exposure Time(0x500d): error 200a on query.
F-Number(0x5007): error 2007 on query.
Property 0xd38c: error 200a on query.
Property 0xd171: error 200a on query.
Property 0xd21c: error 200a on query.
Focus Point(0xd347): error 200a on query.
]

camera configuration output

gphoto2 --list-all-config > list-all-config.txt

and attach list-all-config.txt here

/main/actions/opcode
Label: PTP Opcode
Readonly: 0
Type: TEXT
Current: 0x1001,0xparam1,0xparam2
END
/main/status/serialnumber
Label: Serial Number
Readonly: 0
Type: TEXT
Current: 592D353332303116101068156WL30016
END
/main/status/manufacturer
Label: Camera Manufacturer
Readonly: 0
Type: TEXT
Current: FUJIFILM
END
/main/status/cameramodel
Label: Camera Model
Readonly: 0
Type: TEXT
Current: X-A3
END
/main/status/deviceversion
Label: Device Version
Readonly: 0
Type: TEXT
Current: 2.21
END
/main/status/vendorextension
Label: Vendor Extension
Readonly: 0
Type: TEXT
Current: fujifilm.co.jp: 1.0;
END
/main/status/batterylevel
Label: Battery Level
Readonly: 1
Type: TEXT
Current: 50%
END
/main/other/5001
Label: Battery Level
Readonly: 1
Type: MENU
Current: 1
Choice: 0 0
Choice: 1 1
Choice: 2 2
Choice: 3 3
END
/main/other/d407
Label: PTP Property 0xd407
Readonly: 1
Type: TEXT
Current: 1
END
/main/other/d406
Label: PTP Property 0xd406
Readonly: 0
Type: TEXT
Current:
END
/main/other/d303
Label: PTP Property 0xd303
Readonly: 1
Type: TEXT
Current: 1
END

test if capture perhaps already works:

gphoto2 --capture-image-and-download
*** Error ***
Sorry, your camera does not support generic capture
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***

gphoto2 --capture-preview
*** Error (-1: 'Unspecified error') ***

hfiguiere, 2020-06-18:
It's probably the same case as #484 (Fuji X-E3)

msmeissn, 2020-06-19:
Yes, to our current knowledge it does not support remote control over USB.

(Someone is trying to implement it over Wifi, where it _might_ be possible.)

closed 2020-06-19 by msmeissn