gphoto/libgphoto2 #528: 1000 images issue

dungcules, 2020-06-24:
I do have some issue with my company special camera which is a ptp usb camera. I am using libgphoto2 to test the limitation of the camera. There are two issue but both of them are related to the limitation of 1000 images. 
First issue happens when I take images using "gp_camera_capture" command to capture in my program. Since the image number reach 1000, the camera got disconnected and not recognized by my Coral board.
Second issue is when I use "gphoto2 --list-files", it just list 1000 items inside a folder even though I copied more than 1000 images into the SD card.
I am trying to determine if the problem comes from the camera itself or from the libgphoto2 library. I do not have experience working with the file system of the camera or gphoto2 library before. Can you give me some clue about the cause of these problems?

msmeissn, 2020-06-25:
Usually if the camera crashes ... it is the camera side fault ... gphoto also has no 1000 file limit.

You can enable debugging on gphoto2 side, by adding --debug --debug-logfile=debug.log  to the commands and see if something weird is happening there.

dungcules, 2020-06-25:
Thanks you for your quick respond.

I did add the debug option into the gphoto2 list command and the result is following:
[debug_list_file.log](https://github.com/gphoto/libgphoto2/files/4832091/debug_list_file.log)

I have one folder inside the SD card (DCIM/100_GEOC) and there are 1365 files in this folder. However, the gphoto2 just list 1000 items only (on Linux machine). I also plug the camera directly into my Window machine and it also list 1000 items. Do you think that my camera has limitation or problem?

Also, I just found another issue when my camera use trigger-capture. It work fine at the first capture but it returns busy state and cannot recover until I re-plug the cable. The log file is attached below. The command capture-image  work fine although it has a little delay but I think I can use it.
[debug_second_trigger_capture.log](https://github.com/gphoto/libgphoto2/files/4832175/debug_second_trigger_capture.log)

dungcules, 2020-06-30:
I found the problem on camera itself since I used another camera app to test the camera capture and it stills has the same issue after capturing 1000 images. However, I want to find a solution to work around this problem while waiting for the respond from the camera manufacturer. What I did is capturing, download the image and then deleted it. This temporary solution allowed me to capture more than 1000 images but the delay time 

Do you have any idea about the problem of the camera, Marcus? If you have other solutions to fix this issue, I would like to try them.

msmeissn, 2020-06-30:
no, i have no other solutions.
if this would be a per-directory limitation you could have the camera switch directories every 100 images  perhaps?

dungcules, 2020-11-12:
I found that this camera and SandDisk Card are not compatible. The other branch SD cards work fine and get over this limitation.

Thank you for your supporting!!!

closed 2020-11-12 by dungcules