gphoto/libgphoto2 #529: Supported Camera URL on the readme.md is dead

Mirppc, 2020-06-25:
In the readme.md there is a link leading to what is a supported camera lisit.  I was hoping it would be more up to date than darktable's or the project site's as none of my camera's seem to be supported (that issue is still being tested to rule out PEBKAC)

So the readme says this:

Then, there are cameras supporting the so-called USB Mass Storage protocol. This is a protocol that has been published and lets you access any storage device, be it a camera or a disk connected via USB to your computer. As there are already drivers for this protocol out there, you don't need an additional library like libgphoto2. The list of camera that use USB Mass Storage is getting longer everyday, so we won't publish it.

For a more up to date list, you can consult *camera list with support status.*  

In there the "camera list with support status." is hot linked to http://www.teaser.fr/~hfiguiere/linux/digicam.html  which happens to be a dead url.

msmeissn, 2020-06-25:
What cameras do you have? 
While above url is not functional anymore, we have both the support list http://gphoto.org/proj/libgphoto2/support.php and the http://gphoto.org/doc/remote/ remote list...

Mirppc, 2020-06-25:
Ahh, thanks for the response.  I posted this mainly because a dead url in a readme is not really great.  I am still debugging why multiple digital camera's do not show up with gphoto2 --auto-detect.  I still need to rule out PEBKAC and possibly the USB cable.

Hopefully the hotlinked URL can be fixed in the readme to something more up to date.  Possibly the ones you linked @msmeissn .

As for Camera issue i will post another bug report when i have enough information to make a concise and usable report.

msmeissn, 2020-06-25:
you can use the lowerlevel tool "lsusb" to see if they show there...

Mirppc, 2020-06-25:
there is already a thread for one of my camera's.  I will post there for my main camera's soon.  This post was for the dead url link.

hfiguiere, 2020-06-25:
> In the readme.md there is a link leading to what is a supported camera lisit. I was hoping it would be more up to date than darktable's or the project site's as none of my camera's seem to be supported (that issue is still being tested to rule out PEBKAC)
>
> In there the "camera list with support status." is hot linked to http://www.teaser.fr/~hfiguiere/linux/digicam.html which happens to be a dead url.

This is issue #475 and #481, and already fixed on Feb 24, part of 2.5.25.

Mirppc, 2020-06-25:
> This is issue #475 and #481, and already fixed on Feb 24, part of 2.5.25.

That is incorrect.  Looks like the link was removed 9 hours ago as of me posting this.  no new link was put in it's place but at lead the link to a dead site is gone.

*For a more up to date list, you can consult [camera list with support status].*

README.md
removed the old list, replaced by pointers to our lists
9 hours ago

I am considering this good enough and closing.  Thank you @msmeissn .

closed 2020-06-25 by Mirppc