gphoto/libgphoto2 #530: Gphoto2 writes then deletes images.

goemonburo, 2020-06-26:
I don't know what's going on but I'm using:

    gphoto2 --capture-image-and-download

And each time it gives me a note that it's "new file in /store_blahblah/DCIM/100EOS7D/IMG_Blah.CR2 on the camera", then "saving file as XYZ.CR2" then (why???!!) "Deleting file in /store_blahblah/DCIM/100EOS7D/IMG_Blah.CR2 on the camera."

It does this for both the CR2 and the JPG and I'm unable to ever see any image.  I used the "--set-config capturetarget=1" flag to (presumably) set it to save to the card...why is it deleting?!

Thank you!

msmeissn, 2020-06-26:
If you want to both download and keep them, use the --keep and --keep-raw options

if you do not want to download, just keep on card, use --capture-image  instead

goemonburo, 2020-06-26:
Thank you for the comment.  So...curious... "--capture-image-and-download" takes an image and then doesn't save it to either the computer OR the card?  That's what's apparently happening.  It doesn't seem to be storing the image in either place.  (Or I'm not finding it?)

msmeissn, 2020-06-26:
No, it saves it to the current directory where you are running gphoto2 from.

goemonburo, 2020-06-26:
Thank you again, Msmeissn.  I appreciate the answer -- and found the images tucked into the folder where I was running it from, just as you said.  

For video, do you know how the audio is captured? I've tried a mic into the camera itself, I've tried my hw:0:0 option, don't seem to have audio at all.  (Sorry, this would be a different issue, should I repost??)

closed 2020-06-26 by goemonburo

msmeissn, 2020-06-27:
with --capture-movie, it will only capture preview images , not sound.

you can do movie capture with the methods in http://gphoto.org/doc/remote  , this will record the movie on the camera in h264 or what the camera uses ... this will just mix the camera microphone into the video.