gphoto/libgphoto2 #531: Could not list folders in '/'.  An error occurred in the io-library ('Bad parameters'): No error description available

goemonburo, 2020-06-26:
**Describe the bug**

Every time I start gtkam, I get the following error message which needs to be clicked out before the program works.

"Could not list folders in '/'.

An error occurred in the io-library ('Bad parameters'): No error description available"

This is not a show stopper, but I would like to know what's causing this (if possible to find out) so I can just have Gtkam work normally.

**Name the camera**
Canon EOS 7D                   usb:001,022
 
**libgphoto2 and gphoto2 version**
output of: gphoto2 --version
gphoto 2.5.23
libgphoto2 2.5.25

**To Reproduce**
Steps to reproduce the behavior:

Type "gtkam" at the command line.

msmeissn, 2020-06-27:
gtkam might have old camrea information and trying to access it ... can you remove .gphoto/settings and see if this helps?

goemonburo, 2020-06-27:
That fixed it, yes.  Will it throw that each time I switch cameras then?  Like if I'm using one camera and then switch to a different one and there's the other camera still showing there in the list?

closed 2020-06-27 by goemonburo