gphoto/libgphoto2 #537: Sigma FP support

msmeissn, 2020-07-03:
Fill in fields below. If you don't know how, leave it free.

**Name of the camera*

Sigma FP  (Full Frame format system camera)

Received via user request e-mail.

Need research if possible at all.

https://www.dpreview.com/forums/thread/4500015

msmeissn, 2020-07-03:
https://www.sigma-global.com/en/cameras/fp-series/download/sdk/

qtnemo, 2021-04-05:
Any new development on the Sigma fp camera?
list-config shows only half a dozen entries such as manufacture, serial number, ... but none of the configurations (f-number, iso, ...).  Capture and Trigger show command not supported.  Attached are the log for list-config, capture, and trigger.
If I want to add the support myself, any pointer on where to start (ptp.c, ptp-pack.c, ...)?  

[trigger.log](https://github.com/gphoto/libgphoto2/files/6258885/trigger.log)
[capture.log](https://github.com/gphoto/libgphoto2/files/6258886/capture.log)
[listconfig.log](https://github.com/gphoto/libgphoto2/files/6258887/listconfig.log)

msmeissn, 2021-04-05:
well, mostly some documentation on the vendor ptp opcodes.

if you have those, i can work on it.

msmeissn, 2021-04-07:
i also tried mailing support@sigma... for doc this time, so far no reply, not even auto.

qtnemo, 2021-04-07:
Same here.  Still waiting for a reply from them.

msmeissn, 2021-04-10:
I rented one over the weekend and try to get some things in

msmeissn, 2021-04-10:
Status:

- liveview works
- simple capture works barely, picture is returned, but from earlier captures ... need to work on this more

qtnemo, 2021-04-12:
Thank you so much for the effort.

msmeissn, 2021-04-12:
so, camera returned ... 
- capture is not very reliable, but working.
- liveview seems reliable.

configuration is not implemented, thiis will be a bit longer effort

chubun, 2021-04-13:
Please let us know if we can help in anyway such as paying for the camera rental, etc.

flavioperucchi, 2023-03-27:
What is the current status of the Sigma fp support? 
Do you know if Sigma fp use normal PTP standard ISO15740?

msmeissn, 2023-03-27:
fwiw not much changes since my last updates, since i did not rent it again yet. (currently also have no time to do much work :( )

pianhe, 2023-07-29:
> fwiw not much changes since my last updates, since i did not rent it again yet. (currently also have no time to do much work :( )

Perhaps because of the firmware upgrade, the most basic functions are now useless for sigma fp @msmeissn

Damani2, 2024-01-05:
Came here to join in frustration. I want to use Darktable tethered which uses a gphoto2 dependency. I can't get any of the camera settings to apply for shutter speed, aperture, or iso. I can just trigger a photo. any advice on how to fix this? I'd rather use Darktable than Capture one.

Damani2, 2024-01-05:
I have the FP L

hfiguiere, 2024-01-05:
the frustration is the camera vendor lack of support. You gave them money after all.

Damani2, 2024-01-05:
> the frustration is the camera vendor lack of support. You gave them money after all.

What is it that we need from them? Because they provide a sample sdk application. I want to help push this forward.

hfiguiere, 2024-01-05:
if you read the history of the issue, Marcus doesn't have the camera. He has to spend money to rent it to do Sigma's work, and that's not even a guarantee that things can be resolved. "sample sdk application" likely require sigma proprietary code that doesn't run on Linux and that may not even be useful.

jmathew58, 2024-04-19:
I could also create a development set up with the Sigma FP connected to my Mac via USB and access via anydesk if someone wanted to use my computer.   But only if that can RESULT in an android or iOS app.

jmathew58, 2024-04-20:
> > the frustration is the camera vendor lack of support. You gave them money after all.
> 
> What is it that we need from them? Because they provide a sample sdk application. I want to help push this forward.

I would also like to push the possibility of an iOS or Android app that will Connect to the camera via USB-C.
Am willing to fund such a project.