gphoto/libgphoto2 #538: D500

dufd2304, 2020-07-03:
**Describe the bug**

Image type aliases for Nikon D500 are wrong; setting specific file type (different quality of JPGs, NEF, TIF or NEF+JPG combination) will result in a different setting in the camera and a different file format captured. 

This results in abnormal behaviors in downstream software relying on libgphoto2 to acquire specific file formats. Specifically, using EKOS/INDI to capture images in RAW format, EKOS/INDI won't allow a NEF + JPEG selection, therefore disallowing choices 5, 6 and 7. Choices 8-12 are allowed but client software will randomly download either the NEF or the JPEG acquired, and anything other than NEF/RAW is 

**Name the camera**
Model                          Port                                            
----------------------------------------------------------
Nikon DSC D500                 usb:001,011     

**libgphoto2 and gphoto2 version**
gphoto2 2.5.20
libgphoto2 2.5.24.1
libgphoto2_port 0.12.0

**To Reproduce**

_gphoto2 --get-config imagequality
Label: Image Quality                                                           
Readonly: 0
Type: RADIO
Current: NEF+Normal
Choice: 0 JPEG Basic
Choice: 1 JPEG Normal
Choice: 2 JPEG Fine
Choice: 3 Unknown value 0003
Choice: 4 NEF (Raw)
Choice: 5 NEF+Basic
Choice: 6 NEF+Normal
Choice: 7 NEF+Fine
Choice: 8 Unknown value 0008
Choice: 9 Unknown value 0009
Choice: 10 Unknown value 000a
Choice: 11 Unknown value 000b
Choice: 12 Unknown value 000c
Choice: 13 Unknown value 000d
END_

gphoto2 --set-config imagequality=4 && gphoto2 --capture-image-and-download results in a large format JPEG file (see correspondance below) and gphoto2 --set-config imagequality=7 && gphoto2 --capture-image-and-download is required to acquire a NEF file, even though gphoto2 --set-config imagequality=7 && gphoto2 --get-config imagequality suggests the selection to be NEF+Fine.
[debug.log](https://github.com/gphoto/libgphoto2/files/4871450/debug.log)

Trial and error capture at different settings (gphoto2 --set-config imagequality=0 && gphoto2 --capture-image-and-download && (...) until imagequality=13) shows it should be the following:
_gphoto2 --get-config imagequality
Label: Image Quality                                                           
Readonly: 0
Type: RADIO
Current: NEF+Normal
Choice: 0 JPEG Basic
Choice: 1 JPEG Basic*
Choice: 2 JPEG Normal
Choice: 3 JPEG Normal*
Choice: 4 JPEG Fine
Choice: 5 JPEG Fine*
Choice: 6 TIF
Choice: 7 NEF (Raw)
Choice: 8 NEF + JPEG Basic
Choice: 9 NEF + JPEG Basic*
Choice: 10 NEF + JPEG Normal
Choice: 11 NEF + JPEG Normal*
Choice: 12 NEF + JPEG Fine
Choice: 13 NEF + JPEG Fine*
END_

dufd2304, 2020-07-09:
Addendum:
Modifying the camlibs/ptp2/config.c entry for generic Nikon compression and recompiling fixes this specific problem, but this is a bastardly/inelegant way of doing it and will break every other unreferenced Nikon device. Adding a specific Nikon_D500_Compression or referencing the Nikon_***_Compression of a device that has the same aliases would work, but my coding skills are not to the level that I can confidently do it while being sure I'm not breaking something else.

msmeissn, 2020-07-09:
I have vendor subtables for Nikon D series for these cases... look e.g. for the D90 tree or so

nikon_d90_capture_settings 

there we have already these entries:

        { N_("Image Quality"),                  "imagequality",         PTP_DPC_CompressionSetting,     PTP_VENDOR_NIKON,   PTP_DTC_UINT8,  _get_Nikon_D90_Compression,     _put_Nikon_D90_Co

check if one of threse tables can also be used already for the D500 perhaps

msmeissn, 2020-07-09:
( i will not have time before the weekend to synxc with the SDK tables, sorry ... my work week is just too busy )

dufd2304, 2020-07-09:
The D850 seems to be spot on (lots of internal similarities with the D5 and the D500, so no surprise there).
No rush, my bastard fix makes it functional enough for 30s and less exposure for the time being, and I'm working on fixing my issue with bulb capture in the meantime (will check if a similar bug report was filed, and file one end of day if not - in fact, may just try and recompile using the whole D850 config and see if it fixes the bulb capture as well)

msmeissn, 2020-07-12:
The SDK of the D500 says it has a smilar list for compression setting than the D90, I added this.

dufd2304, 2020-07-14:
"The SDK of the D500 says it has a smilar list for compression setting than the D90, I added this."
It's almost the same, but D90 is lacking the tiff and the higher compression quality JPGs (or else libgphoto has D90's compression aliases wrong).
This is straight out of the D500 SDK (and corresponds to what my camera has in settings):
0 JPEG (BASIC)
1 JPEG (BASIC★)
2 JPEG (NORMAL)
3 JPEG (NORMAL★)
4 JPEG (FINE)
5 JPEG (FINE★)
6 TIFF (RGB)
7 RAW
8 RAW+JPEG (BASIC)
9 RAW+JPEG (BASIC★)
10 RAW+JPEG (NORMAL)
11 RAW+JPEG (NORMAL★)
12 RAW+JPEG (FINE)
13 RAW+JPEG (FINE★)
Whereas, per libgphoto2/camlibs/ptp2/config.c, D90 would have the following:
static struct deviceproptableu8 nikon_d90_compression[] = {
	{ N_("JPEG Basic"),	0x00, PTP_VENDOR_NIKON },
	{ N_("JPEG Normal"),	0x01, PTP_VENDOR_NIKON },
	{ N_("JPEG Fine"),	0x02, PTP_VENDOR_NIKON },
	{ N_("NEF (Raw)"),	0x04, PTP_VENDOR_NIKON },
	{ N_("NEF+Basic"),	0x05, PTP_VENDOR_NIKON },
	{ N_("NEF+Normal"),	0x06, PTP_VENDOR_NIKON },
	{ N_("NEF+Fine"),	0x07, PTP_VENDOR_NIKON },
This one lacks the TIFF, as well as the high compression quality JPGs (the stars)
D850, on the other hand, is spot on:
static struct deviceproptableu8 nikon_d850_compression[] = {
	{ N_("JPEG Basic"),  0x00, PTP_VENDOR_NIKON },
	{ N_("JPEG Basic*"), 0x01, PTP_VENDOR_NIKON },
	{ N_("JPEG Normal"), 0x02, PTP_VENDOR_NIKON },
	{ N_("JPEG Normal*"),   0x03, PTP_VENDOR_NIKON },
	{ N_("JPEG Fine"),   0x04, PTP_VENDOR_NIKON },
	{ N_("JPEG Fine*"),   0x05, PTP_VENDOR_NIKON },
	{ N_("TIFF"),   0x06, PTP_VENDOR_NIKON },
	{ N_("NEF (Raw)"),   0x07, PTP_VENDOR_NIKON },
	{ N_("NEF+Basic"),   0x08, PTP_VENDOR_NIKON },
	{ N_("NEF+Basic*"),   0x09, PTP_VENDOR_NIKON },
	{ N_("NEF+Normal"),   0x0A, PTP_VENDOR_NIKON },
	{ N_("NEF+Normal*"),   0x0B, PTP_VENDOR_NIKON },
	{ N_("NEF+Fine"),   0x0C, PTP_VENDOR_NIKON },
	{ N_("NEF+Fine*"),   0x0D, PTP_VENDOR_NIKON },

Having recompiled camlibs/ptp2/config.c for the D500 to use the D850 table, I can attest that it works properly. Images are captured, and downloaded files are of the proper format - also, I can use gphoto2 to change the settings, and the camera's LCD shows the right setting. Still having no luck on bulb capture however (and yes, using the set-config bulb, wait, set-config bulb, wait and download; puts my camera in a bad state and can't do anything thereafter, trying a few things then filing another bug report).

msmeissn, 2020-07-19:
sorry my d90 comment was wrong, I had it adjusted to use the d850 list in the second commit.

msmeissn, 2020-07-19:
for bulb, it should work. a logfile will be helpful when you have one

jscheidtmann, 2021-02-14:
Hi msmeisen, is this part of the 2.5.26 release?

msmeissn, 2021-02-15:
yes, it is in 2.5.26

closed 2021-02-15 by msmeissn