gphoto/libgphoto2 #539: PTP I/O Error - Nikon J2

jordancormack, 2020-07-04:
I get the following error when I try to capture an image with my Nikon 1 J2:

```
*** Error ***              
PTP I/O Error
ERROR: Could not capture image.
ERROR: Could not capture.

*** Error ***              
PTP I/O Error
```

The camera appears as `Nikon J2` when I run `--auto-detect`, but when I try to run `--capture-image-` I get the above error after about 25-40 seconds.

Using my Canon 6D, `--capture-image-` works as expected and with no delay.

gphoto2/libphoto2 versions:
```
gphoto2         2.5.23.1       gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.25.1       standard camlibs (SKIPPING lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

On the camera compatibility page, it says that image capture is supported for the Nikon J2. I have attached the debug.log output.

[debug.log.txt](https://github.com/gphoto/libgphoto2/files/4873522/debug.log.txt)

jordancormack, 2020-07-06:
UPDATE:

I used https://github.com/gonzalo/gphoto2-updater to update to gphoto2 version 2.5.23.1 and libgphoto2 2.5.22. This lets me capture images with the Nikon J2, but still with a relatively large delay (approx. 20 seconds).

msmeissn, 2020-07-07:
We called some commands that were not available.

This is now fixed in libgphoto2 GIT master, will be in 2.5.26 
(the updater allows a developemtn install ... you can try that if you want)

closed 2020-07-07 by msmeissn