gphoto/libgphoto2 #541: connecting Nikon D5300 under Ekos failed and troubles with gphoto2

Patrick-81, 2020-07-11:
Hello,
I have made a fresh install of Kstars/Ekos/INDI on a Dell D430 running Ubuntu MATE 18.04LTS 32bits. I am using a D5300 to capture images. The DSLR is well recognized by the computer (lsusb), not mounter (doesn't appear mounted with lsblk).
On the advice of Jasem Mutlaq I checked gphoto2 with the following command. The DSLR is responsive but gave me an error.
**gphoto2 --debug --debug-logfile=gphoto.txt --capture-image-and-download**
I join here the log file (Hope the drag and drop will works.)
Best regards
Pat
[gphoto.txt](https://github.com/gphoto/libgphoto2/files/4906840/gphoto.txt)

msmeissn, 2020-07-11:
I see error 0xA200 , which is PTP_RC_NIKON_Bulb_Release_Busy

are you doing bulb capture?

msmeissn, 2020-07-11:
for bulb capture, use --set-config bulb=1 --wait-event=100s --set-config bulb=0
or gphoto2  --bulb=100s  --capture-image-and-download

Patrick-81, 2020-07-11:
<span style="font-family:arial,helvetica,sans-serif; font-size:12px">‌</span>

<div class="gl_quote" style="margin-top: 20px; padding-top: 5px;">De : "Marcus Meissner"<br>
A : "gphoto/libgphoto2" ,"Patrick-81" ,"Author"<br>
Envoyé: samedi 11 juillet 2020 17:36<br>
Objet : Re: [gphoto/libgphoto2] connecting Nikon D5300 under Ekos failed and troubles with gphoto2 (#541)<br>
 
<div class="gl_quoted">
<p>Thank you Marcus, this command wotks well. Maybe the problem is more on Indi driver side. Might it be possible Jasem ?<br>
Best regards<br>
Pat</p>

<p>for bulb capture, use --set-config bulb=1 --wait-event=100s --set-config bulb=0<br>
or gphoto2 --bulb=100s --capture-image-and-download</p>

<p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
</div>

<div class="gl_quoted"> </div>
</div>

knro, 2020-07-12:
@Patrick-81 Can you turn **FORCE BULB** in options to see if that makes any difference?

Patrick-81, 2020-07-12:
<span style="font-family:arial,helvetica,sans-serif; font-size:12px">He‌</span>

<div class="gl_quote" style="margin-top: 20px; padding-top: 5px;">De : "Jasem Mutlaq"<br>
A : "gphoto/libgphoto2" ,"Patrick-81" ,"Mention"<br>
Envoyé: dimanche 12 juillet 2020 07:56<br>
Objet : Re: [gphoto/libgphoto2] connecting Nikon D5300 under Ekos failed and troubles with gphoto2 (#541)<br>
 
<div class="gl_quoted">
<p>Hello<br>
There is no option FORCE_BULB in options. And as the Nikon doesn't connect the option pad remains empty.<br>
The version of the driver is 2.6. On my other PC version of the driver is 2.9. Is there a huge difference between 2.6 and 2.9 ?<br>
Regards</p>

<p><a class="user-mention" data-hovercard-type="user" data-hovercard-url="/users/Patrick-81/hovercard" data-octo-click="hovercard-link-click" data-octo-dimensions="link_type:self" href="https://github.com/Patrick-81">@Patrick-81</a> Can you turn <strong>FORCE BULB</strong> in options to see if that makes any difference?</p>

<p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
</div>

<div class="gl_quoted"> </div>
</div>

Patrick-81, 2020-07-13:
Hello Jasem,
I compiled indi and the indi-gphoto driver. And all works fine now. 
Don't know if the problem comes from the indi driver or the lib of 
gphoto with the legacy version of indi and drivers.
The compiled driver have the number 30, the legacy 26.
Best regards
Pat

msmeissn, 2022-01-04:
seems fixed

closed 2022-01-04 by msmeissn