gphoto/libgphoto2 #544: Nikon J4 cannot set parameters except ISO

simont77, 2020-07-20:
Hi,
I'm trying to use a Nikon J4  with ekos. I soon realized that I'm not able to use BULB (that is somehow expected), but also that I cannot set any exposure parameter except ISO. I then tried directly with gphoto using CLI, and it appears that many config parameters are readonly, and among these also shutterspeed2 (why no shutterspeed?) and f-number, whereas others, such as ISO, are read-write. Is this a limitation of the camera or a bug in libgphoto?
Thanks

simont77, 2020-07-21:
I made some other tests.
Expmode can be changed, but shutterspeed and f-number remains readonly even if a set expmode to M (either in camera or with gphoto).
Controlmode is readwrite and set to 0. I tried to set it to 1 with --set-config controlmode=1, the command issues no error, but controlmode remains to 0 if I read it back.

msmeissn, 2020-07-25:
Can you do both setting controlmode=1 and other variables in the same command?

libgphoto2 will switch it back to 0 when exiting.

simont77, 2020-07-25:
Hi, 
it does not work either, shutterspeed is always readonly. The current value for all properties is read correctly, though. Here is the log of the command.

[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/4975534/my-logfile.txt)

msmeissn, 2020-07-25:
hmm. can you attach gphoto2 --summary   output?

simont77, 2020-07-25:
Here is the summary and its logfile

[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/4975916/my-logfile.txt)
[summary.txt](https://github.com/gphoto/libgphoto2/files/4975917/summary.txt)

simont77, 2020-08-01:
Hi, any thought about this issue? Do you need some more test&log?

msmeissn, 2020-08-02:
I debugged 2 hours with my Nikon J5 which has the same issues ... I did not succeed in changing above so far.

We might be hit by camera side limitation

simont77, 2020-08-02:
That's unfortunate...it's very strange that one can change many exposure setting but not time and aperture. Is it possible to force a write even if the camera reports it is readonly? Can it be that real exptime and aperture to be used are one of the unknown readWrite properties reported by the camera?

msmeissn, 2020-08-02:
I tried writing it regardless of their reported read-only status, got "access denied" errors back.

Nikon might be "hiding" functionality, e.g. not reporting it via PTP introspection, which we already override to some degree ... but its lots of guess work.

I found meanwhile that the (read only) shutter speed variable is hidden this way too.

I tried with controlmode setting, so far did not change behaviour.
Also tried over WLaN again, also no luck.

I did not yet poked at all special variables. 0xf018 is the device language though, as the cameras was suddenly in russian;)

msmeissn, 2020-08-02:
hmm. f004 is shutterspeed readwrite

msmeissn, 2020-08-02:
f003 / f006 seems to be f-number

msmeissn, 2020-08-02:
First trys with setting them and capture-image and EXIF comparison seem to indicate they work.

I will hook those 2 up in git , but probably not today.

simont77, 2020-08-02:
Great! Thanks

msmeissn, 2020-08-02:
you can try locally with gphoto2 --set-config f003=40 for f/10 ... (my lens goes from f/3.5 -> f/16, which is f003 = 22 -> 48 in 2 steps)

simont77, 2020-08-03:
Yes, I confirm that both f003 for aperture and f004 for shutterspeed work. f004=12 sets the shutter to 1/4. Do you already find out the mapping?

msmeissn, 2020-08-03:
i committed the aperture mapping I have to current libgphoto2 git via commit 16e222c5986b345d700b6ab2ea21941067e7d1e3

shutterspeed is next.

simont77, 2020-08-03:
Here is the mapping down to 1.8

20 -> 3.2
18  -> 2.8
16 ->  2.5
14 ->  2.2
12 ->  2
10 ->  1.8

simont77, 2020-08-03:
Different issue, I'm trying to compile the current version with latest additions, but I got this compile error:

```
In file included from ptp2/ptp.c:27:
ptp2/ptp.h:3303:18: error: field has incomplete type 'struct timeval'
        struct timeval          starttime;
                                ^
ptp2/ptp.h:3303:9: note: forward declaration of 'struct timeval'
        struct timeval          starttime;
               ^
1 error generated.
```

The modification was added in commit https://github.com/gphoto/libgphoto2/commit/2ebf9928d0780b3f58e1b749870e2fee49ffd366

I'm using brew on Mac, and I can successfully compile the latest stable release 2.5.25

msmeissn, 2020-08-03:
i fixed the build hopefully

msmeissn, 2020-08-03:
i did a first hookup of shutterspeed, but for the j4 i needed tgo hack a bit and it might not work, so please test

msmeissn, 2020-08-03:
currently avauilable as "shutterspeed1" config option, will likely rename it

simont77, 2020-08-03:
just tried, I do not set any shutterspeed or shutterspeed1, only shutterspeed (readonly). Instead I see and can set aperture

simont77, 2020-08-03:
Sorry, I see only shutterspeed2 (readonly)

msmeissn, 2020-08-03:
can you see if you can "git pull" if there is perhaps somethign more new? if not, 
can you attach
gphoto2 --list-all-config
output?

simont77, 2020-08-03:
I just compiled d86133b, should be the last.
Here is list-all-config output 

[all_config.txt](https://github.com/gphoto/libgphoto2/files/5015978/all_config.txt)

msmeissn, 2020-08-03:
can you also do this with gphoto2 --debug --debug-logfile=listall.log --list-all-config   ? andattach listall.log? 
not sure where I added a bug :/

simont77, 2020-08-03:
Here it is

[listall.log](https://github.com/gphoto/libgphoto2/files/5017145/listall.log)

msmeissn, 2020-08-03:
i enhanced the generic setter/geter code some more.

can you check again? might see "shutterspeed1u" now as variable.

simont77, 2020-08-03:
ok, now there is shutterspeed1u and it can indeed set shutterspeed

simont77, 2020-08-04:
Regarding BULB instead, is there any possibility that the TerminateCapture command is somehow hidden in the same way?

msmeissn, 2020-08-04:
you can try to call it with this cmdline:

gphoto2 --set-config opcode=0x920c,0x0,0x0

for my J5 it reports "Unsupported operation", meaning the opcode is not available (not even hidden).

I will rename shutterspeed1u to shutterspeed ... can you update git and see if it still available for you?

simont77, 2020-08-04:
I will try when at home. As concern the Nikon 1 V1, is there any hope of better support, or do you already know that there are limitations from the camera? I briefly tested it, the reported props are even fewer and also acquisition did not seem reliable. I may try with it more carefully if needed.

simont77, 2020-08-05:
The new shutterspeed works.

Instead when a try the TerminateCapture opcode I get:

```
*** Error ***              
Failed to set new configuration value 0x920c,0x0,0x0 for configuration entry opcode.
*** Error (-7: 'Problema di I/O') *** 
```
with this logfile.  
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/5026838/my-logfile.txt)

The camera stalls, and I have to turn it off.

msmeissn, 2020-08-06:
The camera crashes on the command. so its not working I would say.

If you have a Nikon 1 V1 too... hmm. can you try if there are hidden properties:

gphoto2 --debug --debug-logfile=d100.log --set-config opcode=0x1014,0x5007  
and attach it here? (5007 is f-number)

simont77, 2020-08-06:
The command failed with 
```
*** Error ***              
Failed to set new configuration value 0x1014,0x5007 for configuration entry opcode.
*** Error (-1: 'Unspecified error') ***   
```

The log file is

[d100.log](https://github.com/gphoto/libgphoto2/files/5033221/d100.log)

msmeissn, 2020-08-06:
0.060413 ptp_usb_getresp [usb.c:514] (0): PTP_OC 0x1014 receiving resp failed: PTP Device Prop Not Supported (0x200a)

So not even f-numer is queryable, which means pretty no config works.

Does capture even work on the V1?

simont77, 2020-08-06:
No, it does not seem to work...

```
*** Error ***              
PTP I/O Error
ERROR: Could not capture image.
ERROR: Could not capture.
```

[capture.log](https://github.com/gphoto/libgphoto2/files/5033341/capture.log)

If it means something, I can read the battery level.

simont77, 2020-08-06:
This is the summary, with a lot of errors

[summary.txt](https://github.com/gphoto/libgphoto2/files/5033356/summary.txt)

simont77, 2020-08-07:
I tried to rename shutterspeed to shutterspeed2 to fix the issue with indi, and it actually works, but now with --list-config there are 2 shutterspeed2 properties....see the --list-config output

[config_indiFix.txt](https://github.com/gphoto/libgphoto2/files/5041330/config_indiFix.txt)

Here is what I have done (very simple, maybe too much...):
https://github.com/gphoto/libgphoto2/compare/master...simont77:master

Also, I noticed that compared with the stable release, now there is no more the expprogram2 property, which is the actual one to change PASM mode (and it's readwrite). Only the expprogram property is present, but this is readonly and it seems related to Nikon 1 "smart" program (Sport, Landscape, etc). Also bracketing disappeared. I was not able to find where the modification occurred, though. Here is the -list-config with the stable release vs your latest commit:

[config_HEAD.txt](https://github.com/gphoto/libgphoto2/files/5041295/config_HEAD.txt)
[config_stable.txt](https://github.com/gphoto/libgphoto2/files/5041296/config_stable.txt)

simont77, 2020-08-12:
Hi Marcus, did you have a change to look at my patch above (https://github.com/gphoto/libgphoto2/compare/master...simont77:master)? If it is ok for you (even with the double shutterspeed2 appearing) I can issue a pull request.

msmeissn, 2020-08-12:
I just did this manually.

I think I will also file something for Indi to make their logic a bit better

msmeissn, 2020-08-12:
can you run gphoto2 --list-all-config --debug --debug-logfile=listallconfig.txt   .... and attach the log and outpout here? 

Disappearance of expprogram2 and bracketing is likely caused by my changes to the generic getter/setter logic

simont77, 2020-08-12:
Sorry, I’m on holiday now, I will post it when I will come back home.

msmeissn, 2020-08-12:
no worries, github issues are patient :)

simont77, 2020-09-04:
Hi, I'm back. Here is the log and output of list-all-config

[listallconfig.txt](https://github.com/gphoto/libgphoto2/files/5175917/listallconfig.txt)
[output.txt](https://github.com/gphoto/libgphoto2/files/5175919/output.txt)

msmeissn, 2020-09-06:
I readded the use of GetVendorPropCodes , which should restore the functionality... 
Itweird: does not work on j1, j2,  works on j3, j4, does not work on j5 ... :(

simont77, 2020-09-06:
It does not seem to work. Expprogram2 and bracketing are still missing. Here is the log and output with the latest commit:

[listallconfig.txt](https://github.com/gphoto/libgphoto2/files/5179589/listallconfig.txt)
[output.txt](https://github.com/gphoto/libgphoto2/files/5179590/output.txt)

msmeissn, 2020-09-06:
I relaxed the check a bit more ... can you retest?

simont77, 2020-09-06:
ok, now they appeared and work

closed 2020-10-19 by simont77