gphoto/libgphoto2 #547: Hello, I hope you can update and support this camera：SONY ILCE-7RM3

xdw-123, 2020-08-01:
Fill in fields below. If you don't know how, leave it free.

SONY ILCE-7RM3

Use the Marketing name.

**USB IDs**
After running lsusb on my terminal, I got: bus 001 device 006: ID 054c: 0c33 Sony Corp

**camera summary output**
gphoto2 --summary > summary.txt
[https://github.com/xdw-123/song-camera-ilce-7RM3/blob/master/summary.txt](url)

**camera configuration output**
gphoto2 --list-all-config > list-all-config.txt
[https://github.com/xdw-123/song-camera-ilce-7RM3/blob/master/list-all-config.txt](url)

**test capture**
test if capture perhaps already works:
  gphoto2 --capture-image-and-download.This method is available
  gphoto2 --capture-preview.This method reports an error, see the document.
[https://github.com/xdw-123/song-camera-ilce-7RM3/blob/master/capture-preview.txt](url)

msmeissn, 2020-08-01:
You are using libgphoto2 2.5.9, which is very outdated. (its from 2015)

Please update to 2.5.25 first and test again.

xdw-123, 2020-08-03:
Thank you for your reply. That version of python-gphoto2. is really lower. Can you update that version?Or what should I do next?

msmeissn, 2020-08-09:
python-gphoto2 is unrelated to libgphoto2 ... you need to update the libnrary libgphoto2, the python bindings can stay the same

closed 2022-01-04 by msmeissn