gphoto/libgphoto2 #549: D3400, D3500 Connecting to PC/Smart Device hang

alterillex, 2020-08-07:
**Describe the bug**

Hello, I'm having issues with both my D3400, and D3500 Nikon Cameras. Randomly the camera(s) tend to get "stuck" in a never ending "connecting to camera" screen when taking pictures. I'm currently using --capture-image-and-download to capture and download my images to my Ubuntu PC. I've set a --wait-event=3000ms but this hasn't helped either. I've used an SD card, and no SD card running on camera internal storage with proper settings in gphoto2 config and the issue persists with both. 

My python script runs well, so well that I can take 6,000-8,000 pictures in succession without issue, but regardless of my settings this "connecting to PC" soft-lock appears at random. I've used four different Nikon D3400s and two different Nikon D3500s yet I've experienced no improvement. Ideally the goal is to run the camera seamlessly for 10,000 to 20,000 pictures.

**Name the camera**
Nikon DSC D3500
Nikon DSC D3400

**libgphoto2 and gphoto2 version**
Gphoto2 : 2.5.23
Libgphoto2: 2.5.25
Libgphoto2_port 0.12.0

**To Reproduce**
Happens at random. Cameras suddenly enter a soft-lock type state requiring the USB to be re-seated to restore functionality and clear the "waiting for PC" status/screen therefore requiring my python script to be restarted setting me back to 0 of the 10k,20k picture goal.

msmeissn, 2020-08-09:
does the camera stop working in this case? or is just the message being displayed?

alterillex, 2020-08-09:
The camera enters a soft lock type status and no longer works until the USB
cable has been reseated.

closed 2020-08-13 by alterillex