gphoto/libgphoto2 #553: error: implicit declaration of function 'flock' is invalid in C99

ryandesign, 2020-08-20:
Hi, when building libgphoto2 master on macOS with Xcode 12 beta (after working around #552), it fails with this error:

```
usbscsi/linux.c:104:6: error: implicit declaration of function 'flock' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        if (flock(port->pl->fd, LOCK_EX | LOCK_NB) != 0) {
            ^
usbscsi/linux.c:104:6: note: did you mean 'clock'?
/Applications/Xcode-beta.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.0.sdk/usr/include/time.h:109:9: note: 'clock' declared here
clock_t clock(void) __DARWIN_ALIAS(clock);
        ^
usbscsi/linux.c:128:6: error: implicit declaration of function 'flock' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        if (flock(port->pl->fd, LOCK_UN) != 0) {
            ^
2 errors generated.
```

Clang in Xcode 12 beta is different from previous versions of Xcode in that it considers implicitly declared functions to be an error rather than a warning.

This problem is happening because in libgphoto2_port/gphoto2/gphoto2-port-portability.h it defines `_POSIX_C_SOURCE`. In sys/fcntl.h (included from sys/file.h) the declaration of `flock` is guarded by:

```
#if !defined(_POSIX_C_SOURCE) || defined(_DARWIN_C_SOURCE)
```

I was able to work around it by adding `-D_DARWIN_C_SOURCE` to CFLAGS.

I'm not very familiar with `_POSIX_C_SOURCE` and `_DARWIN_C_SOURCE` so I don't know whether removing the define for `_POSIX_C_SOURCE` in gphoto2-port-portability.h or adding the define for `_DARWIN_C_SOURCE` would be an appropriate fix.

msmeissn, 2020-08-20:
it should probably just not build usbscsi in macos , as it does not make sense there?

ryandesign, 2020-08-20:
I'm not sure. It does build with these changes, but if it's only intended to work on Linux, then sure, only build it on Linux. I see in usbscsi/linux.c references to filesystem paths under /sys, and there certainly isn't any such path on macOS.

msmeissn, 2020-08-20:
libgphoto2_port/configure.ac
        case "$host" in
          *-*-mingw* | *-*-cygwin* | *-*-msvc* )
          IOLIB_LIST="$IOLIB_LIST usb1"
          ;;
          *)
          IOLIB_LIST="$IOLIB_LIST usb1 usbdiskdirect usbscsi"
          ;;
        esac

add the macos parts to the first mingw... case perhaps (if it reports as darwin, try to add it as darwin wildcard mask there)

also the usb scsi interface is likely quiote different under MacOS. We could support it, but it likely will be a different driver.

ryandesign, 2020-08-20:
If usbdiskdirect and usbscsi rely on sysfs, which is a Linux feature, shouldn't the test go the other way around?
```
        case "$host" in
          *-linux*)
          IOLIB_LIST="$IOLIB_LIST usb1 usbdiskdirect usbscsi"
          ;;
          *)
          IOLIB_LIST="$IOLIB_LIST usb1"
          ;;
        esac
```

msmeissn, 2020-08-20:
might be even better

msmeissn, 2020-08-20:
applied a fix

closed 2020-08-20 by msmeissn