gphoto/libgphoto2 #555: Sony Alpha incremental settings and DPD parsing info

dognotdog, 2020-08-24:
I've done some work on interactive control of an A7S, and I believe to have decoded some of the device property description flags that are specific to Sony. I am not sure how this could be integrated into libgphoto2, but I wanted to record what I found.

Essentially, the two hitherto unknown bytes between the data type and default value field are:
- one byte  flags, 
- one byte read/write indicator.

The flags seem to be a bit mask, with 0x01 being an incremental setting (set with ctrl value b, vs ctrl value a for non-incremental), and maybe 0x80 indicates a physical button on the device. There seem to be more values but I haven't found out what they indicate.

The read/write byte seems to be one of 0,1,2 where
- 0 is readonly
- 1 is settable through USB
- 2 is set by the camera (eg. physical dials or auto settings depending on program mode)

The read/write setting can change depending on program mode, eg. switching from manual to aperture priority changes this value for the shutter speed setting from 1 to 2.

see the parsing in [PTP Webcam](https://github.com/dognotdog/ptpwebcam/blob/e91f40035124903f4a3d261481e7800c246319bb/PtpWebcamDalPlugin/PtpCameraSony.m#L578) vs. [ptp_unpack_Sony_DPD](https://github.com/gphoto/libgphoto2/blob/16331a682333d0c1f7380e21f1c411639acc21f5/camlibs/ptp2/ptp-pack.c#L1009).

dognotdog, 2020-09-04:
As an added piece of information for `0x8X` flags, eg. buttons, even though they have bit0 set (eg. `0x81`),indicating being absolute, seem to work by "incerementing" by 2 to press, and by 1 to release.

msmeissn, 2021-10-02:
this seemed not very reliable, occasionaly reporting readonly even if it is readwrite

dognotdog, 2021-10-02:
Just randomly, or depending on exposure mode, etc? Gotta say I haven’t looked at this in a while, though.