gphoto/libgphoto2 #559: issue in using gphoto2 with go pro hero 8 black

rockstar221123, 2020-08-28:
Hi,
 I am trying to use go pro hero 8 black in ubuntu 18.04 through gphoto2.
output of gphoto2 --summary:
Camera summary:                                                                
**Manufacturer: GoPro
Model: HERO8 BLACK
  Version: HD8.01.01.60.00
 Serial Number: xxxxxxxxx
Vendor Extension ID: 0x6 (1.0)
Vendor Extension Description: microsoft.com: 1.0; android.com: 1.0;

Device Capabilities:
	File Download, File Deletion, File Upload
	No Image Capture, No Open Capture, No vendor specific capture

Storage Devices Summary:

Device Property Summary:
Friendly Device Name(0xd402):(read only) (type=0x0) invalid type, expected STR (Undefined)
Synchronization Partner(0xd401):(read only) (type=0x0) invalid type, expected STR (Undefined)**

as it shows "no image capture, no open capture, no vendor specific capture", when trying for "gphoto2 --capture-movie", the program is exiting with error .
**_Capturing preview frames as movie to 'movie.mjpg'. Press Ctrl-C to abort.      
ERROR: Movie capture error... Exiting.
Movie capture finished (0 frames)_**

Please suggest what to do.

msmeissn, 2020-08-28:
While there is a beta firmware for the GOPRO Hero 8 turning it into a webcam, we have not yet tried that or seen that.

The original GOPRO as we know only supported download.

reference to beta webcam code:

https://gopro.com/de/de/news/how-to-use-gopro-for-webcam   (or switch to english)

rockstar221123, 2020-08-28:
as per go pro, the beta firmware is for Mac only. I have flashed my go pro with the beta firmware, but not able to make your module work with it.
Kindly let me know if you are able to resolve it. Thanks !

msmeissn, 2020-08-28:
can you run 

gphoto2 --summary --debug --debug-logfile=debug.txt

and attach debug.txt here?

rockstar221123, 2020-08-29:
[debug.txt](https://github.com/gphoto/libgphoto2/files/5144840/debug.txt)

please find attached the debug.txt file. (I have updated the camera firmware to the latest one which supports webcam mode and have checked the mode by connecting the camera to my macbook and it is working fine. )
I am getting the below error, when running the command _gphoto2 --capture-image_

**_*** Error ***              
Sorry, your camera does not support generic capture
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***_**

msmeissn, 2020-09-01:
Hmm, revere engineering so far did not have any good resuilts.

can you run "lsusb -v" and attach the gopro part (or the whole output) here ?

msmeissn, 2020-09-01:
(I looked into this a bit, and I think GoPro is not implementing the webcam mode over PTP.

They seem to implement a USB network interface, and then talk the regular Wlan access protocol over it with the camera.
)

rockstar221123, 2020-09-04:
[gopro_usb_tree.txt](https://github.com/gphoto/libgphoto2/files/5172821/gopro_usb_tree.txt)

I have attached the result of the "lsusb -v" (go pro part)
In the connection menu, they have two options:
1. go pro connect: "Select go pro connect to stream video from your gopro when connected to your computer"
2. MTP : "Select MTP to download media from your gopro when connected to your computer"

----------------------------------------------------------------------------------------------------------------------------------
So, is there anyway to control the go pro through the USB network interface or should i go for the media mod (for hdmi port) and then hdmi to usb converter?

msmeissn, 2020-09-08:
is lsusb -v output the same when you do it in  "go pro connect" mode?

Can you see if it brings up some network interface in this mode?

I would currently suggest going the HDMI route ... Even on Windows this is less than stable when I read the Facebook group.

KonradIT, 2020-10-09:
This webcam (GoPro Connect) mode just exposes the control/media offload/live feed API over USB Ethernet.

With this, one can use HTTP requests to take a picture, download media, or even get a live feed off the camera (1080p is supported).